ChampSim/ChampSim #277: How to record multi-core traces?

PKUZHOU, 2022-09-22:
Hi dear developers, thank you for developing the useful simulator. 
It seems that the provided tracer can only record the single core trace. Is there any way to record multi-threading traces?  My benchmark   is a multi-threading benchmark and runs on more than one cores. 

Thank you!

ngober, 2022-09-22:
There is currently no support in ChampSim for recording or executing multi-threaded traces. There has been an increasing amount of discussion about this lately, and it's clear the simulator will eventually go this direction, but it will entail a lot of work and will require all existing traces to be recreated. So, this likely won't be supported in the short term, though it is on the roadmap.

If you are interested in this, I would welcome any suggestions or requirements that you have.

PKUZHOU, 2022-09-23:
Got it. Thank you for your kind reply!

closed 2022-09-23 by PKUZHOU