ChampSim/ChampSim #282: run my matrix multiplication on ship replacement policy

ShefaaAbed, 2022-10-11:
Hello,,
I try to run my matrix multiplication on ship replacement policy but i do not know how to do this!! 

Can anyone help please?
Thanks.

ngober, 2022-10-11:
First of all, I recommend you direct your questions to the mailing list and reserve Issues for bugs and feature requests.

You haven't given enough information to appropriately answer your question. I assume you have a program you would like to trace and simulate with ChampSim. You should look at the tracing infrastructure provided under the `tracer/` directory. Once you have traced your program, you can run the trace through ChampSim. I would recommend reading the [configuration file guide on our wiki](https://github.com/ChampSim/ChampSim/wiki/Creating-a-configuration-file).

ShefaaAbed, 2022-10-12:
yes that's what I meant, it worked thanks.
I tried to open mailing list but it doesn't work, I've got content unavailable and i tried to switch my accounts and still get the same problem.

ShefaaAbed, 2022-10-13:
Yes and I send an empty mail but I'm still waiting for the owner to accept me, this is the last email I received from you: "Your join request for champsim has succeeded but is pending approval."

ShefaaAbed, 2022-10-13:
Yes and I send an empty email but I'm still waiting for the owner to accept me, this is the last email I received from you: "Your join request for champsim has succeeded but is pending approval."

GinoAC, 2022-10-13:
Hi Shefaa you should now be able to post to the group.

closed 2022-10-13 by ngober