ChampSim/ChampSim #296: Branch MPKI bug

sawansib, 2022-12-09:
Hello,

Recently the branch MPKI has increased (in my tests and also mentioned by [Daniel A. Jiménez](https://github.com/ChampSim/ChampSim/issues/271) in the current develop branch. I found that the reason is an incorrect update of branch type information when doing the BTB update. 

Consider the [line](https://github.com/ChampSim/ChampSim/blob/7b459db03ce62d8db2c7afc7db16a4609f8b28b7/btb/basic_btb/basic_btb.cc#L124) where by default the branch type is set to ALWAYS_TAKEN. Now if we have a case where the BTB entry for a taken conditional branch already exists then the same entry's branch type will be updated from conditional (original BTB entry branch type) to always taken which is incorrect.

In my tests, I find that fixing this error will revert back the MPKI to the previous version of the develop branch. As my fixes are temporary I'm not including them here. 

Sawan

ngober, 2023-02-22:
I'm looking through the git blame for the line you linked, and it looks like a merge somewhere introduced this bug. Could you give more detail about the case you observe where the behavior is incorrect? Also, if you submit a pull request for this, I'd be eager to see your solution.

sawansib, 2023-03-07:
Hey Nathan,

Sorry for late reply!

You can see the issue in the log i am adding below. The log simply prints which IP is added to the BTB and also the result of the search in the BTB by IPs. The log is for spec_gcc_001 from the 1st IPC. 

`
btb_btbDbasic_btb_update_btb ip? 66b60c target? 66b5f0 **type? 3** taken? 1
btb_btbDbasic_btb_update_btb ip? 66b600 target? 66b60c type? 3 taken? 1
btb_btbDbasic_btb_btb_prediction ip? 66b60c target? 66b5f0 **type? 2**
`

Bug: - You can see for the IP `66b60c` the BTB is updated with branch type 3 (CONDITIONAL), but when the same IP looks in the BTB it gets the target correctly, but the branch type is 2 (ALWAYS_TAKEN) which is incorrect. 

Cause:- It happens due to the line i mentioned in the beginning of this issue. So, when the update is called in the BTB by default the branch type is set to 2 (ALWAYS_TAKEN). In the case of the taken branch there is no condition that update the default branch type of 2 set earlier. 

Solution: - The simple solution is to add a condition for taken branches. For example: -
`
 auto opt_entry = ::BTB.at(this).check_hit({ip, branch_target, type});
  
  if (taken) {
    if (opt_entry->type == ::branch_info::CONDITIONAL && type == ::branch_info::ALWAYS_TAKEN) {
      type = ::branch_info::CONDITIONAL;
    }
  }

  if (opt_entry.has_value()) {
    opt_entry->type = type;
    if (branch_target != 0)
      opt_entry->target = branch_target;
  }

  ::BTB.at(this).fill(opt_entry.value_or(::btb_entry_t{ip, branch_target, type}));
`

Here `opt_entry->type` is the original branch type while `type` is the default. So we check in the BTB if the branch is saved as CONDITIONAL and the default is still ALWAYS_TAKEN we change the default to CONDITIONAL.

I have created a pull request https://github.com/ChampSim/ChampSim/pull/318

Thanks again for looking into this.

Sawan

ngober, 2023-03-07:
Ah, I think I understand this now, and I believe the current behavior is correct. When the BTB entry is marked "always taken", this doesn't mean "unconditional", it simply means that the BTB has never seen the branch be not taken. A few lines earlier on L129, the branch is marked as conditional if it has ever been not taken. The BTB will continue to predict that the branch is taken until it has reason to assume otherwise. I think this is the correct behavior.

sawansib, 2023-03-08:
Ah okay okay i agree the behavior you described is correct. Then do you think the current MPKI is correct? Doing the changes i suggested does roll back the MPKI to the previous develop version (Where the average MPKI of server traces from IPC was about 2), so maybe the bug is somewhere else.

ngober, 2023-03-08:
Depending on how you look at it, this is either a bug (because the branch MPKI changed) or a fix for a long-standing bug (where the branch MPKI was incorrect to begin with). For my part, if I can show that the behavior of the BTB is verifiably correct now, I would say this is a fixed bug, not a bug itself.

ngober, 2023-03-13:
I see now. I've finally managed to write a test that fails before, but succeeds after your patch. I'm going to push the test onto your branch so it gets merged in. I have some another thought, but it's a code quality point and I'll include it in my code review.

Thank you for this work to find this.

sawansib, 2023-03-13:
Hi,

Great! I have written a basic solution, please update it according to your and ChampSim coding style :)

I will close this issue now, again thank you for your time.

Sawan

closed 2023-03-13 by sawansib