ChampSim/ChampSim #299: lower_translation config flag

agusnt, 2022-12-20:
Hello,

I am trying to set up a configuration (1 Core, L1D ip_stride prefetcher running with virtual address, no L2C prefetch, no LLC prefetch) where L1D demand requests go to DTLB and L1D prefetch requests go to STLB (skipping DTLB).

If I understand correctly, the lower_translate key sets the translation level for prefetch requests. However in my test when I use "lower_translate": "cpu0_STLB", all requests (demand and prefetch) go to STLB.

Is this the expected behavior? Is there any way to set the translation level for prefetching without changing the code?

I am using the develop branch.

The configuration json file looks like this:

```json
    "L1D": {                                                                                                                                                                                    
        "sets": 64,                                                                                                                                                                             
        "ways": 12,                                                                                                                                                                             
        "rq_size": 64,                                                                                                                                                                          
        "wq_size": 64,                                                                                                                                                                          
        "pq_size": 8,                                                                                                                                                                           
        "ptwq_size": 5,                                                                                                                                                                         
        "mshr_size": 16,                                                                                                                                                                        
        "latency": 5,                                                                                                                                                                           
        "max_read": 2,                                                                                                                                                                          
        "max_write": 2,                                                                                                                                                                         
        "prefetch_as_load": false,                                                                                                                                                              
        "virtual_prefetch": true,                                                                                                                                                               
        "prefetch_activate": "LOAD",                                                                                                                                                            
        "lower_translate": "cpu0_STLB",                                                                                                                                                         
        "prefetcher": "ip_stride"                                                                                                                                                                      
    },    
```

ngober, 2022-12-20:
Yes, this is the intended behavior. All requests are sent to the specified TLB without distinction.

If you want to achieve this behavior modifying only module code, you could implement a replacement policy for the DTLB that bypasses all prefetches. This would still incur the latencies for prefetches, but might be close to what you're looking for.

agusnt, 2022-12-21:
Thanks for the explanation.

closed 2022-12-21 by agusnt