ChampSim/ChampSim #301: UINT64 value is copied into UINT32 value in WriteToSet() in champsim_tracer.cpp

YashikaVerma156, 2023-01-04:
For writing source_memory/destination_memory addresses when a call is made to WriteToSet(T* begin, T* end, UINT32 r), 64bit value is copied to UINT32 r.

Source memory and destination memory operands are declared as 64bit wide in instruction.h
uint64_t destination_memory[NUM_INSTR_DESTINATIONS_SPARC] = {};
uint64_t source_memory[NUM_INSTR_SOURCES] = {};

Hence the ChampSim trace have inaccurate value for memory operands.

alberto-ros, 2023-01-07:
From what I can see, the value passed to "r" is the register number, not the register content. So 32 bits should be more than enough. Am I missing something here?

YashikaVerma156, 2023-01-08:
Actually, WriteToSet(T* begin, T* end, UINT32 r) is used for both memory and register operands.
So, for register number i.e. when register operands are extracted, it is fine. But, while extracting the memory addresses the 64-bit memory address is copied to UINT32 r, which is wrong.

alberto-ros, 2023-01-08:
I see. You are right. Something to be fixed. Perhaps the best way is to do a pull request fixing the problem.

YashikaVerma156, 2023-01-15:
ok, sure.