ChampSim/ChampSim #306: Why are the total_miss_latency values big?

gangwarady, 2023-02-05:
In _cache.cc_ file and in the _filllike_miss_ function it's given that _total_miss_latency = current_cycle - handle_pkt.cycle_enqueued;_

On printing the above _total_miss_latency_ when making a main memory access the values are in thousands like around 2000 or 3000 and sometimes the values are around 200, which doesn't make any sense.

Can somebody explain why is it like that?

ngober, 2023-02-05:
There was a bug where certain packets (I believe they were writeback packets) would not have their `cycle_enqueued` member initialized. This has been fixed on the develop branch and will be merged soon.

https://github.com/ChampSim/ChampSim/blob/develop/test/402-miss-latency.cc#L137

closed 2023-02-05 by ngober

gangwarady, 2023-02-17:
The above link is showing page not found.

ngober, 2023-02-17:
Fixed the link.