ChampSim/ChampSim #307: Reason for this Deadlock?

gangwarady, 2023-02-06:
On running the Champsim, I have got a deadlock which shows the following:

**cpu0_STLB MSHR empty

DEADLOCK! CPU 0 cycle 3000186
IFETCH_BUFFER head instr_id: 5001431 translated: 2 fetched: 2 scheduled: 0 executed: 0 is_memory: 1 num_reg_dependent: 0 event: 3000185
ROB head instr_id: 5001015 translated: 2 fetched: 2 scheduled: 2 executed: 1 is_memory: 1 num_reg_dependent: 0 event: 2002274
Load Queue Entry..........**

Can anybody tell me what could be the reason?

GinoAC, 2023-02-06:
Please provide the number of cores and benchmarks you're simulating so we can recreate the deadlock.

gangwarady, 2023-02-07:
This happens when in readlike_miss() function I'm forcefully changing the mshr_entry to one of the packets in MSHR, so that an inflight miss can occur.

gangwarady, 2023-02-07:
I change the incoming handle_pkt.address to the address of any packet present in MSHR, so that mshr_entry!=MSHR.end(), but I get this deadlock. Also I've tried directly making the mshr_entry equal to any packet present in MSHR.

gangwarady, 2023-02-07:
This is what is printed on Deadlock:

**cpu0_STLB MSHR empty

DEADLOCK! CPU 0 cycle 4171625
IFETCH_BUFFER empty
ROB head instr_id: 5075262 translated: 2 fetched: 2 scheduled: 2 executed: 1 is_memory: 1 num_reg_dependent: 0 event: 3171625
Load Queue Entry
[LQ] entry: 5 instr_id: 5075456 address: 1abfcb110 translated: 2 fetched: 1
[LQ] entry: 8 instr_id: 5075566 address: 0 translated: 1 fetched: 0
[LQ] entry: 16 instr_id: 5075262 address: 1abfcb108 translated: 2 fetched: 1
[LQ] entry: 24 instr_id: 5075487 address: 1abfcb138 translated: 2 fetched: 1
[LQ] entry: 27 instr_id: 5075293 address: 1abfcb130 translated: 2 fetched: 1

Store Queue Entry
[SQ] entry: 2 instr_id: 5075374 address: 199799a08 translated: 2 fetched: 2

cpu1_ITLB MSHR Entry
[cpu1_ITLB MSHR] entry: 0 instr_id: 5609256 address: 100ab full_addr: 402aeb type: 0 fill_level: 1 event_cycle: 18446744073709551615

cpu1_L1I MSHR empty

cpu0_DTLB MSHR empty

cpu0_ITLB MSHR empty

cpu0_L1I MSHR empty

cpu1_STLB MSHR Entry
[cpu1_STLB MSHR] entry: 0 instr_id: 5609256 address: 100ab full_addr: 402aeb type: 0 fill_level: 1 event_cycle: 18446744073709551615
[cpu1_STLB MSHR] entry: 1 instr_id: 5609258 address: 1fc01bca9b6 full_addr: 7f006f2a6d80 type: 0 fill_level: 1 event_cycle: 18446744073709551615

DEADLOCK! CPU 1 cycle 4171625
IFETCH_BUFFER empty
ROB head instr_id: 5609258 translated: 2 fetched: 2 scheduled: 2 executed: 1 is_memory: 1 num_reg_dependent: 0 event: 4171591
Load Queue Entry
[LQ] entry: 0 instr_id: 5609258 address: 0 translated: 1 fetched: 0
[LQ] entry: 1 instr_id: 5609259 address: 0 translated: 1 fetched: 0
[LQ] entry: 2 instr_id: 5609263 address: 0 translated: 1 fetched: 0
[LQ] entry: 3 instr_id: 5609264 address: 0 translated: 1 fetched: 0

Store Queue Entry

cpu1_PTW MSHR Entry
[cpu1_PTW MSHR] entry: 0 instr_id: 5609256 address: 6600000 v_address: 402aeb type: 0 translation_level: 0 fill_level: 4 event_cycle: 18446744073709551615
[cpu1_PTW MSHR] entry: 1 instr_id: 5609258 address: 191e00008 v_address: 7f006f2a6d80 type: 0 translation_level: 0 fill_level: 4 event_cycle: 18446744073709551615

cpu0_PTW MSHR empty

cpu1_L1D MSHR Entry
[cpu1_L1D MSHR] entry: 0 instr_id: 5609256 address: 198000 full_addr: 6600000 type: 4 fill_level: 4 event_cycle: 18446744073709551615
[cpu1_L1D MSHR] entry: 1 instr_id: 5609258 address: 6478000 full_addr: 191e00008 type: 4 fill_level: 4 event_cycle: 18446744073709551615

cpu0_L1D MSHR Entry
[cpu0_L1D MSHR] entry: 0 instr_id: 5075262 address: 6aff2c4 full_addr: 1abfcb108 type: 0 fill_level: 4 event_cycle: 18446744073709551615

cpu1_L2C MSHR Entry
[cpu1_L2C MSHR] entry: 0 instr_id: 5609256 address: 198000 full_addr: 6600000 type: 4 fill_level: 4 event_cycle: 18446744073709551615
[cpu1_L2C MSHR] entry: 1 instr_id: 5609258 address: 6478000 full_addr: 191e00008 type: 4 fill_level: 4 event_cycle: 18446744073709551615

cpu0_L2C MSHR Entry
[cpu0_L2C MSHR] entry: 0 instr_id: 5075262 address: 6aff2c4 full_addr: 1abfcb108 type: 0 fill_level: 4 event_cycle: 18446744073709551615

LLC MSHR empty

cpu1_DTLB MSHR Entry
[cpu1_DTLB MSHR] entry: 0 instr_id: 5609258 address: 1fc01bca9b6 full_addr: 7f006f2a6d80 type: 0 fill_level: 1 event_cycle: 18446744073709551615**

gangwarady, 2023-02-07:
Number of cores are 2 and traces are bwaves_1861B & bzip2_281B

GinoAC, 2023-02-07:
The original resources allocated in the core are for the packet's original address. When you force it to hit in the MSHR, the request to the original address never completes and never releases the cores' resource.

gangwarady, 2023-02-07:
Could there be a way to perform a hit in the MSHR without deadlock?

GinoAC, 2023-02-07:
Depends on what your end goal is. I recommend reading the code on how requests are sent from ooo_cpu.cc and handled in the caches to decide what would be best for your work. 

I'm going to go ahead and close this issue as the source of the deadlock is not due to a bug in ChampSim. Please feel free to post design-oriented questions to the ChampSim google group, where myself or others may be able to assist you.

closed 2023-02-07 by GinoAC

gangwarady, 2023-02-07:
Thanks can you give me the link of the google group please.

GinoAC, 2023-02-07:
Sure! Here you go:
https://groups.google.com/g/champsim

gangwarady, 2023-02-07:
Hey for some reason, I'm not able to access it. Says content unavailable.

GinoAC, 2023-02-07:
Sorry about that. Send an empty email to champsim+subscribe@googlegroups.com and I'll be able to add you as a member

Upasna55, 2023-02-13:
Hello GinoAC

I get Content Unavailable even after sending email to `champsim+subscribe@googlegroups.com`