ChampSim/ChampSim #313: Segmentation fault on disabled page randomization, specific page randomization seeds

maccoymerrell, 2023-02-24:
When disabling page randomization, or using specific seed values, a segmentation fault occurs at some point during runtime. Varies between traces and seed values, but typically within the first 100 million instructions.

**To Reproduce**
Steps to reproduce the behavior:
1. Disable page randomization by editing vmem.cc
```
VirtualMemory::VirtualMemory(uint64_t capacity, uint64_t pg_size, uint32_t page_table_levels, uint64_t random_seed, uint64_t minor_fault_penalty)
    : minor_fault_penalty(minor_fault_penalty), pt_levels(page_table_levels), page_size(pg_size),
      ppage_free_list((capacity - VMEM_RESERVE_CAPACITY) / PAGE_SIZE, PAGE_SIZE)
{
  assert(capacity % PAGE_SIZE == 0);
  assert(pg_size == (1ul << lg2(pg_size)) && pg_size > 1024);

  // populate the free list
  ppage_free_list.front() = VMEM_RESERVE_CAPACITY;
  std::partial_sum(std::cbegin(ppage_free_list), std::cend(ppage_free_list), std::begin(ppage_free_list));

  next_pte_page = 0xffff'ffff'ffff'ffff;
  ppage_free_list.pop_front();
}
```
2. Run trace 605.mcf_s-1152B.champsimtrace.xz
3. Wait until the segfault occurs

**Expected behavior**
Champsim should not segfault in this case. Several other seeds can generate similar results, but no randomization is the most consistent at causing the segfault.

**Output**
/tmp/par-job-1_t5QM6: line 2:  5360 Segmentation fault      (core dumped) ~/ChampSimRowhammerSPP/bin/champsim -w0 -i2000000000 -o mcf_spp0 -v0 /home/Benchmark/champsim_trace/speccpu/6xx/605.mcf_s-1152B.champsimtrace.xz > champsim_mcf_spp0.txt

**Desktop (please complete the following information):**
 - OS: RHEL
 - Compiler g++

**Additional context**
Occurs with the same seeds, with or without prefetching, at around the same instruction.
Below is a gdb trace of the segfault from previous runs, although this is from the main branch and not develop (this problem still exists in develop, but impacts fewer seeds).
[image]

maccoymerrell, 2023-05-11:
I found a quick segfault that impacts the develop branch from January.
trace: mcf-1536B
prefetcher: bingo
randomization: off

Other changes made to Champsim:
2 MB LLC
1 MB L2

segfaults within 1 second of start.

ngober, 2024-11-08:
I'm revisiting this before #509 gets merged. Is this still occurring?

maccoymerrell, 2024-11-08:
As far as I know, it's not still occurring. I will say that we should probably have randomized address space enabled by default, and have a runtime flag or config option to disable it (or vice-versa). When running champsim with different physical-DRAM mappings this makes a big difference, and right now being able to switch between the two is probably valuable. I think I can open a PR to do this.

closed 2024-11-10 by ngober