ChampSim/ChampSim #319: Questions about virtual/physical addresses with regard to caches and TLBs

Freedom-is-slavery, 2023-03-07:
Hi,
Thank you for the excellent work so I'm able use ChampSim to simulate my own prefetching algorithms, but I'm a little confused when I walk through the source code in master branch and the paper you guys submitted in arXiv last year (actually the architectural diagram Fig.3 in the paper, so I attach it below).
[image]
The core_inst.cc which is generated by config.sh and uses the default configuration json file shows that only L1I and ITLB objects set va_pref = 1 (i.e., use virtual address to do prefetching)
```cpp
CACHE cpu0_L1I("cpu0_L1I", 1.0, 1, 64, 8, 64, 64, 32, 8, 3, 1, 2, 2, LOG2_BLOCK_SIZE, 0, 1, 1, 5, &cpu0_L2C, CACHE::pref_t::CPU_REDIRECT_pprefetcherDno_instr_, CACHE::repl_t::rreplacementDlru);
CACHE cpu0_ITLB("cpu0_ITLB", 1.0, 1, 16, 4, 16, 16, 0, 8, 0, 1, 2, 2, LOG2_PAGE_SIZE, 0, 1, 1, 5, &cpu0_STLB, CACHE::pref_t::pprefetcherDno, CACHE::repl_t::rreplacementDlru);
```
I also found this in `CACHE::readlike_hit, cache.cc`:
```cpp
  // update prefetcher on load instruction
  if (should_activate_prefetcher(handle_pkt.type) && handle_pkt.pf_origin_level < fill_level) {
    cpu = handle_pkt.cpu;
    uint64_t pf_base_addr = (virtual_prefetch ? handle_pkt.v_address : handle_pkt.address) & ~bitmask(match_offset_bits ? 0 : OFFSET_BITS);
    handle_pkt.pf_metadata = impl_prefetcher_cache_operate(pf_base_addr, handle_pkt.ip, 1, handle_pkt.type, handle_pkt.pf_metadata);
  }
```
1. So firstly I want to make sure whether only L1I work on virtual address space?
2. If so, for L1D, unified L2C and shared LLC, do all of them work on the physical address space (i.e., physically indexed and physically tagged, PIPT)? Is there any translation process involving TLB and page table walker between L1I and L2?
3. Additionally, does the out-of-order core issues memory requests for instructions and data directly to L1I and LID, respectively? Is there any translation process between core and L1D since I think the out-of-order core works on virtual address space while L1D on physical address space?
4. What's the meaning of the configurability of virtual and physical address for TLB prefetching?

Thanks a lot~

Freedom-is-slavery, 2023-03-09:
Got it, and thank you for reply! So when L1I prefetcher works in virtual address space, do the prefetching requests go through TLB and PTW to convert VA to PA and then use PAs before fetching cache lines from lower level cache, say L2C? Is the latency of translation process counted in prefetcher's latency thus influencing overall system performance?

ngober, 2023-03-09:
That is correct. The prefetcher does not have a latency separate from the underlying cache/TLB system.

alberto-ros, 2023-03-09:
Nathan, correct me if I am wrong, but in the develop branch, when the prefetcher works with virtual addresses, the time to do the translation is considered (i.e., the packet is sent to TLB for translation in the same way as translation requests).

ngober, 2023-03-09:
Yes, that's correct too. Instruction/data caches check tags in the physical space, so their virtual-space prefetches must be translated. TLBs only ever prefetch in the virtual space, since they tag in the virtual space.

Freedom-is-slavery, 2023-03-10:
Understood it.
BTW, I have another question about RFO access, does it mean request for ownership in MOESI coherence protocol which acquires cache line and invalidate all of other copies? In champsim, does it only triggered by store requests? So can I treat it as a special type of writeback?

ngober, 2023-03-10:
At least right now, ChampSim does not implement any kind of coherence. You can view RFO as a kind of write. It is only issued for stores.

closed 2023-03-27 by ngober