ChampSim/ChampSim #329: Which calculation method is used for memory bandwidth?

Yujie-Cui, 2023-04-22:
Hello, I want to calculate the memory bandwidth of my configuration, but there are so many variables related to bandwidth, which calculation method should I use?

        #define DRAM_CHANNELS 1 // default: assuming one DIMM per one channel 4GB * 1 => 4GB off-chip memory
        #define LOG2_DRAM_CHANNELS 0
        #define DRAM_RANKS 8 // 512MB * 8 ranks => 4GB per DIMM
        #define LOG2_DRAM_RANKS 3
        #define DRAM_BANKS 8 // 64MB * 8 banks => 512MB per rank
        #define LOG2_DRAM_BANKS 3
        #define DRAM_ROWS 32768 // 2KB * 32K rows => 64MB per bank
        #define LOG2_DRAM_ROWS 15
        #define DRAM_COLUMNS                                                                                                   \
            32 // 64B * 32 column chunks (Assuming 1B DRAM cell * 8 chips * 8 transactions = 64B size of column chunks) => 2KB
               // per row
        #define LOG2_DRAM_COLUMNS 5
        #define DRAM_ROW_SIZE (BLOCK_SIZE * DRAM_COLUMNS / 1024)
        
        #define DRAM_SIZE (DRAM_CHANNELS * DRAM_RANKS * DRAM_BANKS * DRAM_ROWS * DRAM_ROW_SIZE / 1024)
        #define DRAM_PAGES ((DRAM_SIZE << 10) >> 2)
    #define DRAM_CHANNEL_WIDTH 32  // 32 bytes
    #define tRP_DRAM_CYCLE 11
    #define tRCD_DRAM_CYCLE 11
    #define tCAS_DRAM_CYCLE 11
    tRP  = tRP_DRAM_CYCLE  * (CPU_FREQ / DRAM_IO_FREQ); 
    tRCD = tRCD_DRAM_CYCLE * (CPU_FREQ / DRAM_IO_FREQ); 
    tCAS = tCAS_DRAM_CYCLE * (CPU_FREQ / DRAM_IO_FREQ); 
    DRAM_MTPS = 1600;  //1600MT/s
    #define CPU_FREQ 4000  //4GHZ
    #define DRAM_IO_FREQ 800 //0.8GHZ
    DRAM_MTPS = 1600;
    #define _DRAM_LATENCY		240	//Cycles

What is the memory bandwidth in this configuration? It's called 37.5GB/s in the paper, but I can't get the same result. The paper's authors did not respond to emails.

First:

The `IO frequency` is `0.8GHZ`, `DRAM_CHANNEL_WIDTH` is `32byte`, so the memory bandwidth is `IO frequency*DRAM_CHANNEL_WIDTH *RANK_number` =   `0.8*32*8/8 =25.6GB/s`
Is that right?

second 
DRAM_MTPS = 1600MT/s, 
memory bandwidth = DRAM_MTPS * data_width/8 = `1600*32byte/1000 = 51.2GB/s` 
How to calculate is right?

ngober, 2023-04-24:
Wow, this must be a very old version of ChampSim, so I can't be sure how you arrived at this configuration. But, I think I see the difference between the two. The DRAM model assumes DDR, so there are 2 transfers per cycle. When you calculated via the frequency, you get a value that is half the value you get when you calculate via the MTPS.

Yujie-Cui, 2023-04-25:
@ngober   Thanks a lot for your answer. I would like to ask another question, is there any additional setup required to run GAP trace(https://utexas.app.box.com/s/2k54kp8zvrqdfaa8cdhfquvcxwh7yn85/folder/132804668078) on ChampSim?  I have an deadlock error when running.

Really Thanks

GinoAC, 2023-05-03:
If they follow the ChampSim trace format, they should not require additional setup.

I'll go ahead and close this since your original concern was addressed.

closed 2023-05-03 by GinoAC