ChampSim/ChampSim #343: Access to RQ and WQ sizes from the prefetcher

agusnt, 2023-05-18:
Hello,

I am trying to get the RQ/WQ size in the prefetcher_initialize() function. In previous version of development I done this with the get_size() function. However, in the current version I was not be able to find any way to do it.

May be possible to implement a get_[rq,wq,pq]_size()?

ngober, 2023-05-18:
That's correct. Ever since #303, `CACHE::get_size()` returned 0 for anything other than the MSHR. This is because each upper level now has its own set of queues, so it is ambiguous which queue's size would be returned. We could implement something, but what value would we expect from it?

agusnt, 2023-05-18:
The size of each queue. For example, if the configuration of L1D is the following one:
    ```
"L1D": {                                                                                                                                                                  
        "sets": 64,                                                                                                                                                           
        "ways": 12,                                                                                                                                                           
        "rq_size": 64,                                                                                                                                                        
        "wq_size": 64,                                                                                                                                                        
        "pq_size": 8,                                                                                                                                                         
        "mshr_size": 16,                                                                                                                                                      
        "latency": 5,                                                                                                                                                         
        "max_tag_check": 2,                                                                                                                                                   
        "max_fill": 2,                                                                                                                                                        
        "prefetch_as_load": false,                                                                                                                                            
        "virtual_prefetch": false,                                                                                                                                            
        "prefetch_activate": "LOAD,PREFETCH",                                                                                                                                 
        "prefetcher": "no"
    },
```

For RQ it should 64.

I do not know if this is the answer for your question.

ngober, 2023-05-18:
I see. I've been thinking about this since the patch I mentioned above. Right now, it's true that all of the queues to a given lower level from the upper levels have the same size, and it's true that the `"rq_size"` keys fix this. I could see how it is reasonable to just return that value.

But, I think eventually I want to move away from specifying the sizes directly in the configuration files. It is really easy to choose a bad configuration value here, for example a number that is less than the number of MSHRs in the upper level. I'm thinking instead to move to a calculated value that is the product of the number of MSHRs in the upper level and the hit latency of the lower level. Maybe then we can expose a `"rq_factor"` key that would then multiply this calculated value. I think it's a lot more expressive to say "I want to make the queue twice as big as it usually is" than to say "I want this queue to be 32 elements long now".

But, if you're researching this, I'll ask you: do you see benefit to configuring different queue sizes? Would it be better to specify the queue size in a different way?

alberto-ros, 2023-05-18:
Hi Nathan,

My opinion is that it is good as it is now. I prefer to set the number that each queue has, so I can tune it. Otherwise, I need to calculate the size based on some formula and it is less expressive in my opinion.

It is true that now some parameters may be set in a "bad" way. But I see better solutions. Two examples are:
 
 1. Offering several config files like "AlderLake", "Zen4", etc. I think by default it is still one similar to "IceLake" (but with a bimodal branch predictor, not prefetch and other not tuned paramaters).

 2. Adding warnings, when the specified value is below a standard calculated value.

I hope this than help.

agusnt, 2023-05-19:
I agree with Alberto, I find easier to specify the size of queues than saying I want a read queue x2 bigger than before. My best solution, as Alberto said, will be specified the size of the queues and if there are not correct trigger a warning.

Anyway, from my perspective to implement a prefetcher I will only need access to a var/function that give me the side of the queues.

ngober, 2023-05-21:
This makes sense. Still, I want to leave open the opportunity to let the queues have different sizes in the future. 

What would you think of a member function like

    std::vector<std::size_t> CACHE::get_rq_size() const;

This gives all of the sizes back in a vector. For now, you can just `.front()` it, since it's guaranteed to have at least one member and they're all the same.

agusnt, 2023-05-25:
`std::vector<std::size_t> CACHE::get_rq_size() const;`

This will be great.

Thank you

commit 5ed1e93e (2023-05-30): Added occupancy and size getters (#347)

Resolves #343 

This patch adds const-qualified getters to the cache model to observe
cache sizes and occupancies.
closed 2023-08-09 by ngober