ChampSim/ChampSim #344: prefetch_hit parameter not passed any more to prefetcher_cache_operate

alberto-ros, 2023-05-18:
Hello, 

Until recently the prefetch_cache_operate function had a parameter called prefetch_hit. It was used by many prefetchers that now cannot be ported without modifying the prefetcher interface.

The information about prefetch_hit is still in cache.cc: (way->prefetch && !handle_pkt.prefetch_from_this)

Do you think it could be added back or otherwise why it was removed?

Thanks!

ngober, 2023-05-18:
Was there actually such a parameter? I don't recall it, and it's pretty far back in the git history if it was there.

The prefetcher does not actually activate on its own requests. [We have a test demonstrating this.](https://github.com/ChampSim/ChampSim/pull/303/files#diff-f12a80a9b06758711decb7a6dadca0e5d80e28206c03953ca5e64551130254f0R552) Prefetches from upper levels can, however. They are identified by the `PREFETCH` type. If a prefetcher could trigger itself, I suspect a lot of designs would be flawed, since they would be speculating on their own speculations.

GinoAC, 2023-05-18:
There was such a parameter that survived longer in the master branch.

That particular parameter was not to intended to be used to trigger prefetches but rather to provide feedback on whether a prefetch was useful. It was extremely useful for training and as it stands, you cannot identify whether a hit was from a prefetch or not without tracking all prefetches in the prefetcher itself.

alberto-ros, 2023-05-18:
This is the possible solution based on Gino's comment:

metadata_thru = impl_prefetcher_cache_operate(pf_base_addr, handle_pkt.ip, hit, handle_pkt.type, way->prefetch && !handle_pkt.prefetch_from_this);

alberto-ros, 2023-05-18:
Actually, as Nathan said, since the function is not called for own prefetchers, it would simply as:

metadata_thru = impl_prefetcher_cache_operate(pf_base_addr, handle_pkt.ip, hit, handle_pkt.type, way->prefetch);

ngober, 2023-05-18:
Oh, I see what you mean. This is present in instruction prefetchers on the master branch. The develop has changed these to match the data prefetchers. 

I can see how this would be useful information. I think I misinterpreted the original post to mean "issue a prefetch that hits", but you meant "demand hit to prior prefetch".

There's not much that we can do about porting the IPC prefetchers right now. We can't easily change the signatures back. However, there's some potential in #330. If that merges, then we can add an alternate interface that includes `prefetch_hit`. See #297 for an example of this.

alberto-ros, 2023-05-18:
With that change, prefetchers from IPC-1 can be ported without much modifications (an no modifications outside the prefetcher code) to the current develop branch.

ngober, 2023-05-18:
OK, I'm marking this as a bug, since this is technically a regression. It can be resolved after #330. For now, do what you can to approximate the behavior.

ngober, 2023-05-23:
Moving the discussion back here.

I'm always hesitant to change any of the module interfaces, even to add a `prefetch_hit` parameter. A huge reason for #330 was to allow these kind of module interface changes to be less painful and to rely on the compiler to detect the correct interface to use. The reason I don't like changing interfaces is because it will cause code to be released that rely on that interface (see the IPC-1 issue that this discussion began with), and so we are in a way obligated to maintain support for that interface in perpetuity.

So, I want to be very careful. What should the hook function signature look like? It currently looks like

    uint32_t prefetcher_cache_operate(uint64_t addr, uint64_t ip, uint8_t cache_hit, uint8_t type, uint32_t, metadata_in)

#297 will add the signature

    uint32_t prefetcher_cache_operate(champsim::address addr, champsim::address ip, uint8_t cache_hit, uint8_t type, uint32_t, metadata_in)

How would we extend these to include `prefetch_hit`?

alberto-ros, 2023-05-24:
Yes, I would extend it that way. Another thing I found is that for some of the prefetchers in IPC-1 the expected value of prefetch_hit was either 0 or 1. It was true for that version of the simulator, and some prefetchers have things like prefetch_hit == 1.  I am just mentioning this because I found that problem when porting them, but I am okay with any decision you take regarding this.

PR #349 (2023-05-30): Added useful_prefetch parameter to prefetchers
Resolves #344 

This patch adds a parameter to the prefetcher cache_operate hook that indicates whether the hit block was a prefetch or not. This will help prefetchers get good feedback on their performance, and it also allows easier migration from the IPC-1 prefetchers. This is, however, a breaking change for all existing code in the name of bringing instruction prefetchers and data prefetchers together.
commit 7a83fae0 (2023-05-30): Added useful_prefetch parameter to prefetchers (#349)

Resolves #344 

This patch adds a parameter to the prefetcher cache_operate hook that
indicates whether the hit block was a prefetch or not. This will help
prefetchers get good feedback on their performance, and it also allows
easier migration from the IPC-1 prefetchers. This is, however, a
breaking change for all existing code in the name of bringing
instruction prefetchers and data prefetchers together.
closed 2023-05-30 by ngober