ChampSim/ChampSim #350: User Defined Inclusivity

GinoAC, 2023-05-31:
There are a few other features ahead in the queue, but we've been talking about adding an inclusivity knob for a while. The caches' queues now allow packets to traverse between higher/lower levels (TODO: reference PR number) which can enable invalidation packets enforcing inclusive caching to be sent from a lower level to a higher level cache (i.e. from the L2 to the L1). This change would at minimum require:

1) Adding an invalidation packet type
2) Adding an additional invalidation request queue
3a) A new cache function that specifically handles invalidation packets and removes the address from the cache. 
3b) Logic to handle any race conditions between invalidation requests and in-flight requests (i.e. an MSHR entry or in-flight request is being handled in the cache when an invalidation request arrives). This can easily be handled by delaying the invalidation until the in-flight request is complete. 

I'm open to any thoughts on this or if anyone would like to tackle this feature.

ngober, 2023-05-31:
I like this. I have a few design thoughts:

- The caches already have `CACHE::invalidate_entry()` that takes an address and invalidates it if it exists in the tag dictionary. It's not tested because it's never called, but a good idea would be to start with a test for invalidation.
- The invalidation packet type should follow the pattern of `champsim::channel::request_type` and `champsim::channel::response_type`, and be given as a using-member of the channel model.
- We will definitely need tests for race conditions. If I've learned one thing working on this project, it's that the edge case I think will never happen in fact happens in a half dozen traces.

GinoAC, 2023-05-31:
Ah I thought the `invalidate_entry()` function was removed so that's certainly a good place to start. 
Good point, there will need to also be an `invalidation_ack` response type to mirror the invalidation request. 
Coming from coherence-land, this isn't very prone to race conditions so long as request collisions are properly handled. This will become more of a concern if we begin to support inter-core requests.

ngober, 2023-06-15:
I set up the `feature/350-inclusivity` branch for work on this. I've added a basic implementation where if a cache encounters an RFO request, it issues invalidation requests to all its upper levels. It passes a basic test. If we want to move from that to strict forms of clusivity, it should be something like this.

GinoAC, 2023-06-16:
Future additions to this need to specify how to handle an invalidation when there are inflight MSHR entries.

I'm looking through this and overall I like the design. In particular, I like that the channel handles the packet types themselves which will remove traffic the cache model does not need to interact with. This opens up modeling colocated metadata or coherence structures which may only need to signal the cache model instead of requiring the packets to travel through the cache model.

ngober, 2023-06-16:
I just pushed a patch to propagate the invalidation to all upper levels when the invalidation is handled. So, now RFOs are in fact properly owning.

But, you're right, there's still some edge cases, and outstanding misses are most of them. Thinking about it, here are the cases I can see:

1. An invalidation request arrives from the lower level while a tag check is in flight. This implies that further upper levels have an open MSHR. Should the tag check be cancelled?
2. An invalidation request arrives from the lower level while a TLB lookup is occurring. The invalidation request gives a physical address, so it may be difficult to determine that the invalidation request applies to the packet.
3. An invalidation request may pass by a packet in a channel. The invalidation request will encounter an MSHR when it arrives. Should the packet be cancelled?
4. The packet in question may itself be an RFO. Does that change anything?

I'll ask your opinion as the doctor of memory consistency. What do you think needs to happen in these cases?

ngober, 2023-06-16:
I also have the question of acknowledgements. The invalidations probably need to acknowledge completion. However, the invalidation requests are sent at tag check time, the instant before the hit returns. Should it be the case that a hit does not return until the invalidations are acknowledged? This will add quite a bit of of latency to certain misses. It would also mean we have to move the sending of invalidation requests to somewhere other than `CACHE::try_hit()`.

GinoAC, 2023-06-18:
First off, welcome to Coherence Land! These are great questions. Adding proper RFOs implies Ownership and raises the complexity of the interactions you're describing. 

Our first steps should be as simple as possible to lay the foundation for even more complex interactions. This will result in much lower performance due to many stalling requests but will make adding optimizations much easier later. The rule of thumb, for now, is First Come, First Serve. Whatever reaches the resource first is what decides the next interaction. Assuming the most straightforward approach without regard to performance: 

1) The tag check should continue; the invalidation will wait behind the demand request and then propagate to the upper levels. 
2) Invalidations should not affect packets, only memory and metadata structures (for now). The invalidation will need to stall until the previous request is complete.
3) The packet should not be canceled. The following interaction depends on which cache levels the invalidation and LOAD/RFO occur if we enable shared memory (which does not necessarily mean we need to implement a shared state), the caches' inclusivity, and what coherence model we're applying.
4) Github was breaking my format; see 3.
5) A hit should return once the acknowledgment is received. Imagine core A and core B with 2 levels of private cache and a shared LLC. 
5A) Since dirty data only propagates during writebacks, if Core A has the data dirty in the L1 and we allow Core B's requests to hit in the LLC before acknowledgment completes, it will receive stale data. 
5B ) If Core A has clean data in its L1 and Core B's RFO hits in the LLC. If Core B completes the RFO and writes to its L1 before the invalidation reaches Core A's L1, then Core A and Core B will have different values in their private caches, risking incoherence if Core A accesses its L1. 

These are the high-level answers, but we'll need to define the scope of this patch to implement/enforce them. 

1. Do we want to allow shared memory between cores? This may seem obtuse, but are you envisioning these interactions occurring due to evictions or requests from other cores? 

If we do not allow shared memory, then the coherence model gets tabled for now, and RFOs are back to being read requests to the LLC... for now.  

2. (ignore if you said no to 1) Are we intending this to be the beginning of a (substantial) coherence patch, or are we aiming to have the correct interactions for now?  

If we want a coherence foundation, we'll start separating the cache model from the logic that decides which requests stall, send responses, and interact with the cache model. We would also need to add a buffering structure for stalling requests, logic to allow stalling requests to continue and add metadata to the MSHR to track transient state. This list grows depending on if we want to start with a broadcast or directory protocol (I'd prefer a directory, but broadcast is technically easier to implement). 

Adding invalidations, responses, acknowledgments, and a notion of ownership begins the journey into coherence. If we want this patch to be focused on user-defined inclusivity, we may want to only allow invalidations for evictions and table RFOs for a coherence patch (discussion: #360). I'd also be delighted to build a simple model in this patch and then extend it later.

ngober, 2023-06-19:
I'm going to respond to your points in reverse. I agree that we should be clear about the scope of this patch. This patch should focus on inc/exclusivity, not on coherence. However, having the tools in place for invalidations is essential for both, so we should be rigorous here for the sake of less work later.

I do want to eventually allow shared memory. I have users that have requested it from me, especially as it pertains to atomics. But, that's probably for #360.

I'm learning here that invalidation requests should _not_ be handled in parallel with tag checks, but rather in sequence. What I'm seeing from number 5 in particular is that we will need a set-aside (MSHR) for invalidations, like the translation setaside that's there already. Will all exclusive hits be stalled by invalidation up-propagations? Will this stall the cache, or are there MSHRs for this?

GinoAC, 2023-06-23:
Ok, so I've had some time to think about this. I think if we're keeping it inclusivity-oriented, we can limit the effects of the RFO's by making them only invalidate the lower levels (i.e. a write to an exclusive L1 should invalidate the cache entry in the LLC). 

Correct, for now at least, we need to handle these in sequence. We do need a set-aside for invalidations whether that's in the MSHR, or we can add a simple Transition Buffer (TB) implementation that can track cache lines moving from valid to invalid. This might be a good starting point to build #360 off of since it moves towards separating tracking cache contents and the contents' coherence state. 

Again, I like these questions. As a starting point, if a cache line is being invalidated and receives a demand we can immediately treat it like a cache miss and send the request to the next level of the cache. I'm sure there's an optimization we can apply, but for now I'm going to push for keeping it simple at the cost of performance.

ngober, 2023-06-25:
OK. This is all coming together. Do you think I should remove the RFO invalidations from this patch and push it to #360 instead? Then this patch would just focus on clusivity invalidation.

I see a wrinkle we need to iron out. I may need to dig into the literature to answer this well, but here it is.

Inclusivity refers only to a relationship between a cache and its immediate lower level. That is, an L1C cannot be exclusive of the L3C, only the L2C. When we say that we have an "exclusive L1", what we mean is that anything present in the L1 is **not** in the L2. It could be in the L3, if the L2/L3 relationship is non-exclusive.

Now from specification to design. How do we specify this on the user side? We could allow the `"lower_level"` key to be an object, where we can specify

    {
      "L1D": {
        "lower_level": {
          "name": "my_L2C",
          "inclusivity": "exclusive"
        }
      }
      "caches": [ { "name": "my_L2C" } ]
    }

where the legal values for `"inclusivity"` are `["inclusive", "non_inclusive", "exclusive"]`, and the current specification (specifying the `"lower_level"` key as a name of a cache), defaults to non-inclusive. We could also add `"inclusivity"` as a member of the cache object and call it good, as in

    {
      "L1D": { "inclusivity": "exclusive" }
    }

I think the first option is more clear, but not actually more specific.

On the C++ side, it's a little bit weirder, since both upper and lower level need to know about the clusivity of the relationship. The lower level needs to know to invalidate on a hit (if exclusive), and to invalidate on eviction (if inclusive). So, we will have to specify that somehow. It's ugly, but we could specify the `CACHE::Builder::upper_levels::value_type` as a `std::pair<champsim::channel*, champsim::inclusivity>`, where the second member is an enum class with the three options. That makes some parts of the code that care about upper levels but not clusivity (looking at `CACHE::get_rq_occupancy()` especially) a bit clunky, and it makes the specifications for `CACHE::Builder` pretty weird, because now we'd have to specify

    CACHE my_llc{CACHE::Builder{champsim::defaults::default_llc}
        .upper_levels({std::pair{&to_cpu0_l2_channel, champsim::inclusivity::non_inclusive}, std::pair{&to_cpu1_l2_channel, champsim::inclusivity::non_inclusive}})
    };

which is mind-numbingly tedious.

I'll iterate over the design ideas in my head, but if you have any thoughts, drop them here.

GinoAC, 2023-06-29:
As excited as I am to see RFOs move closer to being fully implemented, I think it might be best to wait and introduce them later. 
 
Yikes, that is text-book tedious.
 
I don't agree that clusivity is solely defined between immediate cache levels. It's fairly common to group caches' clusivity by whether or not they're shared. A common configuration in literature is a noninclusive private cache and an inclusive shared cache, with the L1 and L2 being noninclusive of one another. However, the L3 contains both the L1 and L2 cache lines. 
 
Would adding a field to describe caches' clusivity relative to one another across multiple levels be viable? 
 
I don't have a solid plan on how to do this, but… I believe that the design should ideally aim for generalized clusivity. When an event (hit, miss, eviction) happens in a cache with particular clusivity, the cache reacts appropriately, and its requests/responses should enforce its clusivity. (This is where the foundation for the coherence model begins to creep in) We can generalize what each level should do based on:
1) The request it receives
2) the level's clusivity relative to the request's origin
The most challenging part of this approach is that the first level can slightly vary for certain clusivities (for example, if all levels are exclusive, only the L1 receives a fill from a demand request).
 
I'm unsure if I'm overthinking this…but I'll continue thinking about it.

ngober, 2023-06-29:
OK, I'll cherry-pick 0f35273c3c04189ca099a3987695c99a9eb06fa2..e3260fcfded0e327636f776ebecf4141751d6a18 onto a branch for #360, and I'll hard reset `feature/350-inclusivity` unless anyone was working on it.

I think we might be able to make the C++ stuff less tedious by adding fields to the cache member data. We could give `champsim::channel::request_type` an `clusivity` member with type `champsim::inclusivity'. If it is "exclusive", we request that the lower level bypass fill and invalidate on hit. If it is "inclusive", we request to be notified if the filled block is ever evicted. For this, we could give `CACHE::BLOCK` an `inclusive_evict` member of type `champsim::channel*` that reminds us to send invalidations to those channels on eviction.

### Two examples:

1. L1 and L2 are strict-inclusive, L3 is exclusive.
  - A miss in the L1 sends an inclusive request to the L2.
  - The miss in the L2 sends an exclusive request to the L3.
  - The L3 misses and bypasses fill.
  - Fill L2, mark that the L1 wishes to be invalidated on eviction.
  - Fill L1.

2. L1 and L2 are exclusive, L3 is non-inclusive.
  - A miss in the L1 sends an exclusive request to the L2.
  - The miss in the L2 sends a weak request to the L3.
  - The L3 misses and fills. No note is made that the L2 wants eviction notifications.
  - Bypass L2.
  - Fill L1.

So, I now do think that this can be accomplished by only knowing the clusivity of the lower level, and I think the clusivity relationships can be expressed entirely pairwise, without a global understanding. The request can encode the upper level's inclusive behavior.

ngober, 2023-06-29:
@all The linked branch for this issue has been force-updated. Please pull before doing any further work.

GinoAC, 2023-07-31:
I've been thinking on this more and I think the key here is that clusivity defines what a lower level does when it receives a request from an upper level or has an event occur. I'm hoping to get a working patch pushed in the next month or so and I'm going to use this issue as a scratchpad and to get feedback. Whatever sticks will be added to the wiki.

I believe we can just define clusivity per level since the only request generation dependent on clusivity are write backs and invalidations. To break down exactly what I mean:

Exclusive Caches
- Any hit in a lower cache from an upper request results in an invalidation in the lower level
- Demand misses only fill the first level cache
- Evictions in upper cache fill the lower cache
Inclusive Caches
- Lower cache evictions generates an invalidation to the upper levels (this requires tracking in the lower level)
- Misses in any level of the cache cause a fill
Noninclusive Caches
- Demand misses fill any level of the cache
- Hits do not invalidate

So the three main events are demand hits, demand misses, and evictions. 

| Clusivity/Event  | Hit | Miss | Eviction |
| ------------- | ------------- | ------------- | ------------- |
| Inclusive | Nothing | Fill | Writeback, send invalidation upper |
| Non-Inclusive | Nothing | Fill | Writeback |
| Exclusive | Invalidate | Nothing - fill first level | Writeback |

We don't need the other levels to be aware of their clusivity, just send the right request and let the state move the expected way (i.e. receiving an invalidation request results in an ack, regardless of clusivity). What's interesting about this is technically we never have to define the L1's clusivity. Hopefully this gets rid of the tedium of tying the caches' clusivities and just rely on the request handling to do the right thing.

I'm open to any thoughts on this, otherwise I'll be working on a draft when I have time in the coming weeks.

ngober, 2023-07-31:
You make a great point that it doesn't make much sense to define the L1 as exclusive, and the table demonstrates that: invalidating on hit and bypassing on miss basically mean it's just not a cache.

The table does assume that a cache has the uniform clusivity with respect to its upper levels. If it is, well, that's a design decision. If not, we end up with something like what I suggested above:

> 
> ```
> CACHE my_llc{CACHE::Builder{champsim::defaults::default_llc}
>     .upper_levels({std::pair{&to_cpu0_l2_channel, champsim::inclusivity::non_inclusive}, std::pair{&to_cpu1_l2_channel, champsim::inclusivity::non_inclusive}})
> };
> ```
> 

Do you plan to push your work to `feature/350-inclusivity`?

GinoAC, 2023-07-31:
I do plan to push to the 350-inclusivity branch unless there's other work being done on it.

Ah good point about the table assuming uniformity, I should have specified that the table assumes the clusivity is defined as the relationship between a lower cache and its upper cache level(s), regardless of where its placed in the hierarchy. An exclusive LLC would try to invalidate the line in a noninclusive or inclusive L2 and the L1. An inclusive LLC contains everything present in its upper caches, but the L2 is free to be exclusive by invalidating the L1's entries. So the clusivity of the level just defines what it does when it receives an event/request.

This does feel more like a design decision and I'd prefer to stick with existing conventions for the sake of any students using this feature.

ngober, 2023-07-31:
`feature/350-inclusivity` is good. Go ahead and use it.