ChampSim/ChampSim #371: Branch misprediction detection

alberto-ros, 2023-08-12:
Hi everyone! 

I was doing some test in the new master branch and I found some behavior that I wanted to discuss here. 

Imagine that a BTB miss takes place for an indirect or return branch. Since there is a BTB miss, the type of the instruction is not known, so the target that would be provided by the Indirect Target predictor or the RAS is neglected.

Later at decode, the branch is known, and in case of an indirect or return branch, the correct target could be used, thus squashing the pipeline at that time. However, ChampSim waits until execution to perform the squash, as it assumes that the miss was due to an incorrect target instead of a missing instruction type (but with a correct target).

So the questions is... is this relevant? It needs to be changed? I guess any modern processor would recover in that case when the branch is decoded.

I think that there is a "simple" way to implement this change without affecting much the BTB interface: returning both the indirect and RAS predicted targets, and handling that case when the prediction is done.

ngober, 2023-08-12:
I'm not completely sure that I understand which information is unknown. If there is an indirect/return miss, is that not because the target cannot be determined? So, the type and the target are both mispredicted. 

I do wonder if there is a minor feature that could be used here. For a while, ChampSim has had support for multiple modules to be specified. For BTBs, [the prediction is simply the last module's prediction](https://github.com/ChampSim/ChampSim/blob/73b62eb147f5006537c411dcb66e17a194bf786e/inc/ooo_cpu.h#L346) (note the use of the comma operator in the fold). Recently, #330 also [subdivided the BTB into smaller components](https://github.com/ChampSim/ChampSim/tree/develop/btb/basic_btb).

Would it make sense to make the BTB components (RAS, Indirect predictor, etc) into full-fledged BTB modules, then to modify the way the multiple module resolution works to keep more information? This is only a loose thought right now, but it could help solve what you're saying.

alberto-ros, 2023-08-13:
Right now, in the master branch, on a BTB miss, the Indirect and RAS predictors are not checked (https://github.com/ChampSim/ChampSim/blob/2bba2bd44e32fdeeb030ceded02a97ce9273ff91/btb/basic_btb/basic_btb.cc#L67).

The RAS should offer high accuracy, but right now on a BTB miss, return branches that miss in the BTB cause a misprediction that is recovered on the execution stage. 

I like the idea of splitting the BTB, the Indirect and the RAS predictors (not use about the functionality of direct in #330; I guess it is that providing the target on a BTB hit on a direct branch, but that is what the BTB should do, so in that case I would merge both). If the ooo_cpu code would them call all modules (BTB, Indirect and RAS) and take a decision there based on their outcome (which seems more realistic to me), then the problem can be easily solved.