ChampSim/ChampSim #382: Towards a new trace format

ngober, 2023-09-04:
The existing ChampSim trace format has a few issues that have limited development. As an initial example, the Cloudsuite traces and the standard traces have different number of destination registers, and these values are fixed, making the traces difficult to interoperate.

We ought to consider moving to a new trace format that removes the impedance of the old format and leaves the opportunity open for future growth. I'm opening this Issue to document the steps we will be taking towards making a new format. This might be a place where new requirements are added, and it might be a place where alternatives are considered.

### Desired Features

Here is a list of features that should be in the new format:

- In-built compression: the existing trace is emitted as raw values that must be immediately compressed by an external tool. The existing trace is typically too large to be stored on disk.
- File validation: the file should have a header and/or footer that validates the file contents. This can include a magic number, an instruction count, a checksum, etc.
- Source program information: the current trace format has no information about the underlying program. This information is usually encoded in the file name of the trace, but this is extremely fragile.
- Branch information in instructions: the current trace does not encode the branch type for any branch instruction or the branch target for non-taken branches. This makes branch prediction difficult.
- Memory access sizes: the size of a load, store, or a fetch is not encoded. If the access were to span multiple blocks, we would want to issue multiple memory requests.
- Well-defined format: The current format does not make strict correctness guarantees about the order in which bytes are stored on disk. If endianness or padding of struct elements varies between C++ implementations, two traces will be horribly broken between the two machines.
- Not using zero as a non-value: In both the registers and the memory accesses, the value 0 is used as an empty value. However, it may be that 0 should be a meaningful value, particularly in register values. 
- Support for multiple threads and thread interactions: the current trace format is strictly sequential and single-threaded. This makes support for multiple threads impossible since they cannot be sequenced.

### Deliverables

The deliverables on this issue are:

- A detailed file format specification, perhaps in the same format as an RFC
- A PIN tracer to replace the current tracer
- A conversion tool for the current trace format to a new format. This does not need to be an online tool (to stream the old format into ChampSim), though that could be an interesting alternative.

alberto-ros, 2023-09-04:
I would add:
 
 - The target of a branch should be present in the trace, even if the branch is not taken.

 - Optionally, more information about FP operations: e.g., a FP mul takes more cycles than a FP add and more than an integer operation.

ngober, 2023-09-04:
Ah, yes. Having opcodes in the trace would be very useful. I'll add that to the list.

aperais, 2023-09-05:
Hello @ngober, if I may, I have some thoughts on the new trace format based on some recent work I participated to.

One high level thought is that, if we look at what traces are used with ChampSim today, they are not all x86. In the future I half expect people to want to use RISC-V traces as well. If ChampSim owners want to take a stand and say only x86 support is provided, that is understandable, but given the large amount of ARMv8 traces that are available, I think there is something to be done here (there's actually cvp2champsim). More generally, my opinion is that any new trace format should make sure to not corner itself into supporting x86 only. Your proposal for branch information goes in that direction and I am 100% for it. 

Another issue is variable vs. fixed-width encoding. With fixed-width, you want to right shift PCs by 2 (4B instructions) before accessing BTB and branch predictor, otherwise you are losing 1/4 of the entries. With VLE, you should not shift. I think this could be addressed by a header featuring a shift amount that corresponds to the ISA.

Another issue is addressing mode. It is currently a pain point for ARMv8 because pre/post indexing increment has the load/store update the base register, and currently it is updated with the memory latency (not ALU, see #112). There are three ways to handle this case:

- Attaching an op-type or latency to destination registers, which feels like an overhaul because it means changing the code to deal with instructions on the scheduling side, but operands on the execution side.
- Adding a fake instruction in the trace to do the base register update. This is fairly transparent but i) What PC should it use ? ii) This changes the number of architectural instructions hence *PKI statistics.
- Supporting uops in the trace and in the simulator. Basically, each trace instruction may have several uops with op-type, num dests and dests, num srcs and srcs. All share the same PC. Fetch/Decode and the ROB only see one instruction, as a collection of uop. The backend can then see uops. The code changes would be to emit the uops in Decode and to change the Commit logic to wait until all uops are done for a given architectural instructions.

ngober, 2023-09-05:
I agree in the most part with these ideas.

A lot of what you describe with addressing modes is already possible. The base address register should be listed as a source register and, if it is updated either pre- or post-increment, it should be listed as a destination register also. This brings us to an interesting point that there is also currently no way to know if a particular register is an address source/destination or a data source/destination. These may have different implications for dependency resolution.

I think ChampSim should be (and, to an extent, is) ISA-agnostic. Right now, there are only a few assumptions about the ISA.

First, some of the included branch predictor and BTB models perform a right shift by 2 before their operations. Technically, this idealizes an ARMv8/RISC-V architecture over x86, since some x86 instructions may thus be aliased. I think this could be solved by including instruction length in the trace, similar to data access length described above. Overall, since modules are the responsibility of the user, I think this is not a problem the simulator should be required to solve. If a user is not downshifting PCs that they know to be multiples of 4, that's not a problem that ChampSim can help with.

Second, there are the `champsim::REG_STACK_POINTER` and similar constants. These have been there for a long time. They are mostly descriptive, showing the particular way traces represented the stack pointer, instruction pointer, and flags at a certain time. We could standardize these, making all traces regardless of ISA use these registers. Or, we could have these values be specified in a trace header. That way, each trace specifies what it uses for the stack pointer.

I had a discussion yesterday with someone about whether the translation from a particular ISA to ChampSim's agnostic internal representation should happen in the tracer or in the simulator. I'm inclined to say that the tracer should do this. I think the traces should be the same regardless of the source ISA. It eliminates the possibility of having two cores in ChampSim running different ISAs. Although, maybe people want that? :man_shrugging:

ngober, 2024-06-16:
I was looking though some old files today and I found a sketch of a trace format that I had made. I'm posting it here for posterity's sake.

```
enum branch_type
{
    NOT_BRANCH = 0,
    DIRECT_JUMP,
    INDIRECT_JUMP,
    CONDITIONAL,
    DIRECT_CALL,
    INDIRECT_CALL,
    RETURN,
    OTHER
};

enum access_length
{
    8BIT = 0,
    16BIT,
    32BIT,
    64BIT
};

struct trace_instr
{
    // Number of sources and destinations
    unsigned num_source_regs : 2;
    unsigned num_destination_regs : 2;
    unsigned num_source_mems : 2;
    unsigned num_destination_mems : 2;

    // Sources and destinations
    std::byte source_regs[num_source_regs];
    std::byte destination_regs[num_destination_regs];
    uint64_t source_mems[num_source_mems];
    uint64_t destination_mems[num_destination_mems];
    // maybe these should be fixed-length

    // Source and destination memory access lengths
    access_length source_al[4];
    access_length destination_al[4];

    // Branch information
    bool is_branch;
    bool is_taken_branch;
    branch_type type : 3;
    bool reserved[3];

    uint64_t ip;
    std::byte reserved;
    std::byte instruction_length;
    uint16_t opcode;
    uint16_t asid;
    uint16_t thread_id;

    // Optional: If the trace header has the data bit enabled
    std::byte data[]; // of length based on destination_al

    // Optional: if the trace header has debugging symbols bit enabled
    std::string debug;
};

struct trace_format
{
    char magic_number[4] = "CST2";
    std::string trace_name;
    bool has_data;
    bool has_debugging_symbols;
    bool reserved[6];
    std::byte padding[];

    std::byte stack_pointer_register_index;
    std::byte flags_register_index;
    std::byte instruction_pointer_register_index;
    std::byte reserved[5];

    uint32_t instruction_count;
    std::byte reserved[4];

    trace_instr instrs[instruction_count];
};
```

arav12ind, 2024-08-22:
Can DynamoRio traces be used for ChampSim?

ngober, 2024-08-22:
Not out of the box. I'm not aware of anyone who has written a tracereader for DR traces.

ngober, 2024-10-12:
Another feature for the new traces: traces should document their register aliasing. For example, in x86, rax and eax are aliases to the same hardware registers and the nonoverlapping bits are preserved. This information could be contained in a segment of the trace. Here's a sketch of what I'm envisioning:

    reg | aliased over | bitmap                  | zero on write?
    53  | 7            | 00 00 00 00 FF FF FF FF | 0
    54  | 8            | 00 00 00 00 FF FF FF FF | 0
    ....

maccoymerrell, 2024-11-05:
We had a good discussion about the trace format, and had the following ideas:

1. The header should be ascii, and describe certain things per line. No reason to use binary for the header. It might be good to dump some important information in the header that can be read as text. We think isa is probably good to include here. We weren't sure whether register aliasing should be in the header or not, or left to the tracer.
2. The header should describe the format of each instruction. Order of instruction fields, and the length of each field in bytes. The idea behind this is to support the old trace format, as well as any extensions to the trace format we may add in the future.
3. A binary portion of the file that describes each instruction as defined in the header. Each instruction begins with the # of bytes used to describe the instruction. Not all fields need described per instruction, but the fields must be described in-order and up to the last used field. Not sure for certain, but I think the ability to describe something like a noop as briefly as possible will help with trace size. Also, we need to eventually decompress the trace, so uncompressed size does matter for performance.
4. Regarding old traces, how are we going to reconcile instruction type between the two trace formats? Do we interpret all ops in old traces integer operations? Is it okay to leave generic execution latency in champsim?

@joshualmashburn probably has more to say

ngober, 2024-11-05:
1. I kinda agree. There's some things that should be ascii, like the trace name and the timestamp, but other things like the instruction count definitely should not be. I suspect y'all wouldn't dispute that, though.
2. This can be accomplished by a version number. Version 1 has XYZ fields and in that order, version 2 has ABC fields and in that order. I don't think the header should fully specify every possible combination.
3. I like the idea of possibly having multiple optional segments of the file, perhaps a segment having a map of instruction pointers to function names is one example. These optional segments need not be present, but can provide information if given. Your suggestion sounds like an example of that.
4. The old traces would be detectable by the lack of the magic number at the beginning. We can just slap a middling latency on every instruction and call it a day. If the user uses traces that lack information, we can't solve that.

ngober, 2024-12-06:
Adding another feature idea to the trace format: an initial state for page mapping. Multiple people have told me how the simulator is highly sensitive to memory request sequencing at the very beginning of the simulation (during warmup, typically), because that's when page mapping gets assigned. It can lead to measurable performance swings. An initial state would reduce this significantly. The mapping doesn't have to be "true", in the sense that it is the only mapping that could be. It can be a little arbitrary, it just needs to be there.