ChampSim/ChampSim #392: trace generator: assertion failed on generated trace

Pravesh-Jamgade, 2023-09-21:
**Describe the bug**
I am playing around generating my own traces for gapbs workloads. The generated traces do not work gives assertion error saying v_address!=0.

**To Reproduce**
1. $PIN_ROOT/pin -t obj-intel64/champsim_tracer.so -o v1.champsim  -s 100000 -t 100000000 -- ../../../Champsim2/tracer/pin/gapbs-exp/./cc -g 4

2. xz -z v1.champsim
3. bin/champsim --warmup_instructions 200000000 --simulation_instructions 500000000 tracer/pin/v1.champsim.xz

**Expected behavior**
address should be zero, thats correct. That means during only trace generation pintool was not able to pick the addresses.

**Output**
champsim: /mnt/A/Champsim3/src/channel.cc:117: bool champsim::channel::do_add_queue(R&, std::size_t, const typename R::value_type&) [with R = std::deque<champsim::channel::request>; std::size_t = long unsigned int; typename R::value_type = champsim::channel::request]: Assertion `packet.address != 0' failed.
Aborted (core dumped)

**Desktop (please complete the following information):**
 - OS: Linux Mint 20
 - Compiler [e.g. GCC, LLVM] gcc (Ubuntu 9.4.0-1ubuntu1~20.04.2) 9.4.0
 - Version [e.g. 22]

**Additional context**
I am trying to change trace format. I have observed these issue and asked in pinhead and stackoverflow as well. I have observed these issue when i was trying to update a certain variable which sometimes come with zero (expected was non-zero value). 
Please refer the issue i have posted on pinheads [#](https://groups.io/g/pinheads/message/13722) and stackoverflow [#](https://stackoverflow.com/questions/77133428/why-there-are-inconsistent-values-in-pin-tool-generated-trace?noredirect=1#comment135980575_77133428)

drcut, 2023-10-01:
I got the similar issue, and I found it was due to the '-t 100000000' is too large, larger than the real number of instructions. Thus, some dummy data was filled into the 'v1.champsim', and these dummy data cannot be parsed by ChampSim.
Hope this can help you solve your problem

drcut, 2023-10-02:
Hi, I came across the same issue. Do you have any update about it? Thanks

arunkp1986, 2023-10-02:
Hi, I also came across same bug for gapbs benchmark

arunkp1986, 2023-10-10:
I observed that trace files generated by the old tracer program have no issues with Gapbs.

Pravesh-Jamgade, 2023-10-10:
@drcut is it okay turn off. I mean it is a benchmark application. We expected it to run it as it is ??

Pravesh-Jamgade, 2023-10-10:
@arunkp1986 is it true, old tracer is working ?? can you share old repo ??

arunkp1986, 2023-10-11:
@Pravesh-Jamgade I checked for old forks from Champsim repo, tryout this https://github.com/Quangmire/ChampSim/tree/master/tracer, comment out "UINT32 read_size = INS_MemoryReadSize(ins);" and pass read_size as 0 as this PIN API is deprecated

rachanaaithal, 2024-04-09:
Hi, I am having the same issue when I add any new fields in the trace file. Were you guys able to solve it?

Pravesh-Jamgade, 2024-04-10:
@rachanaaithal Yes, please follow https://github.com/ChampSim/ChampSim/issues/392#issuecomment-1756570156

rachanaaithal, 2024-04-10:
@Pravesh-Jamgade , Thank you, I will try using the tracer in that fork, but were you able to figure out why this version of the tracer works as opposed to the current version?