ChampSim/ChampSim #404: [Clarification] : Type of trace file of extension .log

anubhavjana, 2023-11-08:
Hello, wanted to know if we can run trace files with .log extension i.e. pass .log file to champsim executable instead of .xz?

wilsonwang881, 2023-11-08:
Please refer to `src/tracereader.cc` line 38 to line 40. Looks like you need to have the files end with *gz*, *xz* or *bz2*.

You can change *log* extension to any of the one mentioned above but the file should be compressed in one of the above 3 formats.

Hope that helps.

anubhavjana, 2023-11-08:
Thanks @wilsonwang881 for your response. Had a look at the source file. Will try that out.

ngober, 2023-11-08:
The trace file does not need to be compressed. If you pass an uncompressed file, the simulator will read it just fine. We usually compress them because practically useful traces can often be quite large.

Note that the compression detection is very rudimentary: it makes an assumption based on the file name. So, if you are passing a zlib-compressed trace named `picture_of_cat.jpg`, the simulator will give you, well, unhelpful results. One of these days someone should get around to better compression detection (:point_right: #382).