ChampSim/ChampSim #414: Heap overflow

Setu-Gupta, 2023-12-15:
**Describe the bug**
I've implemented some cache replacement policies on a fork of this repository. You can find the fork that I'm working on [here](https://github.com/Setu-Gupta/ChampSim_SOTA). When I use the LFU, MRU, or the PseudoLRU policy, I get heap corruption issues. I found this out by enabling address sanitizer. This bug corrupts the objects that my policies creates which leads to assertion failures and segfaults.

**To Reproduce**
Steps to reproduce the behavior:
1. Clone my ChampSim repository and install the dependencies.
```
$ git clone https://github.com/Setu-Gupta/ChampSim_SOTA.git
$ cd ChampSim_SOTA
$ git submodule update --init
$ vcpkg/bootstrap-vcpkg.sh
$ vcpkg/vcpkg install
```
2. Checkout the `heap-overflow-bug` branch. The `Makefile` on this branch builds `ChampSim` with `-fsanitize=address -fno-omit-frame-pointer`.
```
$ git checkout remotes/origin/heap-overflow-bug
```

3. Configure and build ChampSim with the buggy configuration.
```
$ ./config.sh SOTA_configs/tests/replacement_policies/PseudoLRU_buggy.json
$ make -j
```
4. Run ChampSim with `400.perlbench-41B.champsimtrace.xz` trace which you can find [here](https://dpc3.compas.cs.stonybrook.edu/champsim-traces/speccpu/).
```
$ ./bin/champsim_test_PseudoLRU_debug --warmup_instructions 200000000 --simulation_instructions 500000000 ~/path/to/trace/400.perlbench-41B.champsimtrace.xz
```

**Expected behavior**
The heap buffer should not overflow.

**Output**
```
WARNING: physical memory size is smaller than virtual memory size.
WARNING: option --warmup_instructions is deprecated. Use --warmup-instructions instead.
WARNING: option --simulation_instructions is deprecated. Use --simulation-instructions instead.

*** ChampSim Multicore Out-of-Order Simulator ***
Warmup Instructions: 200000000
Simulation Instructions: 500000000
Number of CPUs: 1
Page size: 4096

Off-chip DRAM Size: 4 GiB Channels: 1 Width: 64-bit Data Race: 3200 MT/s
=================================================================
==713956==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x629000004618 at pc 0x563f9b2772b6 bp 0x7ffc02889e10 sp 0x7ffc02889e00
READ of size 8 at 0x629000004618 thread T0
    #0 0x563f9b2772b5 in CACHE::repl_replacementDpseudoLRU_update_replacement_state(unsigned int, unsigned int, unsigned int, unsigned long, unsigned long, unsigned long, unsigned int, unsigned char) (/home/grads/s/setu/ChampSim_SOTA/bin/champsim_test_PseudoLRU_debug+0x42f2b5)
    #1 0x563f9aea8805 in CACHE::handle_fill(CACHE::mshr_type const&) (/home/grads/s/setu/ChampSim_SOTA/bin/champsim_test_PseudoLRU_debug+0x60805)
    #2 0x563f9aeab68f in std::_Deque_iterator<CACHE::mshr_type, CACHE::mshr_type const&, CACHE::mshr_type const*> std::__find_if<std::_Deque_iterator<CACHE::mshr_type, CACHE::mshr_type const&, CACHE::mshr_type const*>, __gnu_cxx::__ops::_Iter_negate<CACHE::operate()::{lambda(auto:1 const&)#4}> >(std::_Deque_iterator<CACHE::mshr_type, CACHE::mshr_type const&, CACHE::mshr_type const*>, std::_Deque_iterator<CACHE::mshr_type, CACHE::mshr_type const&, CACHE::mshr_type const*>, __gnu_cxx::__ops::_Iter_negate<CACHE::operate()::{lambda(auto:1 const&)#4}>, std::random_access_iterator_tag) [clone .constprop.0] (/home/grads/s/setu/ChampSim_SOTA/bin/champsim_test_PseudoLRU_debug+0x6368f)
    #3 0x563f9aee15dc in CACHE::operate() (/home/grads/s/setu/ChampSim_SOTA/bin/champsim_test_PseudoLRU_debug+0x995dc)
    #4 0x563f9af17675 in champsim::do_phase(champsim::phase_info, champsim::environment&, std::vector<champsim::tracereader, std::allocator<champsim::tracereader> >&) (/home/grads/s/setu/ChampSim_SOTA/bin/champsim_test_PseudoLRU_debug+0xcf675)
    #5 0x563f9ae730ea in champsim::main(champsim::environment&, std::vector<champsim::phase_info, std::allocator<champsim::phase_info> >&, std::vector<champsim::tracereader, std::allocator<champsim::tracereader> >&) (/home/grads/s/setu/ChampSim_SOTA/bin/champsim_test_PseudoLRU_debug+0x2b0ea)
    #6 0x563f9ae7f746 in main (/home/grads/s/setu/ChampSim_SOTA/bin/champsim_test_PseudoLRU_debug+0x37746)
    #7 0x7f829ee29d8f in __libc_start_call_main ../sysdeps/nptl/libc_start_call_main.h:58
    #8 0x7f829ee29e3f in __libc_start_main_impl ../csu/libc-start.c:392
    #9 0x563f9ae81e64 in _start (/home/grads/s/setu/ChampSim_SOTA/bin/champsim_test_PseudoLRU_debug+0x39e64)

0x629000004618 is located 1040 bytes to the right of 16392-byte region [0x629000000200,0x629000004208)
allocated by thread T0 here:
    #0 0x7f829f6b6357 in operator new[](unsigned long) ../../../../src/libsanitizer/asan/asan_new_delete.cpp:102
    #1 0x563f9b276b6d in CACHE::repl_replacementDpseudoLRU_initialize_replacement() (/home/grads/s/setu/ChampSim_SOTA/bin/champsim_test_PseudoLRU_debug+0x42eb6d)
    #2 0x563f9ae71b04 in champsim::main(champsim::environment&, std::vector<champsim::phase_info, std::allocator<champsim::phase_info> >&, std::vector<champsim::tracereader, std::allocator<champsim::tracereader> >&) (/home/grads/s/setu/ChampSim_SOTA/bin/champsim_test_PseudoLRU_debug+0x29b04)
    #3 0x563f9ae7f746 in main (/home/grads/s/setu/ChampSim_SOTA/bin/champsim_test_PseudoLRU_debug+0x37746)
    #4 0x7f829ee29d8f in __libc_start_call_main ../sysdeps/nptl/libc_start_call_main.h:58

SUMMARY: AddressSanitizer: heap-buffer-overflow (/home/grads/s/setu/ChampSim_SOTA/bin/champsim_test_PseudoLRU_debug+0x42f2b5) in CACHE::repl_replacementDpseudoLRU_update_replacement_state(unsigned int, unsigned int, unsigned int, unsigned long, unsigned long, unsigned long, unsigned int, unsigned char)
Shadow bytes around the buggy address:
  0x0c527fff8870: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c527fff8880: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c527fff8890: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c527fff88a0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c527fff88b0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
=>0x0c527fff88c0: fa fa fa[fa]fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c527fff88d0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c527fff88e0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c527fff88f0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c527fff8900: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c527fff8910: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==713956==ABORTING
```

**Desktop:**
 - OS: Ubuntu 22.04.3 LTS
 - Compiler: gcc version 11.4.0
 - LibC version: Ubuntu GLIBC 2.35-0ubuntu3.5

**Additional context**
The only difference between this repository and my repository is that my repository has more replacement policies. Moreover, the `heap-overflow-bug` branch on my repository builds ChampSim with address sanitizer. Apart from these two differences, both the repositories are completely identical.

Setu-Gupta, 2023-12-20:
Note that this bug only occurs when I do manual memory management (by using `new` and `delete`). If I don't allocate memory manually, then this bug disappears.

ngober, 2024-01-02:
If your bug occurs only when you are doing manual memory management, are you confident this is a bug with ChampSim and not with your module code?

(Also, stop using manual memory management. You probably don't need it. Encapsulate it in a constructor and destructor.)

Setu-Gupta, 2024-01-02:
I've re-implemented my code such that it no longer uses manual memory management. However, considering that manual memory management should technically not have resulted in heap overflow, I'm now very skeptical of the data that ChampSim generates. If heap overflow is possible, then it is also possible, albeit unlikely, that the statistics may have been overwritten. I want to find the source of this bug.

I'm fairly confident that the bug is not with my implementation as I've added various `assert`s in the code to ensure that I never access memory which is not allocated. But it is also possible that I may have messed something up somewhere.

ngober, 2024-01-02:
I'm not too worried about what you report. Both the illegal access and the allocation occurred within your module. ChampSim can't provide infinite protection against accessing bad data. I think you should use `std::vector` and its bounds-checked `.at()` function.

ngober, 2024-03-01:
Closing this unless we can confirm that the bug is internal to ChampSim.

closed 2024-03-01 by ngober