ChampSim/ChampSim #424: Runtime configuration

ngober, 2024-01-04:
Seems like this would still be tedious with Champsim's current need to recompile after altering the config file. I saw that simulators like Ramulator2 built their entire system based on the config file passed at runtime rather than hardcoding and selective compiling. The con of course is that you need to compile in all prefetching schemes, replacement policies, etc....

I have thought about something like this a lot when having to make 6 or 7 directories with different configurations of champsim. It feels wrong to just have a directory full of nothing but different champsim binaries with different configurations.

_Originally posted by @maccoymerrell in https://github.com/ChampSim/ChampSim/discussions/415#discussioncomment-7891215_

In response to this comment, I made some efforts to move the configuration to the run step, rather than the compile step. Progress can be tracked on the `feature/runtime-config` branch.

There are a few challenges that need to be overcome:

 - All constexpr references to `BLOCK_SIZE` or `PAGE_SIZE` need to be converted to object-local values. We can't have static sizes with a runtime configuration. We probably also need to be careful here to make sure that upper and lower levels either have the same block size or convert to their lower level's block size when sending a packet.
 - The DRAM model needs to take its parameters in its constructor
 - The build step needs a system that converts module names as strings to C++ object identifiers. This probably has a few surprises behind it.
 - A C++ replica of the Python configuration parser needs to be created. This will take a lot of work.
 - Generation of the files that support legacy modules must be maintained.

I have already succeeded on the last point. The makefile now generates the support files as necessary when building ChampSim.

The branch currently will probably not build a proper executable, because the instantiation file is no longer generated. The environment needs to be created by the C++ side instead. But, all the tests compile and pass so there is some good things happening.

I would appreciate any help on this. I worked on the first part out of sheer obsession today, but I probably can't afford to work on it much for a while.

ngober, 2024-01-10:
This branch has significant merge conflicts after the merge of #297. The `champsim::block_number` and similar types can no longer be statically defined, since `BLOCK_SIZE` is known only at runtime.

ngober, 2024-02-06:
The merge conflicts have been resolved. I developed a pattern where the runtime `BLOCK_SIZE` can be examined:

    struct block_offset_extent : champsim::dynamic_extent {
        block_offset_extent() : champsim::dynamic_extent(champsim::data::bits{champsim::lg2(BLOCK_SIZE)}, champsim::data::bits{}) {}
    };

This way, the extent is both default constructible, so it can be used like `champsim::address_slice<champsim::block_offset_extent> offset{address};`, and still examine the value at run time. However, this still requires that the value be global, which I would eventually like to change.

maccoymerrell, 2024-09-16:
When I have time, I would like to take a crack at part of this.
I think by using derived classes (for the modules), and a json parsing library, reading in the config becomes a lot simpler.
There is an interesting thing Ramulator does with maps and constructors that allows different derived classes of a base class to be looked up via string, and then constructed and attached to a base class pointer. Very useful for swapping modules at runtime or construction.

Ideally, I don't think it would be a bad idea to enable runtime configuration for all aspects of ChampSim, for example, the dram controller. I don't see an issue with supporting this, and I think it would be interesting to move ChampSim towards an entirely module-based system. I think if its designed well, this isn't something that needs to be complicated. I think the current python config scheme is more limiting than we think.

Think about this: Instead of dealing with all these complex constructors, we instead pass it a portion of the json file, along with a parent pointer.  The portion of the json file passed is the part relevant towards creating that module. Then, the implementation of that specific module decides which parameters it wants in that portion of the config. It can look for any variable it wants, and use it however it wants. Now imagine modules being able to recursively do this, so we start at the environment then cache then replacement. Since we don't have a hard-coded config format, new modules can be parameterized at runtime according to what the writers of that module want.
New cache model? Write your own, no need to modify ChampSim's base source code and you can use the existing replacement policies and prefetchers. 
I think these types of changes can really leverage ChampSim's module system to become something competitive with gem5 in its own way.

Regarding legacy, I honestly think migrating code from legacy to the new module system is not difficult (I did spp in about an hour). If someone is complaining about not being able to use a piece of code written 5 years ago, then maybe we should just call the next release ChampSim2 and be done with it.

This is all stuff that definitely needs to come after what I am working on though, regardless.

ngober, 2024-09-16:
I don't disagree with your legacy comment, but I suspect certain people might. I'll leave it at that.

It looks like ramulator is doing some kind of pimpl for their implementation, which probably ends up being just the same as virtual inheritance, but with some flexibility. I've been a fan of type erasure for the modules because it allows us to do compile-time feature detection, but I suppose that can change.

Regarding passing the config down recursively, I want to think through that. Currently there are 3 places that one could specify the L1D sets:

    {
        "L1D": { "name": "my_l1d", "sets": 256 },
        "ooo_cpu": [ { "L1D": { "name": "my_l1d", "sets": 512 } } ],
        "caches": [ { "name": "my_l1d", "sets": 1024 } ]
    }

(Exercise for the reader: how many sets are in the L1D in this configuration?)

How would these things get passed recursively down? How do you handle the loop where the L1D has a TLB that descends to a PTW backed by the same L1D?

maccoymerrell, 2024-09-16:
FYI when I say "module" here I am going to be referring to basically any ChampSim building block, operable, module, etc... not just replacement/bp/pref. Also I started writing this to respond to you, and then it kind of veered into things I want to change about how the config file works, and then a complete overall of the config file.

Well first, I think its clear the PTW and all the caches are at the same level of module hierarchy, not a member of each other, so recursive passing wouldn't (and probably couldn't) be used in that case. This will be restricted to modules which have modules within them (not chasing down channel connections). So a custom cache that uses a prefetcher will receive the part of the json that defines the parameters for that cache and for the prefetcher inside it, and pass just the prefetcher portion to the prefetcher.
I think it might also be feasible to create a wrapper around the entire config file and pass that instead, so that if a module wants a parameter from a different module, its still available, it just needs to know what its the module's name is. Otherwise it can look up a parameter with its own name. Ideally, I think we should move towards a place where all modules are treated the same in the config file, so we don't need to make special considerations for where and how things like caches can be defined.

I guess what you are worried about is actually finding and passing the correct parameters that may be in multiple locations, and how recursive passing might work with that.
I don't like the idea of supporting multiple ways to define a single parameter. I think the config file should imply a hierarchy in terms of modules, so if something is defined within a module, it is a child of that module and is instantiated within it. Things defined at the same level are siblings in whatever parent module they are defined in. Modules are only defined once, with a complete definition. Essentially, all the parameters needed for the constructor are defined in a single place or default values are used. To avoid just dropping support outright for old config files, I think this can be managed with some pre-processing.

Connecting modules at runtime is another issue. I like the idea of being able to pass a module's name as a parameter in another module and being able to use that as a reference to that module. I think that gives a lot of flexibility to the config file, and modules users might write. There are a couple of ways that could be supported, but I think its a good idea for supporting things like VMEM as a module. I think channel connections can be handled the same way, where references to channels are passed. I think this would require channels be declared explicitly in the config though.

So you can define them like this in the config (borrowing some of Ramulator's yaml formatting here):
channel:
  name: "L1D_to_L2C"
  queue_size, etc...
channel:
  name: "L1I_to_L2C"
  queue_size, etc...
cache:
  name: "L1D"
  sets, etc...
  downstreams: {"L1D_to_L2C"}
cache:
  name: "L2C"
  sets, etc...
  upstreams: {"L1D_to_L2C", "L1I_to_L2C"}

Alternatively, we do what the config file currently does and instead pass module names, then replace those references with channels on the backend and do the connecting. I think that is probably doable as well, but makes config parsing harder.

I actually kind of like the idea of channels being modules. I think that might allow for some pretty interesting designs. I also like that we wouldn't implicitly define a channel between some types of modules (caches) and not others (vmem and dram). If you pass a reference to one module to another, that should give you a reference to the module, not a reference to a channel between the modules. 

For things like VMEM, I think either it will need pulled into the PTW, or we will need a similar reference-passing system that can be applied to modules directly. So if a module wants a reference to another module passed as a parameter, ChampSim can do it. That's a lot more difficult to do, especially because ChampSim currently uses constructors exclusively for setting up operables.  To do this the easiest way, we would need to allow modules to be instantiated with no constructor, and then passed parameters later on. Or they can be passed a reference that doesn't get set with a real value until after all modules are constructed. I'm sure there's a fancy standard library container or type that does what I'm describing.

ngober, 2024-09-16:
I think a lot of what you're saying makes good sense, and I'm following what you're thinking about. I want to pull us back, however, and consider the minimal changes that need to be made to eliminate the Python-based configuration step. This is not for the sake of not wanting a future vision for where ChampSim can develop, but for the sake of wanting to get there without rewriting the entire thing.

I think the terminology of "module" has been settled into the discussion and documentation of ChampSim for long enough now that if we are to broaden the definition to include caches/ptws/vmem/etc., then we need a different word to describe it. Perhaps "component" would work. I like the idea of envisioning how components compose, and externally imposing the connection network (of channels) between them. The idea of passing configuration down so that each component knows how its subcomponents need to be initialized makes a lot of sense. I think a wrapper type is the best design here, so that we can present an interface to the components independent of the user-facing (JSON, YAML, TOML...) format.

But, even discussions about changing the format seem premature to me. It's not clear that they must be changed. In the Python parser right now, the different sources of information (see my example above) are composed into a single dict that gets transformed (mostly directly) into a C++ builder class. To me, that strategy seems like the best way to proceed immediately. Then, we can discuss, perhaps in process, whether the existing format is too difficult to work with. There are currently multiple ways to pass in a parameter. We might not like that it is, but at present, it is and we should seek to maintain it because it's user-facing. The builder classes are quite friendly to this, though. We can instantiate builders with the root config, then call member functions on those builders with the caches defined in `"ooo_cpu": [ { ... } ]` (which overwrites), then call the member functions again with the caches defined in `"caches": [ ... ]`. This feels like it will work.

My point is this: let's limit the scope for this change and view config file stuff as part of a larger project.