ChampSim/ChampSim #440: Mismatch between `champsim_tracer` and `input_instr` regiter size

0x161e-swei, 2024-02-25:
In `tracer/pin/champsim_tracer.cpp` line [128](https://github.com/ChampSim/ChampSim/blob/master/tracer/pin/champsim_tracer.cpp#L128) and line [136](https://github.com/ChampSim/ChampSim/blob/master/tracer/pin/champsim_tracer.cpp#L136), the `regNum` variable is sent as UINT32
However, the `input_instr` struct defined in `inc/trace_instruction.h` uses `unsigned char` array for the `source_registers` and `destination_registers` members.

Intel [PIN manual](https://software.intel.com/sites/landingpage/pintool/docs/98547/Pin/html/group__REG.html) (for the version referenced in this repo), seems to indicate that there are more than 256 possible values for the registers enum.

Does this cause a problem?

ngober, 2024-02-26:
I'm not sure whether it does or not. A significant number of the upper registers listed are PIN-specific virtualized registers. I don't have any observations that would tell us whether these never occur, but I suspect they might not. Either way, I think it could be a good step for ChampSim to standardize the way that we represent registers in traces, as the discussion for opcodes in #382.

joshualmashburn, 2024-10-10:
It does not. See the Pin header file $PIN_ROOT/source/include/pin/gen/reg_ia32.PH

In the x86_64 builds of Pin, the REG enum value for the last machine register is 243. There are two more "application registers", then 246 and beyond are Pin registers, which are never referenced in instrumented instructions.
(update: sometime between Pin 3.22 and the latest Pin, they added one more hardware register to the enum)

If you build the 32-bit version of Pin, there are at least 40-50 fewer machine registers, and the REG enum values shift accordingly (see the if defined clauses in the aforementioned header file).

Does this mean if Intel introduces yet another set of hardware registers (as opposed to extending existing ones), Pin will use more than 256 identifiers for machine registers? Most likely. However, this isn't an issue for existing traces or traces yet to be created on the latest machines.

ngober, 2024-10-12:
It looks like this is in fact a type mismatch, but it isn't the cause of any bad behavior. I'm leaving this open so that we can fix it eventually, but marking it low-priority. If anyone wants to patch this up, go ahead, otherwise we'll roll it into another tracer change.

ngober, 2024-11-08:
I'm realizing that this is a duplicate of #301, so I'm closing this issue. I reiterate that this is not the cause of any bugs that we are aware of, but should probably still be patched.

closed 2024-11-08 by ngober