ChampSim/ChampSim #442: btb `return_stack::calibrate_call_size` for syscalls/sysret

0x161e-swei, 2024-02-27:
**Describe the bug**
In function `return_stack::calibrate_call_size` ([btb/basic_btb/return_stack.cc#L29](https://github.com/ChampSim/ChampSim/blob/feature/350-inclusivity/btb/basic_btb/return_stack.cc#L29)), the distance between `call_ip` and `branch_target` was computed using `champsim::offset`. However, for instruction traces with mode switches, the call distance can be pretty large.

I got `call_ip=0xffff000008e9f8f8` and `branch_target=0x4003b8` and `champsim::offset` throws an exception as follows:
```
terminate called after throwing an instance of 'std::overflow_error'
  what():  The offset cannot be represented in the difference type. Consider using champsim::uoffset() instead.
```

I am using trace from ipc1_public/server_014.champsimtrace.xz .

ngober, 2024-02-27:
Do you know why this occurs? Is it for syscalls?

0x161e-swei, 2024-02-27:
I do not have clear answers for ipc1_public traces. But this happened on my own Intel PT traces.
The `call_ip` was pointing to an instruction that is near syscall/sysret but not exactly on a syscall instruction.

0x161e-swei, 2024-02-27:
This is happening in many ipc1_public traces.   
All exceptions are caused by computing offet between a seemingly kernel address and a user address.   
Some logs attached:
```
$ ./bin/champsim -w 100000000000 workload_traces/ipc1_public/server_002.champsimtrace.xz                                                               
WARNING: physical memory size is smaller than virtual memory size.                                                                                                                                                                                            
                                                                                                                                                                                                                                                              
*** ChampSim Multicore Out-of-Order Simulator ***                                                                                                                                                                                                             
Warmup Instructions: 100000000000                                                                                                                                                                                                                             
Simulation Instructions: 9223372036854775807                                                                                                                                                                                                                  
Number of CPUs: 1                                                                                                              
Page size: 4096                                                
                                                               
Off-chip DRAM Size: 4 GiB Channels: 1 Width: 64-bit Data Rate: 3205 MT/s
terminate called after throwing an instance of 'std::overflow_error'
  what():  The offset between 0xffff000008e9f8f8 and 0x4003b0 cannot be represented in the difference type. Consider using champsim::uoffset() instead.
Aborted       

                                                 
$ ./bin/champsim -w 100000000000 workload_traces/ipc1_public/server_003.champsimtrace.xz                                                           
WARNING: physical memory size is smaller than virtual memory size.  
                                                                                                                                                                                                                                                              
*** ChampSim Multicore Out-of-Order Simulator ***
Warmup Instructions: 100000000000           
Simulation Instructions: 9223372036854775807
Number of CPUs: 1                                              
Page size: 4096                                                

Off-chip DRAM Size: 4 GiB Channels: 1 Width: 64-bit Data Rate: 3205 MT/s
terminate called after throwing an instance of 'std::overflow_error'
  what():  The offset between 0xffffb7e24708 and 0xffff00000812eb70 cannot be represented in the difference type. Consider using champsim::uoffset() instead.
Aborted        

                           
$ ./bin/champsim -w 100000000000 workload_traces/ipc1_public/server_003.champsimtrace.xz 
WARNING: physical memory size is smaller than virtual memory size.

*** ChampSim Multicore Out-of-Order Simulator ***
Warmup Instructions: 100000000000
Simulation Instructions: 9223372036854775807
Number of CPUs: 1
Page size: 4096

Off-chip DRAM Size: 4 GiB Channels: 1 Width: 64-bit Data Rate: 3205 MT/s
terminate called after throwing an instance of 'std::overflow_error'
  what():  The offset between 0xffffb7e24708 and 0xffff00000812eb70 cannot be represented in the difference type. Consider using champsim::uoffset() instead.
Aborted

$ ./bin/champsim -w 100000000000 workload_traces/ipc1_public/server_004.champsimtrace.xz 
WARNING: physical memory size is smaller than virtual memory size.

*** ChampSim Multicore Out-of-Order Simulator ***
Warmup Instructions: 100000000000
Simulation Instructions: 9223372036854775807
Number of CPUs: 1
Page size: 4096

Off-chip DRAM Size: 4 GiB Channels: 1 Width: 64-bit Data Rate: 3205 MT/s
terminate called after throwing an instance of 'std::overflow_error'
  what():  The offset between 0xffff000008222e08 and 0xffffb32fec98 cannot be represented in the difference type. Consider using champsim::uoffset() instead.
Aborted
```

ngober, 2024-03-01:
This is interesting. I've dispatched someone on my team to look at it. This may be something we need to address in the trace format. The hardware should reasonably assume that calls return to the same place they left from, and if the trace is inaccurately representing syscalls, we should fix that.

alberto-ros, 2024-03-01:
I do not know if this is the reason, but we found out that the original IPC1 traces were not well trasnlated regarding calls/returns (please see "Rebasing Microarchitectural Research with Industry Traces", IISWC 2023.) Do you know if the same problem occurs for the new traces?

0x161e-swei, 2024-03-03:
Thank you for looking into this!
Hi @alberto-ros , I do not have access to the new traces, are you referring to traces from your paper artifact?  

And hi @ngober, for my PT traces (and I think this could be happening in the CVP/IPC1 traces too), one problem I found is around the interrupt handlers. The `iret` instructions are correctly identified as returns as they read and write rsp and write rip. However, the entry point to the handler can really just be any instructions when an OS tick happens or a memory instruction when a page fault happens. 
Regarding the new trace format discussed in #382, perhaps a note about interrupts should be added if full-system traces are of interest.

The instructions that throw this mismatch exceptions in simulation can be really misleading. They are around syscalls and sysret because the exceptions are only triggered if the offset is really large (which usually happens when the call-return mismatch happen to be between a kernel and a user instructions).

A second problem I identified from my PT traces (and this one is x86/64 only) is the return stack stuffing mitigation (cref. [source code](https://elixir.bootlin.com/linux/latest/source/arch/x86/entry/entry_64.S#L203)) employed in Linux kernels to defend transient execution attacks on the x86 return stack buffer. These call instructions are meant to be unbalanced. A similar issue (but I did not see from my traces) could potentially exist in the Retpoline mitigation against branch target injection, where [Linux kernel replaced indirect calls/jmps](https://elixir.bootlin.com/linux/latest/source/arch/x86/include/asm/nospec-branch.h#L243) with return instructions.

alberto-ros, 2024-03-04:
Yes, by new traces I mean the ones generated with the artifact. In the artifact repository you can find the IPC-1 traces already converted after our paper findings.

ngober, 2024-11-14:
I'm closing this because it is technically resolved by #543. However, we should pay attention to this in the future when we're making new traces. The fact that this and the other error occurred means that something is not quite right in the tracer code.

closed 2024-11-14 by ngober