ChampSim/ChampSim #447: How to have a shared LLC with multi-core?

ShefaaAbed, 2024-03-02:
I want the LLC to be shared and L1D and L2C to be private. 

This is my config file for two cores.
[my_config.json](https://github.com/ChampSim/ChampSim/files/14469898/my_config.json)

ngober, 2024-03-04:
Have you read the documentation at https://champsim.github.io/ChampSim/master/Creating-a-configuration-file.html ?

The default configuration has a shared LLC and private L1 and L2 caches.

ShefaaAbed, 2024-03-04:
Yes, I did that, but what is confusing me is that I used 401 in the LRU replacement policy, and got LLC results for each CPU, you can check the config file and result file below.
Thank you.

The config file: 
[my_config.json](https://github.com/ChampSim/ChampSim/files/14485048/my_config.json)

The result file:
[401.bzip2-7B.champsimtrace.xz.txt](https://github.com/ChampSim/ChampSim/files/14485050/401.bzip2-7B.champsimtrace.xz.txt)

ngober, 2024-03-04:
Yes, the statistics are reported separately, but the LLC is still unified.

ShefaaAbed, 2024-03-04:
Ok thank you.

ShefaaAbed, 2024-03-04:
> Yes, the statistics are reported separately, but the LLC is still unified.

Why are the LLC results divided among each CPU? Does this mean the access result shown by the LLC is specific to each CPU, so the total would be the sum of these?

ngober, 2024-03-04:
The results are divided because it is often useful to know the effect that one core or another has on the shared cache. It is a simple thing to add the results to get the total, but would be a much harder thing to divide them if they were reported together.

closed 2024-03-19 by ngober