ChampSim/ChampSim #454: Multi-core Simulations Silent Deadlock

maccoymerrell, 2024-03-23:
When running MC simulations on master, long-running simulations see some cores stall indefinitely. Some cores remain active, thus Champsim doesn't seem to detect any deadlock.

**To Reproduce**
Steps to reproduce the behavior:
1. Create an 8-core champsim config, default values except 16MB LLC
2. Run 605.mcf_s-1536B.champsimtrace.xz

**Expected behavior**
Cores eventually make progress on instruction stream regardless of traffic on other cores.

**Output**
All cores report initial heartbeats, but several cores will eventually drop-out of the run, stalling indefinitely. It appears highly correlated with core number, so core 0 will never stall but core 7 will always stall.

**Desktop (please complete the following information):**
 - OS: Ubuntu

**Additional context**
These runs were utilizing Ramulator as the dram backend. I am doubtful that could cause this type of stalling, since presumably a stall somewhere in the dram would impact all cores.

Multicore runs take time, so I will look into this issue further when possible.

chaseblock, 2024-09-04:
I ran into this the other day. Assuming it's the same issue, this is caused by a contentious resource prioritizing lower-numbered cores over higher-numbered cores.

Running a gcc trace (8 cores, 8-way LLC, defaults otherwise) for a short amount of instructions provides the following, where we see that the simulation does not deadlock, but some cores see a much lower IPC:

```
⇒ bin/champsim -w 0 -i 1000000 403.gcc-16B.champsimtrace.xz 403.gcc-16B.champsimtrace.xz 403.gcc-16B.champsimtrace.xz 403.gcc-16B.champsimtrace.xz 403.gcc-16B.champsimtrace.xz 403.gcc-16B.champsimtrace.xz 403.gcc-16B.champsimtrace.xz 403.gcc-16B.champsimtrace.xz |& tee sim.out                             
WARNING: physical memory size is smaller than virtual memory size.

*** ChampSim Multicore Out-of-Order Simulator ***
Warmup Instructions: 0
Simulation Instructions: 1000000
Number of CPUs: 8
Page size: 4096

Off-chip DRAM Size: 64 MiB Channels: 1 Width: 64-bit Data Rate: 3205 MT/s
Warmup finished CPU 7 instructions: 0 cycles: 1 cumulative IPC: 0 (Simulation time: 00 hr 00 min 00 sec)
Warmup finished CPU 6 instructions: 0 cycles: 1 cumulative IPC: 0 (Simulation time: 00 hr 00 min 00 sec)
Warmup finished CPU 5 instructions: 0 cycles: 1 cumulative IPC: 0 (Simulation time: 00 hr 00 min 00 sec)
Warmup finished CPU 4 instructions: 0 cycles: 1 cumulative IPC: 0 (Simulation time: 00 hr 00 min 00 sec)
Warmup finished CPU 3 instructions: 0 cycles: 1 cumulative IPC: 0 (Simulation time: 00 hr 00 min 00 sec)
Warmup finished CPU 2 instructions: 0 cycles: 1 cumulative IPC: 0 (Simulation time: 00 hr 00 min 00 sec)
Warmup finished CPU 1 instructions: 0 cycles: 1 cumulative IPC: 0 (Simulation time: 00 hr 00 min 00 sec)
Warmup finished CPU 0 instructions: 0 cycles: 1 cumulative IPC: 0 (Simulation time: 00 hr 00 min 00 sec)
Warmup complete CPU 7 instructions: 0 cycles: 1 cumulative IPC: 0 (Simulation time: 00 hr 00 min 00 sec)
Warmup complete CPU 6 instructions: 0 cycles: 1 cumulative IPC: 0 (Simulation time: 00 hr 00 min 00 sec)
Warmup complete CPU 5 instructions: 0 cycles: 1 cumulative IPC: 0 (Simulation time: 00 hr 00 min 00 sec)
Warmup complete CPU 4 instructions: 0 cycles: 1 cumulative IPC: 0 (Simulation time: 00 hr 00 min 00 sec)
Warmup complete CPU 3 instructions: 0 cycles: 1 cumulative IPC: 0 (Simulation time: 00 hr 00 min 00 sec)
Warmup complete CPU 2 instructions: 0 cycles: 1 cumulative IPC: 0 (Simulation time: 00 hr 00 min 00 sec)
Warmup complete CPU 1 instructions: 0 cycles: 1 cumulative IPC: 0 (Simulation time: 00 hr 00 min 00 sec)
Warmup complete CPU 0 instructions: 0 cycles: 1 cumulative IPC: 0 (Simulation time: 00 hr 00 min 00 sec)
Simulation finished CPU 0 instructions: 1000000 cycles: 8278726 cumulative IPC: 0.1208 (Simulation time: 00 hr 02 min 09 sec)
Simulation finished CPU 1 instructions: 1000000 cycles: 8825715 cumulative IPC: 0.1133 (Simulation time: 00 hr 02 min 18 sec)
Simulation finished CPU 4 instructions: 1000000 cycles: 9078507 cumulative IPC: 0.1102 (Simulation time: 00 hr 02 min 22 sec)
Simulation finished CPU 2 instructions: 1000000 cycles: 9232957 cumulative IPC: 0.1083 (Simulation time: 00 hr 02 min 25 sec)
Simulation finished CPU 3 instructions: 1000000 cycles: 9506841 cumulative IPC: 0.1052 (Simulation time: 00 hr 02 min 29 sec)
Simulation finished CPU 5 instructions: 1000000 cycles: 22855797 cumulative IPC: 0.04375 (Simulation time: 00 hr 06 min 02 sec)
Simulation finished CPU 6 instructions: 1000000 cycles: 42553363 cumulative IPC: 0.0235 (Simulation time: 00 hr 11 min 19 sec)
Simulation finished CPU 7 instructions: 1000000 cycles: 56068658 cumulative IPC: 0.01784 (Simulation time: 00 hr 14 min 59 sec)
Simulation complete CPU 7 instructions: 1000000 cycles: 56068658 cumulative IPC: 0.01784 (Simulation time: 00 hr 14 min 59 sec)
Simulation complete CPU 6 instructions: 1512395 cycles: 56068658 cumulative IPC: 0.02697 (Simulation time: 00 hr 14 min 59 sec)
Simulation complete CPU 5 instructions: 2949202 cycles: 56068658 cumulative IPC: 0.0526 (Simulation time: 00 hr 14 min 59 sec)
Simulation complete CPU 4 instructions: 5428515 cycles: 56068658 cumulative IPC: 0.09682 (Simulation time: 00 hr 14 min 59 sec)
Simulation complete CPU 3 instructions: 5431588 cycles: 56068658 cumulative IPC: 0.09687 (Simulation time: 00 hr 14 min 59 sec)
Simulation complete CPU 2 instructions: 5431588 cycles: 56068658 cumulative IPC: 0.09687 (Simulation time: 00 hr 14 min 59 sec)
Simulation complete CPU 1 instructions: 5477111 cycles: 56068658 cumulative IPC: 0.09769 (Simulation time: 00 hr 14 min 59 sec)
Simulation complete CPU 0 instructions: 5907687 cycles: 56068658 cumulative IPC: 0.1054 (Simulation time: 00 hr 14 min 59 sec)

ChampSim completed all CPUs

```

In particular, I believe that this is caused by the tag check limit, which first serves upstream 0, then 1, ... until the maximum number of tags in a cache have been checked, meaning that 7 rarely gets any bandwidth.

ngober, 2024-09-04:
OK, this is interesting and gets me thinking. Is the problem that the bandwidth is less than the number of upstreams, so the bandwidth is often expended by the time that the higher-indexed upstreams are tried? If so, could we emit a warning when the bandwidth is too small? I appreciate #490 as a demonstration of the problem, but I anticipate that the users would prefer a deterministic solution.

maccoymerrell, 2024-09-04:
I have left a comment there as to how a more deterministic solution is possible. If you want some pseudo-randomness, then see my comments on using a permutation-based solution. Its deterministic, but depending on implementation can give a nice uniform distribution.

chaseblock, 2024-09-04:
> OK, this is interesting and gets me thinking. Is the problem that the bandwidth is less than the number of upstreams, so the bandwidth is often expended by the time that the higher-indexed upstreams are tried? If so, could we emit a warning when the bandwidth is too small? 

Low-bandwidth configurations are potentially interesting targets of investigations, so I don't think a warning would be sufficient. The issue here is not that the simulated machine's caches are under-provisioned, but rather that those caches prioritize some upstreams over others, leading to what appears to be exponentially-decreasing performance on higher-indexed cores.

> I appreciate #490 as a demonstration of the problem, but I anticipate that the users would prefer a deterministic solution.

Agreed that a deterministic solution would be better.

ngober, 2024-09-04:
There is a certain merit to prioritizing some upstreams over others. For example, #363 fixed a deadlock where the page walks would never be serviced because data requests (that were dependent on them) were consuming bandwidth. We want the translation requests to have priority, and that's why they have their own upstream. If `bandwidth < size(upper_levels)` is an interesting case, we should provide some protection against deadlocks, even if the user's own choices got them to that point.

chaseblock, 2024-09-05:
My concern is that there should be some amount of fairness enforced between the upstreams (even if that "enforcement" is stochastic) so that core7 doesn't turn out to be 10x slower than core0 for no good reason when running the same workload. Not merely deadlock/livelock protection.

maccoymerrell, 2024-09-05:
We can directly enforce fairness by dropping all randomness, and going straight for round robin. Just one line of code I think.

Maybe just insert 
`std::rotate(upper_levels.begin(), upper_levels.begin() + 1, upper_levels.end());`
After line 453 in cache.cc?

ngober, 2024-09-05:
That will accomplish it, I think. I'm trying to think though what effects that would have.

The bandwidth is consumed in this way:
 - for each upper level:
   - for each of RQ, WQ, PQ,
     - Consume as much remaining bandwidth as possible.

Maybe a broader change is needed where just one element from each upper level is consumed, and where that one comes with priority RQ > WQ > PQ (maybe that needs to be WQ > RQ > PQ?).

maccoymerrell, 2024-09-05:
Would that require multiple passes through the upstreams until all queues are either empty or bandwidth is consumed? That sounds like it potentially adds a lot of simulation overhead for large bandwidths. Maybe instead of just 1 at a time, it consumes max(bandwidth/upstream,1) at a time.

ngober, 2024-09-05:
I think that makes sense, consuming up to `max(bandwidth/size(upstreams),1)`. This should be per-upstream, not per-queue.

maccoymerrell, 2024-09-06:
Something like this then:

`  std::rotate(upper_levels.begin(), upper_levels.begin() + 1, upper_levels.end());

  //upper levels get an equal portion of the remaining bandwidth
  champsim::bandwidth::maximum_type per_upper_bandwidth{(champsim::bandwidth::maximum_type)(std::max((size_t)initiate_tag_bw.amount_remaining() / std::size(upper_levels),size_t{1}))};
  bool upper_consumed;
  do {
    upper_consumed = false;
    for (auto* ul : upper_levels) {
      for (auto q : {std::ref(ul->WQ), std::ref(ul->RQ), std::ref(ul->PQ)}) {
        champsim::bandwidth per_upper_tag_bw{std::min(per_upper_bandwidth,champsim::bandwidth::maximum_type{initiate_tag_bw.amount_remaining()})};
        auto bandwidth_consumed =
            champsim::transform_while_n(q.get(), std::back_inserter(inflight_tag_check), per_upper_tag_bw, can_translate, initiate_tag_check<true>(ul));
        channels_bandwidth_consumed.push_back(bandwidth_consumed);
        initiate_tag_bw.consume(bandwidth_consumed);
        upper_consumed |= bandwidth_consumed > 0;
      }
    }
  } while(upper_consumed && initiate_tag_bw.has_remaining());`
  
  This doesn't seem to incur any high simulation performance overheads with the testcase provided by Chase, though I noticed the IPC drops from an average of 0.07692 to 0.07352 when evenly distributing the bandwidth this way.
  With just round-robin, it performs about the same as random, if not better in regards to IPC.

maccoymerrell, 2024-09-09:
@ngober Should the technique you suggested still be incorporated, given that it appears to be somehow less performant on this workload? Or should we just stick with the round-robin?
@chaseblock Could you update your pull request to use this instead (dependent on ngober's response)

ngober, 2024-09-09:
The reason for my per-upstream bandwidth suggestion is that it requires only one pass through the upstreams, which I think is a better way to handle the code. Your snippet above wraps in a do-while. I get why this matters: you're trying to exhaust the bandwidth if at all possible. But, it introduces a third level of looping in this code block that I'm not comfortable with because it reduces code comprehensibility. I'm willing to wager that the loop only sees one pass in low-bandwidth cases, and also that other workloads would see different performance diff.

chaseblock, 2024-09-09:
> Maybe just insert `std::rotate(upper_levels.begin(), upper_levels.begin() + 1, upper_levels.end());` After line 453 in cache.cc?

That's probably better than the shuffle that I had. Changed the PR to this for now.

> The reason for my per-upstream bandwidth suggestion is that it requires only one pass through the upstreams, which I think is a better way to handle the code.

What's the objection to just doing the `rotate` without per-upstream bandwidth limitations?

ngober, 2024-09-10:
> What's the objection to just doing the `rotate` without per-upstream bandwidth limitations?

Timing and fairness, really. In an 8-core system at full load, a request could be delayed 1-7 cycles arbitrarily because it arrived on the wrong cycle. That seems to be against the spirit of the fix we're discussing. If we spread out the bandwidth we both model parallelism better and also provide more forward-progress guarantees.

maccoymerrell, 2024-09-12:
```
std::rotate(upper_levels.begin(), upper_levels.begin() + 1, upper_levels.end());

  //upper levels get an equal portion of the remaining bandwidth
  champsim::bandwidth::maximum_type per_upper_bandwidth{(champsim::bandwidth::maximum_type)(std::max((size_t)initiate_tag_bw.amount_remaining() / std::size(upper_levels),size_t{1}))};
  for (auto* ul : upper_levels) {
    for (auto q : {std::ref(ul->WQ), std::ref(ul->RQ), std::ref(ul->PQ)}) {
      champsim::bandwidth per_upper_tag_bw{std::min(per_upper_bandwidth,champsim::bandwidth::maximum_type{initiate_tag_bw.amount_remaining()})};
      auto bandwidth_consumed =
          champsim::transform_while_n(q.get(), std::back_inserter(inflight_tag_check), per_upper_tag_bw, can_translate, initiate_tag_check<true>(ul));
      channels_bandwidth_consumed.push_back(bandwidth_consumed);
      initiate_tag_bw.consume(bandwidth_consumed);
    }
  }
```

ngober, 2024-09-12:
The amount of type casting in the declaration of `per_upper_bandwidth` makes me wince a little, but other than that it's good. Someone should probably write a test with like 20 upstreams and 2 bandwidth to make sure things do what we intend them to do.

maccoymerrell, 2024-09-16:
@chaseblock Could you update your pr to include this change as well? I agree with Nathan that in a multi-core situation, we should be guaranteeing bandwidth to all cores regardless of cycle if possible.

ngober, 2024-09-17:
Yea, this feels like it exceeds the bug reported here and starts going into new-feature territory. I'm fine with modeling banks if someone wants to it would be particularly interesting to model a sliced LLC.

chaseblock, 2024-09-19:
Apologies for the delay. I agree that it would be better to model the banks eventually, but the currently-proposed fix probably works well enough for now.

Running on an 8-core configuration with the `LLC.max_tag_check=1` provides the following result.

```
⇒ bin/champsim -w 0 -i 1000000 403.gcc-16B.champsimtrace.xz 403.gcc-16B.champsimtrace.xz 403.gcc-16B.champsimtrace.xz 403.gcc-16B.champsimtrace.xz 403.gcc-16B.champsimtrace.xz 403.gcc-16B.champsimtrace.xz 403.gcc-16B.champsimtrace.xz 403.gcc-16B.champsimtrace.xz |& tee sim.out
WARNING: physical memory size is smaller than virtual memory size.

*** ChampSim Multicore Out-of-Order Simulator ***
Warmup Instructions: 0
Simulation Instructions: 1000000
Number of CPUs: 8
Page size: 4096

Off-chip DRAM Size: 64 MiB Channels: 1 Width: 64-bit Data Rate: 3205 MT/s
Warmup finished CPU 7 instructions: 0 cycles: 1 cumulative IPC: 0 (Simulation time: 00 hr 00 min 00 sec)
Warmup finished CPU 6 instructions: 0 cycles: 1 cumulative IPC: 0 (Simulation time: 00 hr 00 min 00 sec)
Warmup finished CPU 5 instructions: 0 cycles: 1 cumulative IPC: 0 (Simulation time: 00 hr 00 min 00 sec)
Warmup finished CPU 4 instructions: 0 cycles: 1 cumulative IPC: 0 (Simulation time: 00 hr 00 min 00 sec)
Warmup finished CPU 3 instructions: 0 cycles: 1 cumulative IPC: 0 (Simulation time: 00 hr 00 min 00 sec)
Warmup finished CPU 2 instructions: 0 cycles: 1 cumulative IPC: 0 (Simulation time: 00 hr 00 min 00 sec)
Warmup finished CPU 1 instructions: 0 cycles: 1 cumulative IPC: 0 (Simulation time: 00 hr 00 min 00 sec)
Warmup finished CPU 0 instructions: 0 cycles: 1 cumulative IPC: 0 (Simulation time: 00 hr 00 min 00 sec)
Warmup complete CPU 7 instructions: 0 cycles: 1 cumulative IPC: 0 (Simulation time: 00 hr 00 min 00 sec)
Warmup complete CPU 6 instructions: 0 cycles: 1 cumulative IPC: 0 (Simulation time: 00 hr 00 min 00 sec)
Warmup complete CPU 5 instructions: 0 cycles: 1 cumulative IPC: 0 (Simulation time: 00 hr 00 min 00 sec)
Warmup complete CPU 4 instructions: 0 cycles: 1 cumulative IPC: 0 (Simulation time: 00 hr 00 min 00 sec)
Warmup complete CPU 3 instructions: 0 cycles: 1 cumulative IPC: 0 (Simulation time: 00 hr 00 min 00 sec)
Warmup complete CPU 2 instructions: 0 cycles: 1 cumulative IPC: 0 (Simulation time: 00 hr 00 min 00 sec)
Warmup complete CPU 1 instructions: 0 cycles: 1 cumulative IPC: 0 (Simulation time: 00 hr 00 min 00 sec)
Warmup complete CPU 0 instructions: 0 cycles: 1 cumulative IPC: 0 (Simulation time: 00 hr 00 min 00 sec)
Simulation finished CPU 4 instructions: 1000000 cycles: 11662371 cumulative IPC: 0.08575 (Simulation time: 00 hr 03 min 24 sec)
Simulation finished CPU 5 instructions: 1000000 cycles: 12779259 cumulative IPC: 0.07825 (Simulation time: 00 hr 03 min 44 sec)
Simulation finished CPU 2 instructions: 1000000 cycles: 13073287 cumulative IPC: 0.07649 (Simulation time: 00 hr 03 min 49 sec)
Simulation finished CPU 6 instructions: 1000000 cycles: 13134292 cumulative IPC: 0.07614 (Simulation time: 00 hr 03 min 50 sec)
Simulation finished CPU 7 instructions: 1000000 cycles: 13259040 cumulative IPC: 0.07542 (Simulation time: 00 hr 03 min 52 sec)
Simulation finished CPU 1 instructions: 1000000 cycles: 13333219 cumulative IPC: 0.075 (Simulation time: 00 hr 03 min 53 sec)
Simulation finished CPU 3 instructions: 1000000 cycles: 13793452 cumulative IPC: 0.0725 (Simulation time: 00 hr 04 min 01 sec)
Simulation finished CPU 0 instructions: 1000000 cycles: 13952744 cumulative IPC: 0.07167 (Simulation time: 00 hr 04 min 04 sec)
Simulation complete CPU 7 instructions: 1061135 cycles: 13952744 cumulative IPC: 0.07605 (Simulation time: 00 hr 04 min 04 sec)
Simulation complete CPU 6 instructions: 1064047 cycles: 13952744 cumulative IPC: 0.07626 (Simulation time: 00 hr 04 min 04 sec)
Simulation complete CPU 5 instructions: 1090767 cycles: 13952744 cumulative IPC: 0.07818 (Simulation time: 00 hr 04 min 04 sec)
Simulation complete CPU 4 instructions: 1201263 cycles: 13952744 cumulative IPC: 0.0861 (Simulation time: 00 hr 04 min 04 sec)
Simulation complete CPU 3 instructions: 1009551 cycles: 13952744 cumulative IPC: 0.07236 (Simulation time: 00 hr 04 min 04 sec)
Simulation complete CPU 2 instructions: 1076815 cycles: 13952744 cumulative IPC: 0.07718 (Simulation time: 00 hr 04 min 04 sec)
Simulation complete CPU 1 instructions: 1058799 cycles: 13952744 cumulative IPC: 0.07588 (Simulation time: 00 hr 04 min 04 sec)
Simulation complete CPU 0 instructions: 1000000 cycles: 13952744 cumulative IPC: 0.07167 (Simulation time: 00 hr 04 min 04 sec)

ChampSim completed all CPUs

```

The PR has been updated with this change.

maccoymerrell, 2024-09-19:
See comment on PR

ngober, 2024-09-30:
Resolved with #490

closed 2024-09-30 by ngober