ChampSim/ChampSim #455: Tracer Does Not Stop Tracing After the Set Number is Reached

WEIwjr, 2024-03-28:
**Describe the bug**
I'm trying to use the tracer to trace memory operations of a program, however I found that the tracer does not stop tracing when the set number of instructions to trace is reached.

To fulfill my need I made the following edit to the original tracer:
1. I rewrite the WriteCurrentInstruction function so that it writes PC and memory addresses directly to a file. 
2. I comment out the code that instruments branch instructions and register read/write instructions. 

**To Reproduce**
$PIN_ROOT/pin -t $USER/ChampSim-master-5/tracer/pin/obj-intel64/champsim_tracer_edit.so -o ./raw_0.txt -t 10000 -- ls

**Output**
For the output file I append the instrCount and KnobTraceInstructions.Value() to the end of each line, and the output file looks like this:
```
2b17a96c7140 0 0 0 0 0 0 1 2710
2b17a96c7143 3c9a8c78 0 0 0 0 0 2 2710
2b17a96c7850 3c9a8c70 0 0 0 0 0 3 2710
...
2b17a96cf76b 0 0 be680815 0 0 0 2fde1 2710
2b17a96cf772 0 0 0 0 0 0 2fde2 2710
2b17a96cf778 0 0 a98e7c00 0 0 0 2fde3 2710
2b17a96cf77f 0 0 0 0 0 0 2fde4 2710
2b17a96cf785 0 0 be6807ec 0 0 0 2fde5 2710
2b17a96cf78b 0 0 0 0 0 0 2fde6 2710
2b17a96cf78d 0 0 0 0 0 0 2fde7 2710
...
```

**Desktop (please complete the following information):**
 - OS: 
NAME="Red Hat Enterprise Linux Server"
VERSION="7.9 (Maipo)"
ID="rhel"
ID_LIKE="fedora"
VARIANT="Server"
VARIANT_ID="server"
VERSION_ID="7.9"
PRETTY_NAME="Red Hat Enterprise Linux"
ANSI_COLOR="0;31"
CPE_NAME="cpe:/o:redhat:enterprise_linux:7.9:GA:server"
HOME_URL="https://www.redhat.com/"
BUG_REPORT_URL="https://bugzilla.redhat.com/"

REDHAT_BUGZILLA_PRODUCT="Red Hat Enterprise Linux 7"
REDHAT_BUGZILLA_PRODUCT_VERSION=7.9
REDHAT_SUPPORT_PRODUCT="Red Hat Enterprise Linux"
REDHAT_SUPPORT_PRODUCT_VERSION="7.9"
 - Compiler [e.g. GCC, LLVM]
gnu/8.4.0/bin/gcc

**Additional context**
Here is the edited code:
```
/*
 *    Copyright 2023 The ChampSim Contributors
 *
 * Licensed under the Apache License, Version 2.0 (the "License");
 * you may not use this file except in compliance with the License.
 * You may obtain a copy of the License at
 *
 * http://www.apache.org/licenses/LICENSE-2.0
 *
 * Unless required by applicable law or agreed to in writing, software
 * distributed under the License is distributed on an "AS IS" BASIS,
 * WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
 * See the License for the specific language governing permissions and
 * limitations under the License.
 */

/*! @file
 *  This is an example of the PIN tool that demonstrates some basic PIN APIs
[... 168 lines cut ...]
int main(int argc, char* argv[])
{
  // Initialize PIN library. Print help message if -h(elp) is specified
  // in the command line or the command line is invalid
  if (PIN_Init(argc, argv))
    return Usage();

  outfile.open(KnobOutputFile.Value().c_str(), std::ios_base::binary | std::ios_base::trunc);
  if (!outfile) {
    std::cout << "Couldn't open output trace file. Exiting." << std::endl;
    exit(1);
  }

  // Register function to be called to instrument instructions
  INS_AddInstrumentFunction(Instruction, 0);

  // Register function to be called when the application exits
  PIN_AddFiniFunction(Fini, 0);

  // Start the program, never returns
  PIN_StartProgram();

  return 0;
}
```

joshualmashburn, 2024-11-11:
I can confirm this bug on my machine using the unmodified tracer. I get a 35MB trace for "10000" instructions; that should be 640000B. I'm using Pin 3.30 however since my Fedora install is too new for 3.22.
I have an older Ubuntu install I'll test this evening, to rule out some change Pin made since 3.22.

ngober, 2024-11-11:
Is there something wrong with the way that the tracer does the `INS_InsertIfCall(..., ShouldWrite, ...)` statement?

joshualmashburn, 2024-11-11:
I'm investigating that now. I'm getting weird behavior. As a first pass fix, I inserted the following, at the end of the Instruction() function:
```
  if (instrCount >= (KnobTraceInstructions.Value() + KnobSkipInstructions.Value()))
    PIN_ExitApplication(0);
```
I then run the tracer for 256 instructions (on his code for readability, but I'll confirm in a moment on the unmodified tracer).
It keeps going til 0x10a instructions (I first used equals, but it missed it and kept going to completion). I'm wondering if either the instruction count isn't being synchronized somehow, or if the inserted calls are subject to instrumentation? If I go for only 64 instructions, it does terminate right at 64.

joshualmashburn, 2025-12-04:
Versus:

[image]

joshualmashburn, 2025-12-04:
GOT IT. Fix inbound. We shouldn't be calling the knobs' value() in instrumentation, and there may be synchronization issues.

closed 2025-12-04 by ngober