ChampSim/ChampSim #457: Build Error: Tukani-Project Dependencies

yashgyy, 2024-04-04:
Having issues building Champsim as dependency tukani-project is disabled by github for violation of terms. There is a need to ensure vcpkg handle this dependencies because currently it is not doing so

ngober, 2024-04-04:
Would you go into more detail about the issue you're experiencing? I don't see that Github marks xz as a dependency of ChampSim.

yashgyy, 2024-04-04:
Apologies @ngober The issue is with vcpkg which is used to manage the dependencies . Apparently one of depend there gone bust https://github.com/microsoft/vcpkg/issues/37962#issuecomment-2036223908

They are actively resolving there it seems ..

ngober, 2024-04-04:
You may be able to get away with deleting the dependency from `vcpkg.json`, since an un tainted version is likely already present on your system.

yashgyy, 2024-04-04:
@ngober Unfortunately it isnt possible for we are making a separate a docker container for champsim and working on it

closed 2024-11-08 by ngober