ChampSim/ChampSim #461: use doubt:  IPC before and after warm up

Maxpicca-Li, 2024-04-22:
* Why is there such a big gap between IPC before and after warming up?
* Logically, the IPC should be lower before the warm-up.
[image]

ngober, 2024-04-22:
You say that logic demands the IPC should be lower in the warmup, but it is not clear why that should be the case. In the warmup phase, most latencies are skipped, then the statistics are discarded. The final statistics reported consist only of non-warmup phases.

Maxpicca-Li, 2024-04-23:
I got that, thank you very much.

As for the lower IPC in warm-up, both the cache and the prefetcher will be cold-started during the warm-up phase, so the normal IPC will be lower. It turns out that the IPC calculation in warmup phase is not the same as the normal calculation, thank you for the answer.

closed 2024-04-23 by ngober