ChampSim/ChampSim #464: Support for Simultaneous Multi-Threading

Deb-ncsu, 2024-04-25:
Hello, my research specifically demands simulation with simultaneous multi-threading. At this point, I understand that it is not supported. My question is is there a plan to support SMT? Additionally, if I want to modify the simulator to add the support for the SMT, is it doable?

ngober, 2024-04-25:
There is currently no SMT support in ChampSim. It could be possible, but nobody has come forward to put in the work required. Here is my estimate of what would be required.

- Currently, each core is statically assigned a trace to read from. This would need to be modified to switch between 2 (or more) traces. This would still be multi-program, not multi-threaded programs. That support would take significantly more work.
- These traces would need to have different ASIDs. I'm aware of someone working to fix ASID support in ChampSim, and it has been a tricky process, the feature has been vestigial for some time.
- The statistics should be modified to report the different threads.

Deb-ncsu, 2024-04-25:
Thanks for the very quick response. 
Currently, I am mostly interested in multi-programmed workloads colocated in a physical core and use SMT to share resources.

qiao-huan, 2024-11-05:
Hello, may I ask if you have implemented SMT? The work I am currently doing also requires SMT support. Thanks.

ngober, 2024-11-05:
No, there is still no support. Nobody has stepped forward to implement it. Would you like to help with that?

qiao-huan, 2024-11-05:
> No, there is still no support. Nobody has stepped forward to implement it. Would you like to help with that?

Sure, thank you. 
Yes, I'm currently trying to modify ChampSim to support SMT. If I make any progress, I'll be happy to share it here with you buddies.