ChampSim/ChampSim #468: Cache Latency

agusnt, 2024-05-02:
Hi everyone,

I'm trying to change the latency of the L1D cache. I changed the "latency" parameter in the JSON file from 5 to 200 cycles, but the simulation output is the same.

After digging into the code, it seems that to change the actual latency, you need to modify the "hit_latency" field instead of "latency." What is the purpose of the "latency" field? To change the cache latency, do I only need to modify the "hit_latency" and "fill_latency" fields?

Thank you.

ngober, 2024-05-03:
I think this was a bug that got fixed in 889b5a4. There appears to be [a test for it](https://github.com/ChampSim/ChampSim/blob/396ff4ae34b55c78765301e13a9aff2e1bd9cdf0/test/cpp/src/091-cache-builder.cc#L218) now.

agusnt, 2024-05-06:
I check with this configuration file (https://pastebin.com/N1PUenLy) and the HIT_LATENCY value is still 4 (I print the value in the `operate()` function of `src/cache.cc`). 

Maybe the issue comes from the JSON parser?

Commit number: 2b8d3fc28abb6072d7675228418fa7cfe862d4dc (latest one)

ngober, 2024-05-06:
Did you try it on develop? The commit I referenced is not on master.

agusnt, 2024-05-09:
I am sorry.

I try it on develop, it does not build (default configuration): 

`ChampSim/.csconfig/core_inst.cc.inc:18:41: error: use of undeclared identifier 'v' champsim::channel{queue_fmtstr.format(**v)},`

ngober, 2024-05-09:
Oops, that looks like a bad python format string. It's fixed now. Good catch.

agusnt, 2024-05-10:
I pull the develop branch and build it.

I change the L1D latency field from 5 to 200, and I have the same simulation output: [5 cycles L1D latency](https://pastebin.com/2UhC9AM3) and [200 cycles L1D latency](https://pastebin.com/Y9PGKERW)

GinoAC, 2024-06-02:
Try defining "hit_latency" in the configuration file for the L1D. You can also do the same for "fill_latency".

agusnt, 2024-06-10:
@GinoAC That solves the problem. 

Thank you