ChampSim/ChampSim #492: MacOS build errors - Makefile Issues

maccoymerrell, 2024-09-06:
It seems develop does not build successfully on MacOS

From what I can tell, this is related to the ordering of targets in the makefile.
The directory creation portion of the makefile is making the *.d targets as directories as well instead of just their parent directories, which causes those *.d to never be made and gcc to complain since it doesn't take directories as inputs.

ngober, 2024-09-06:
I've seen this in the CI more than once. @GinoAC did you ever figure out a solution to it?

GinoAC, 2024-09-06:
My workaround ended up breaking a different portion of the build script. I ended up just switching back to WSL...

ngober, 2024-09-06:
Is this just a result of the `.d` suffix? Would a different suffix work?

maccoymerrell, 2024-09-06:
I don't think so. I think it has something to do with make interpreting wildcards differently on Mac. So when directories are made, with rules targeting those directories, it is also applying the rule to files within those directories.

maccoymerrell, 2024-09-06:
I was able to replicate this behavior on linux by moving the rules making directories to the bottom of the makefile.

ngober, 2024-09-06:
So is the solution to reorder things in the Makefile?

maccoymerrell, 2024-09-06:
Probably, but I don't have a good way to test my changes locally for Mac

maccoymerrell, 2024-09-12:
I think this issue might be resolved. My most recent tests on pr #390 all passed, and I didn't touch anything.

ngober, 2024-09-12:
Resolved... how?

maccoymerrell, 2024-09-12:
I did two things in the recent commit that might have fixed it.
1. I removed legacy spp, so the makefile bug my not be triggered from that anymore
2. Line 92 in the makefile, I changed the shell command from python to python3 since that was causing a lot of headache and seemed inconsistent with the rest of the makefile.

Otherwise, it was a ghost. 👻

maccoymerrell, 2024-09-12:
I am pretty confident at this point its the makefile python version change on line 92.
The pull request with just the spp changes fail.

ngober, 2024-09-12:
I wouldn't be astonished if that were true, but it seems that neither [ubuntu-latest](https://github.com/actions/runner-images/blob/main/images/ubuntu/Ubuntu2204-Readme.md) nor [macos-latest](https://github.com/actions/runner-images/blob/macos-14-arm64/20240903.5/images/macos/macos-14-arm64-Readme.md) have python2 installed. What else would `python` symlink to?

maccoymerrell, 2024-09-12:
I think the problem may be that it doesn't symlink to anything. Maybe

ngober, 2024-09-13:
Would you PR a change with just this, so that we can see if it passes the Mac CI?

closed 2024-09-13 by ngober