ChampSim/ChampSim #499: ChampSim develop branch build runs crash with non-zero Warmup Instructions

Rajarshii, 2024-09-13:
Possible Memory leak when running the latest develop branch build with non-zero warmup instructions.

Steps to reproduce the behavior: -
1. checkout the latest develop branch: ChampSim/develop
2. Follow the usual build steps (only one core) - 
   a. ./config.sh your_config.json
   b. make 
3. Run a single core simulation: e.g.: bin/champsim --warmup-instructions=100000000 --simulation-instructions=250000000 619.lbm_s-4268B.champsimtrace.xz 

Expectation: -
 Simulation should not hang/ consume too much memory 

Observation: -
Simulation consumes several GBs of memory (over 24GiB - after which simulation gets killed on my system)

Setup: -
 - OS: Ubuntu 22.04 LTS (22.9GiB of usable memory, 2GiB swap)
 - Compiler: GCC 
 - Version: 11.4.0

Additional Info: -
Changing main.cc line 104:
warmup_instructions = simulation_instructions / 5; 
    to warmup_instructions = 0;
And simulate without passing --warmup-instruction 
i.e. 
bin/champsim --simulation-instructions=250000000 619.lbm_s-4268B.champsimtrace.xz 
So that no warmup instructions are executed. The simulation does not hang or use large amounts of memory.

maccoymerrell, 2024-09-13:
I can verify this appears to be an issue. Something to do with warmup not emptying a queue maybe?

ngober, 2024-09-13:
Does the simulation make it past warmup? Does it execute any instructions at all?

maccoymerrell, 2024-09-13:
If warmup is set low, or 0, it appears to run normally. Mem leaking stops once warmup ends. So if you have a long warmup, you will likely crash from oom.

maccoymerrell, 2024-09-13:
I have had students running on my DRAM_Controller branch. Typically only short warmups, and it seems to be working.

maccoymerrell, 2024-09-13:
champsim::chrono::duration{} evaluates to 0 right? @ngober
I have been trying to diff master and develop on my phone by sight. I would think that giving something a completion time infinitely in the future would cause queues to fill up, but...

maccoymerrell, 2024-09-13:
~\ One of these things is not like the others /~

DIB HIT BUF SIZE: 182644 IFETCH BUF SIZE: 12 DECODE BUF SIZE: 0
DIB HIT BUF SIZE: 182646 IFETCH BUF SIZE: 12 DECODE BUF SIZE: 0
DIB HIT BUF SIZE: 182648 IFETCH BUF SIZE: 12 DECODE BUF SIZE: 0
DIB HIT BUF SIZE: 182650 IFETCH BUF SIZE: 12 DECODE BUF SIZE: 0
DIB HIT BUF SIZE: 182652 IFETCH BUF SIZE: 12 DECODE BUF SIZE: 0
DIB HIT BUF SIZE: 182654 IFETCH BUF SIZE: 12 DECODE BUF SIZE: 0
DIB HIT BUF SIZE: 182656 IFETCH BUF SIZE: 12 DECODE BUF SIZE: 0
DIB HIT BUF SIZE: 182658 IFETCH BUF SIZE: 11 DECODE BUF SIZE: 0
DIB HIT BUF SIZE: 182660 IFETCH BUF SIZE: 11 DECODE BUF SIZE: 0
DIB HIT BUF SIZE: 182661 IFETCH BUF SIZE: 12 DECODE BUF SIZE: 0
DIB HIT BUF SIZE: 182663 IFETCH BUF SIZE: 12 DECODE BUF SIZE: 0
DIB HIT BUF SIZE: 182665 IFETCH BUF SIZE: 12 DECODE BUF SIZE: 0
DIB HIT BUF SIZE: 182667 IFETCH BUF SIZE: 12 DECODE BUF SIZE: 0
DIB HIT BUF SIZE: 182669 IFETCH BUF SIZE: 12 DECODE BUF SIZE: 0
DIB HIT BUF SIZE: 182671 IFETCH BUF SIZE: 12 DECODE BUF SIZE: 0
DIB HIT BUF SIZE: 182673 IFETCH BUF SIZE: 12 DECODE BUF SIZE: 0
DIB HIT BUF SIZE: 182675 IFETCH BUF SIZE: 12 DECODE BUF SIZE: 0
DIB HIT BUF SIZE: 182677 IFETCH BUF SIZE: 12 DECODE BUF SIZE: 0
DIB HIT BUF SIZE: 182679 IFETCH BUF SIZE: 12 DECODE BUF SIZE: 0
DIB HIT BUF SIZE: 182681 IFETCH BUF SIZE: 12 DECODE BUF SIZE: 0
DIB HIT BUF SIZE: 182683 IFETCH BUF SIZE: 12 DECODE BUF SIZE: 0
DIB HIT BUF SIZE: 182685 IFETCH BUF SIZE: 12 DECODE BUF SIZE: 0
DIB HIT BUF SIZE: 182687 IFETCH BUF SIZE: 12 DECODE BUF SIZE: 0
DIB HIT BUF SIZE: 182689 IFETCH BUF SIZE: 12 DECODE BUF SIZE: 0
DIB HIT BUF SIZE: 182691 IFETCH BUF SIZE: 12 DECODE BUF SIZE: 0
DIB HIT BUF SIZE: 182693 IFETCH BUF SIZE: 12 DECODE BUF SIZE: 0
DIB HIT BUF SIZE: 182695 IFETCH BUF SIZE: 12 DECODE BUF SIZE: 0
DIB HIT BUF SIZE: 182697 IFETCH BUF SIZE: 12 DECODE BUF SIZE: 0
DIB HIT BUF SIZE: 182699 IFETCH BUF SIZE: 12 DECODE BUF SIZE: 0
DIB HIT BUF SIZE: 182701 IFETCH BUF SIZE: 12 DECODE BUF SIZE: 0
DIB HIT BUF SIZE: 182703 IFETCH BUF SIZE: 12 DECODE BUF SIZE: 0
DIB HIT BUF SIZE: 182705 IFETCH BUF SIZE: 12 DECODE BUF SIZE: 0
DIB HIT BUF SIZE: 182707 IFETCH BUF SIZE: 12 DECODE BUF SIZE: 0

ngober, 2024-09-13:
@zmkzLe did we make sure we capped the maximum size of that buffer?

I have learned the hard way: always limit the size of your buffers. The limits will always be tested.

zmkzLe, 2024-09-15:
@ngober We did cap the size but there is actually a bug. In ooo_cpu.cc, the promote_to_decode() that I changed does not check for the limits in DECODE_BUFFER and DIB_HIT_BUFFER so instructions keep getting back_insert() into these two, making entries exceeded the limits.

ngober, 2024-09-16:
Could either @Rajarshii or @maccoymerrell please confirm that #503 resolved this?

maccoymerrell, 2024-09-16:
@ngober I ran this on my own machine and it seems to have fixed the issue. The buffer sizes remain capped, and base memory usage is down from 700MB+ to ~70MB during and after warmup.

closed 2024-09-16 by ngober