ChampSim/ChampSim #502: ChampSim DRAM Size is Unclear

maccoymerrell, 2024-09-14:
**Is your feature request related to a problem? Please describe.**
DRAM is indexed per-address as channel-width blocks. The DRAM can prefetch some arbitrary multiple of this at once to serve large blocks of memory at once.

DRAM modules have banks/bankgroups, rows, and columns. These are defined in a bit-address space. Each DRAM module outputs some fraction of a full channel-width for each [bank,row,column] index.

Above the modules exist ranks, which represent a collection of dram modules that provide the full channel-width of data.
Above that exists channels which represent collections of ranks under a single memory channel.

ChampSim instead provides channels, ranks, banks, rows, and columns as input parameters. All fields are accurate, except columns, which appear to be given in # of cache blocks that can fit in the column space.

To calculate size, ChampSim uses the address space to calculate the size of dram, and assumes a certain amount of data available at each address (on master, each address indexes a block, while on develop each address indexes a channel width [which is correct, assuming columns are addressing bytes, which they aren't]). 
Both of these sizes may be inaccurate depending on the prefetch size and channel width of the dram.

**Describe the solution you'd like**
Inside the dram memory section of the config, there needs to be additional parameters
1. prefetch_width = number of columns grouped together to serve requests, this should generally be (block_size/channel_width)*N.
2. columns should describe available BITS, not bytes or blocks.

From here, we can calculate size accurately:
chip bit density = (banks * rows * columns)
rank bit density = chip bit density * channel_width
channel bit density = rank bit density * ranks
size (bytes) = (channel bit density * channels) / 8

To address:
Before slicing, each address is right-shifted by the channel size.
Slice columns, rows, banks, ranks, channels
Zero out the lower log2(prefetch_size) bits. The packets should be served by the same dram operation.

ngober, 2024-09-15:
I confess to not being a CPU expert, not a DRAM expert. You're probably the first person in ChampSim's history to put significant thought into this.

I'm trying to understand the utility of the prefetch width. Requests to the controller will only ever have width `1*block_width`. So, every time the request goes out, it seems all `prefetch_width` of the columns will be activated and used in the return. There's no prefetching going on.

I agree with the point about column widths being described in bits. It was described in bytes a long time ago, before me, and I simply maintained the behavior. In fact, the opportunity to describe `champsim::bytes` and `champsim::bits` is only very recent, and is only on the develop branch. That change should certainly be made, but probably we should deprecate and replace, rather than change. People don't like it when behavior changes out from under them.

maccoymerrell, 2024-09-15:
I am currently working on a branch to do these changes.

Interestingly, I thought the same about prefetch_width until DDR5. DDR5 supports a prefetch width of up to 16, meaning that two blocks can be bursted from the DRAM at once.
What this means in controller: Requests can be merged into prefetch blocks, saving queue space (which is important at high bandwidths) and we also avoid the overhead of issuing two separate bursts to obtain the data. You can imagine that consecutive cache blocks is actually a pretty common occurrence in some workloads, especially in machines with a lot of cache stress.

I actually have all these changes made already. The only caveat right now is that when we merge, we assume the block address is the same between the two blocks. Instead, queue entries in the memory controller will need to have a vector of packets that its serving, instead of a single one.

maccoymerrell, 2024-09-15:
[](https://github.com/maccoymerrell/ChampSimFix/tree/DRAM_Config_Dev) has my current branch with all these changes functional.

The way I am managing the prefetch_size packet merging is killing performance though. I am using a vector of packets, which I think is incurring a lot of overhead in initialization time when moving packets into the RQ and WQ of the memory controller.
I am not a container wizard, so I am sure there is a much better solution out there. Maybe a const vector of std::optionals of prefetch size entries?

ngober, 2024-09-22:
Resolved with #504

closed 2024-09-22 by ngober