ChampSim/ChampSim #518: Fetching cache lines whose instructions are all hits in the DIB

dpuckett98, 2024-10-09:
**Describe the bug**
`fetch_instruction` doesn't check whether or not the instructions in a cache block have hit in the DIB before it issues a fetch for that cache block. This isn't necessarily incorrect behavior, but it seems suboptimal to me (sorry if this should've been a feature request instead of a bug report). Based on my testing, it frequently occurs that all of the instructions in a cache block hit in the DIB. For example, I tested a SPEC 2017 wrf SimPoint and found that checking if the instructions in a cache block hit in the DIB before fetching that cache block reduced the number of L1I accesses by 15x. Interestingly, this only corresponded to a 1.0002x decrease in the number of cycles.

**To Reproduce**
Steps to reproduce the behavior:
1. Get a fresh install of ChampSim (develop branch)
2. Insert the following code at line 255 of src/ooo_cpu.cc:
```
auto not_dib_hit = [](const ooo_model_instr& x) {
  return !x.decoded;
};
if (std::find_if(l1i_req_begin, l1i_req_end, not_dib_hit) == l1i_req_end) {
  fmt::print("Every instr in fetch was a DIB hit\n");
}
```
3. Run ChampSim (any trace or configuration seems to reproduce the bug; one configuration I've specifically tested is with SPEC 2017 602-wrf-8100B, 100K warmup instructions, 1M simulation instructions)

**Expected behavior**
Nothing is printed by the `fmt::print` inserted in step 2.

**Output**
"Every instr in fetch was a DIB hit" is printed many, many times.

ngober, 2024-10-09:
I remember having this conversation, and I think it had something to do with wanting coherence between the DIB and L1I, and it's related to the fact that the trace is the thing driving fetch, not the branch predictor. This feels like a microarchitectural discussion, not a simulator bug. As in: what is the thing that we want to model, irrespective of whether that model is correct or not.

dpuckett98, 2024-10-15:
After some discussion, ChampSim's current behavior is the one that we want to model because a) we're currently modeling what happens if we check the L1I cache and DIB simultaneously, which is probably more realistic than checking DIB first, then the L1I cache; b) there isn't a significant difference in IPC either way; and c) it may even be better for performance to keep the L1I replacement policy aware of recently used cache lines.

closed 2024-10-15 by dpuckett98