ChampSim/ChampSim #519: VA_AMPM_LITE slows simulation to crawl

maccoymerrell, 2024-10-10:
VA_AMPM_LITE seems to be doing something that kills performance.
If you run these long enough, eventually some sort of live-lock takes place and the simulation never finishes.

**To Reproduce**
1. Use va_ampm_lite as a prefetcher, enable virtual prefetching in the config.
2. Run trace 602.gcc_s-2226B.champsimtrace.xz for at 30M instructions (this will take a few hours for some reason)

**Expected behavior**
The ChampSim run should complete at 30M instructions in about 10-15 minutes.

**Output**

WITH AMPM (300k instructions)

(base) bash-4.4$ ./bin/champsim -i300000 -w0 /mnt/research/Gratz_Paul_V/Shared/champsim-traces/speccpu/6xx/602.gcc_s-2226B.champsimtrace.xz
WARNING: physical memory size is smaller than virtual memory size.

*** ChampSim Multicore Out-of-Order Simulator ***
Warmup Instructions: 0
Simulation Instructions: 300000
Number of CPUs: 1
Page size: 4096

Off-chip DRAM Size: 16 GiB Channels: 1 Width: 64-bit Data Rate: 3205 MT/s
Warmup finished CPU 0 instructions: 0 cycles: 1 cumulative IPC: 0 (Simulation time: 00 hr 00 min 00 sec)
Warmup complete CPU 0 instructions: 0 cycles: 1 cumulative IPC: 0 (Simulation time: 00 hr 00 min 00 sec)
Simulation finished CPU 0 instructions: 300001 cycles: 283756 cumulative IPC: 1.057 (Simulation time: 00 hr 01 min 53 sec)
Simulation complete CPU 0 instructions: 300001 cycles: 283756 cumulative IPC: 1.057 (Simulation time: 00 hr 01 min 53 sec)

ChampSim completed all CPUs

=== Simulation ===
CPU 0 runs /mnt/research/Gratz_Paul_V/Shared/champsim-traces/speccpu/6xx/602.gcc_s-2226B.champsimtrace.xz

Region of Interest Statistics

CPU 0 cumulative IPC: 1.057 instructions: 300001 cycles: 283756
CPU 0 Branch Prediction Accuracy: 99.08% MPKI: 3.283 Average ROB Occupancy at Mispredict: 110.6
Branch type MPKI
BRANCH_DIRECT_JUMP: 0.06667
BRANCH_INDIRECT: 0
BRANCH_CONDITIONAL: 3.193
BRANCH_DIRECT_CALL: 0.01
BRANCH_INDIRECT_CALL: 0
BRANCH_RETURN: 0.01333

cpu0_STLB TOTAL        ACCESS:      24314 HIT:      13703 MISS:      10611
cpu0_STLB LOAD         ACCESS:      24314 HIT:      13703 MISS:      10611
cpu0_STLB RFO          ACCESS:          0 HIT:          0 MISS:          0
cpu0_STLB PREFETCH     ACCESS:          0 HIT:          0 MISS:          0
cpu0_STLB WRITE        ACCESS:          0 HIT:          0 MISS:          0
cpu0_STLB TRANSLATION  ACCESS:          0 HIT:          0 MISS:          0
cpu0_STLB PREFETCH REQUESTED:          0 ISSUED:          0 USEFUL:          0 USELESS:          0
cpu0_STLB AVERAGE MISS LATENCY: 6.704 cycles
cpu0_L2C TOTAL        ACCESS:      45381 HIT:       4910 MISS:      40471
cpu0_L2C LOAD         ACCESS:      21137 HIT:       4871 MISS:      16266
cpu0_L2C RFO          ACCESS:          1 HIT:          0 MISS:          1
cpu0_L2C PREFETCH     ACCESS:      24133 HIT:          3 MISS:      24130
cpu0_L2C WRITE        ACCESS:         29 HIT:         29 MISS:          0
cpu0_L2C TRANSLATION  ACCESS:         81 HIT:          7 MISS:         74
cpu0_L2C PREFETCH REQUESTED:     579638 ISSUED:      24172 USEFUL:       5590 USELESS:        322
cpu0_L2C AVERAGE MISS LATENCY: 124.2 cycles
cpu0_L1I TOTAL        ACCESS:      21907 HIT:      21890 MISS:         17
cpu0_L1I LOAD         ACCESS:      21907 HIT:      21890 MISS:         17
cpu0_L1I RFO          ACCESS:          0 HIT:          0 MISS:          0
cpu0_L1I PREFETCH     ACCESS:          0 HIT:          0 MISS:          0
cpu0_L1I WRITE        ACCESS:          0 HIT:          0 MISS:          0
cpu0_L1I TRANSLATION  ACCESS:          0 HIT:          0 MISS:          0
cpu0_L1I PREFETCH REQUESTED:          0 ISSUED:          0 USEFUL:          0 USELESS:          0
cpu0_L1I AVERAGE MISS LATENCY: 137.4 cycles
[... 118 lines cut ...]
cpu0_DTLB LOAD         ACCESS:      46051 HIT:      40506 MISS:       5545
cpu0_DTLB RFO          ACCESS:          0 HIT:          0 MISS:          0
cpu0_DTLB PREFETCH     ACCESS:          0 HIT:          0 MISS:          0
cpu0_DTLB WRITE        ACCESS:          0 HIT:          0 MISS:          0
cpu0_DTLB TRANSLATION  ACCESS:          0 HIT:          0 MISS:          0
cpu0_DTLB PREFETCH REQUESTED:          0 ISSUED:          0 USEFUL:          0 USELESS:          0
cpu0_DTLB AVERAGE MISS LATENCY: 15.77 cycles
LLC TOTAL        ACCESS:      21224 HIT:         23 MISS:      21201
LLC LOAD         ACCESS:      21136 HIT:          0 MISS:      21136
LLC RFO          ACCESS:          1 HIT:          0 MISS:          1
LLC PREFETCH     ACCESS:          0 HIT:          0 MISS:          0
LLC WRITE        ACCESS:         19 HIT:         19 MISS:          0
LLC TRANSLATION  ACCESS:         68 HIT:          4 MISS:         64
LLC PREFETCH REQUESTED:          0 ISSUED:          0 USEFUL:          0 USELESS:          0
LLC AVERAGE MISS LATENCY: 138.2 cycles

DRAM Statistics

Channel 0 RQ ROW_BUFFER_HIT:      19544
  ROW_BUFFER_MISS:       1645
  AVG DBUS CONGESTED CYCLE: 3.003
Channel 0 WQ ROW_BUFFER_HIT:          0
  ROW_BUFFER_MISS:          0
  FULL:          0
Channel 0 REFRESHES ISSUED:         31

maccoymerrell, 2024-10-10:
Here's a quick look at how that time breaks down
Each sample counts as 0.01 seconds.
  %   cumulative   self              self     total
 time   seconds   seconds    calls  us/call  us/call  name
 72.59     12.63    12.63 29989252     0.42     0.43  va_ampm_lite::check_cl_access(champsim::address_slice<champsim::block_number_extent>)
 15.98     15.41     2.78  6448238     0.43     0.44  va_ampm_lite::check_cl_prefetch(champsim::address_slice<champsim::block_number_extent>)
  1.90     15.74     0.33  1848168     0.18     8.99  CACHE::operate()
  1.26     15.96     0.22    31584     6.97   501.84  va_ampm_lite::prefetcher_cache_operate(champsim::address_slice<champsim::static_extent<(champsim::data::bits)64, (champsim::data::bits)0> >, champsim::address_slice<champsim::static_extent<(champsim::data::bits)64, (champsim::data::bits)0> >, unsigned char, bool, access_type, unsigned int)
  1.21     16.17     0.21 36437490     0.01     0.01  std::pair<champsim::address_slice<champsim::page_number_extent>, champsim::address_slice<va_ampm_lite::block_in_page_extent> > va_ampm_lite::page_and_offset<champsim::address_slice<champsim::block_number_extent> >(champsim::address_slice<champsim::block_number_extent>)

maccoymerrell, 2024-10-10:
30M calls to check_cl_access() for 300k instructions?

ngober, 2024-10-12:
It's probably the result of [this find operation](https://github.com/ChampSim/ChampSim/blob/3b9e7c70678d972bb627c239f225c1bf83cb1d5e/prefetcher/va_ampm_lite/va_ampm_lite.cc#L18). The operation scans all regions each time the function is called, and it does it 4 times per cache access. This is probably expensive.

There are ways to cut down the scaling factor by reducing the number of searches, but you'd have to be sensitive to page crossings. Reducing the search to nonlinear complexity would require changing the data structure or the algorithm (such as sorting the region array by vpn).

maccoymerrell, 2024-10-12:
I can look into that.
I am confused as to why this wasn't an issue previously for AMPM. Seems like this is a recent behavior.

maccoymerrell, 2024-10-12:
This actually seems like a great application the lru table. Its not a perfect substitution, but it seems like it would be much more efficient to store regions instead of an unordered vector.

maccoymerrell, 2024-10-12:
@ngober Are you familiar with ChampSim's implementation of AMPM? By looking at it, it doesn't seem to update the LRU values inside each region. The original paper states we should be using LRU, but the implementation looks closer to just evicting the oldest entry, since it never updates the LRU value.

ngober, 2024-10-12:
I think you're right, that the replacement isn't doing LRU. The tests don't seem to cover that, they only cover the prefetch behavior in a single region.

closed 2024-10-17 by maccoymerrell