ChampSim/ChampSim #525: Hierarchy level implementation in TLB

Sanoj-S-Vijendra, 2024-10-22:
I want to implement certain functionalities on different levels of TLBs. Where should I make changes (like in which file)? I want to know in which files different levels of TLBs are handled.

Thanks

maccoymerrell, 2024-10-22:
The TLBs in ChampSim are defined in cache.cc/.h. All caches as well as the TLBs use the CACHE class in cache.h. If you want to make modifications, they would need to be done there.

Depending on what you are implementing, ChampSim modules can be used to substitute replacement policies and  prefetchers in each cache. Depending on the changes you plan to make, that may be a good place to start, since these can be assigned to specific caches (in your case the TLBs) in the config file.

Sanoj-S-Vijendra, 2024-10-22:
Yes, like I want to implement inclusiveness, exclusiveness etc. in different levels of TLB (similar to what present in cache levels). Then how should I start and what should be my approach for handling and changing cache.cc.

Thanks

ngober, 2024-10-22:
That's an excellent question. You should contribute to #350, we have been looking for someone to take on this project.