ChampSim/ChampSim #535: Cache miss latency misreporting

maccoymerrell, 2024-11-02:
**Describe the bug**
When reporting the miss latency of the cache, it is done in the following way:
1. Record when a MSHR type is created, this becomes the "time_enqueued" value
2. When merging, "time_enqueued" is set to the predecessor when the successor is a prefetch and the successor otherwise
3. Total miss latency cycles is the sum of the differences between cycle when the MSHR is closed and the time enqueued.
4. The average miss latency is calculated by dividing this value by ALL cache misses

**Expected behavior**
Average miss latency should report the average lifespan of an MSHR entry.
By allowing time_enqueued to be altered when multiple demands merge into a MSHR, it reduces the apparent cache latency.
By dividing by the total misses to the cache, we reduce the apparent miss latency by counting MSHR merges towards our total count.

**What are we trying to track?**
It seems to me that its better to track miss latency by recording the total cycles MSHR entries stay open and dividing it by the number of miss (non-writeback) packets sent downstream.
It also seems more correct to only update the time_enqueued value from the initial MSHR entry if a demand merges into a prefetch. Otherwise, to keep it unchanged. This _should_ make it so the reduction in cache latency from prefetching is apparent in the reporting, if there is a lot of demand->prefetch merging in the MSHR.

I have noticed that prefetches often repeatedly issue into open MSHRs, resulting in much higher "misses" in a cache than actual packets sent downstream. This results in our cache latency reporting at something ridiculously low (like 30 or 40 cycles) due to the over-report of misses. Without prefetching, the latency returns to reasonable values like 140-170 cycles. My suggested changes to how its calculated  resolve this.

ngober, 2024-11-02:
Ah, this makes sense. We're calculating the average latency in the stats printers with a denominator of all misses. That's almost certainly wrong.

It looks like our prefetching tests examine the `time_enqueued` member, but don't actually test the effect on the statistics. This is why I'm always so fixated on testing externally visible effects, rather than internal state. We should probably change [this line](https://github.com/ChampSim/ChampSim/blob/60af833777c459f8eadcedf955ee717af1ad66d3/test/cpp/src/431-merge-prefetch.cc#L101) or maybe remove it. There also could be something in test 425 to test that the denominator of the average (however we get there) does not increase.

maccoymerrell, 2024-11-02:
With my current changes on the pr, merging demands into prefetch retains this behavior so the test passes. It didn't seem wrong to do that, but I don't think its very meaningful since prefetches that never merge still contribute to average latency.

ngober, 2024-11-02:
What I'm saying is that 431 currently tests the claim "the MSHR's time_enqueued member is updated", when it should claim "the average latency is reduced". Then, 425 should test the claim "A prefetch that is never used does not cause the average latency to increase."

ngober, 2024-11-08:
Resolved by #536.

PR #536 (2024-11-05): Added downstream stats for cache, fixed cache miss latency calculations to use downstream.
Three things this pr covers:
1. Added downstream stats to the cache. This shows how many of each type of packet went downstream from said cache. This should always be <= misses, but does not count misses that hit in the mshr. It should match up with accesses to the downstream cache/dram.
2. Cache miss latency is now calculated as the total time each MSHR was open divided by the number of downstream packets (excluding writes) from that cache. If a demand merges into a prefetch, then right now it updates t [... cut]
closed 2024-11-08 by ngober