ChampSim/ChampSim #547: Configuration .json and Time Simulation

raffael-daltoe, 2024-11-15:
Hello,
I'm attempting to modify the number of cores in my champsim_config.json file. The default configuration specifies num_cores as 1, but I am trying to use 32. Essentially, I'm only changing the "num_cores" field in the JSON file.

Here’s the command I am running:
```
../tools/ChampSim/bin/champsim --warmup_instructions 50000000 --simulation_instructions 50000000 -- traces/400.perlbench-41B.champsimtrace.xz 
```

However, I encounter the following output:
```
WARNING: physical memory size is smaller than virtual memory size.
traces: At least 32 required but received 1
Run with --help for more information.
```

I have followed the guidelines provided in the [Champsim Configuration](https://champsim.github.io/ChampSim/master/Creating-a-configuration-file.html#multi-core-configurations) and also reviewed the details in this [Paper Champsim](https://arxiv.org/pdf/2210.14324). Despite these efforts, I haven't been able to achieve the desired result. Do you know how I can properly configure more cores for ChampSim simulations? I’d appreciate any guidance or tips you can provide.

Additionally, I wanted to ask if it's normal for the simulation of 400.perlbench-41B.champsimtrace.xz to take around 1 hour and 14 minutes with the default configuration. My last output was:
`Simulation complete CPU 0 instructions: 450000004 cycles: 228983896 cumulative IPC: 1.965 (Simulation time: 01 hr 14 min 30 sec)`

I'm using the default config.json provided in the repository for this test.

maccoymerrell, 2024-11-15:
A trace must be provided for each core in the simulation. That error is indicating that you only provided 1 of 32 traces.

That is not an uncommom simulation time. It will be even longer with 32 cores, just a fair warning.

raffael-daltoe, 2024-11-15:
So, is it not possible to configure 8, 16, or any other number of cores in the configuration file to simulate a single trace?
My goal is to represent the parallelism of a real architecture. I'm aiming to simulate how multiple cores would interact with a single workload to reflect realistic behavior. Is there a way to achieve this with ChampSim?

maccoymerrell, 2024-11-15:
It depends what you are looking for. If you just want to run the same trace on all cores, then you can specify 32 identical traces. If instead you are wanting to process scheduled across multiple cores, ChampSim does not currently have that feature, but we are always open to people contributing code. See #464

raffael-daltoe, 2024-11-21:
Yes, of course. Currently, I'm working with both ChampSim and Scarab, although I spend most of my time on ChampSim. For ChampSim, I developed a Python script to trace multiple traces by creating multiple ChampSim processes. If is interessed for you, is possible to find my progress on this in this repository:[Benchmarking](https://github.com/raffael-daltoe/Benchmarking)

Once I finish my research, I plan to focus on improving ChampSim's multi-threading processing capabilities.

closed 2024-11-25 by ngober

xzx0805, 2025-01-06:
Hi, What kind of CPU do you use? My simulation takes several hours on a SPEC17 trace like (623.xalancbmk_s-10B.champsimtrace_2.xz) to run. Could it be that my CPU is outdated？ @raffael-daltoe @maccoymerrell

CugPaul, 2025-01-06:
> 000004

I also simulated 500 million instructions.

raffael-daltoe, 2025-01-06:
Normally I use my I9 14900HX, what's your CPU?

CugPaul, 2025-01-06:
My CPU is an Intel Xeon Silver 4208, which is a server-grade CPU from a few years ago. Is your CPU from a laptop? 
According to single-thread benchmark scores, my CPU is far behind the Intel i9-14900HX. Could the performance gap between CPUs cause such a large time difference? I'm not sure about it.

raffael-daltoe, 2025-01-06:
> My CPU is an Intel Xeon Silver 4208, which is a server-grade CPU from a few years ago. Is your CPU from a laptop? According to single-thread benchmark scores, my CPU is far behind the Intel i9-14900HX. Could the performance gap between CPUs cause such a large time difference? I'm not sure about it.

Yes, is from my laptop, depending on your processor, the simulations will be faster or not, it may involve other things such as main memory (RAM).
This is my opinion, I'm not a dev from ChampSim, but if you have a latest processor of market is expected to have lowest times of simulation

CugPaul, 2025-01-06:
Appreciate your help!

> Yes, is from my laptop, depending on your processor, the simulations will be faster or not, it may involve other things such as main memory (RAM). This is my opinion, I'm not a dev from ChampSim, [...]

TripleS1404, 2026-02-18:
> Hi, What kind of CPU do you use? My simulation takes several hours on a SPEC17 trace like (623.xalancbmk_s-10B.champsimtrace_2.xz) to run. Could it be that my CPU is outdated？ [@raffael-daltoe](https://github.com/raffael-daltoe) [@maccoymerrell](https://github.com/maccoymerrell)

Hey, May I know how long does it take to run entire Spec2017 traces? I am only collecting end of simulation stats for entire spec2017 but it is taking more than 2 days. I am running for 100M simulation instructions and 10M warmup.