ChampSim/ChampSim #562: Issue with compiling the documentation of ChampSim.

satanu01, 2024-11-30:
Followed each instruction in `doc/` directory, but `sphinx-build -c . src _build/html` giving the following error. Pls help! 

```
$ sphinx-build -c . src _build/html
Running Sphinx v1.8.5

Configuration error:
There is a programmable error in your configuration file:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/sphinx/config.py", line 368, in eval_config_file
    execfile_(filename, namespace)
  File "/usr/lib/python3/dist-packages/sphinx/util/pycompat.py", line 150, in execfile_
    exec_(code, _globals)
  File "/mnt/c/Users/satan/Desktop/Research/Simulator/ChampSim/docs/conf.py", line 67, in <module>
    @functools.cache
AttributeError: module 'functools' has no attribute 'cache'
```

ngober, 2024-11-30:
What is your Python version? That Sphinx looks very old.

The problem here probably comes from the fact that the requirements file doesn't list any version numbers. Would you please checkout `bugfix/sphinx-version-number` and see if that resolves your issue?

satanu01, 2024-11-30:
Hurray… Thanks for your support @ngober. Finally, we successfully compiled the doc file. Actually, it needed the following versions.
- Python 3.8
- Sphinx<=1.8.0
- jinja2==2.5
- markupsafe==0.23
- breathe<4.13.0

Also need to rename the `index.rst` to `contents.rst` and change `@functools.cache` to `@functools.lru_cache(None)`.

ngober, 2024-11-30:
Did you checkout the branch that I mentioned? It already has most of the fixes you mentioned.

satanu01, 2024-12-01:
No sir.

ngober, 2024-12-01:
Would you please checkout that branch and see if it resolves your issues? The changes that I made were minimal, and I want to see if they were sufficient. For example, the ChampSim documentation does not directly require Jinja2, so I would like to keep any requirements on it transitory.