ChampSim/ChampSim #573: Register Allocation Deadlock

ChristopherThomasHill, 2025-01-07:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Compiles ChampSim with `"register_file_size":128`
2. run x264_s-39B.champsimtrace.xz from [https://dpc3.compas.cs.stonybrook.edu/champsim-traces/speccpu/](https://dpc3.compas.cs.stonybrook.edu/champsim-traces/speccpu/)
3. See deadlock

**Expected behavior**
Completes without deadlock.

**Output**
```
Frontend Register Allocation Table        Backend Register Allocation Table
Arch reg:   0    Phys reg:  -1            Arch reg:   0    Phys reg:  -1
Arch reg:   1    Phys reg:  -1            Arch reg:   1    Phys reg:  -1
Arch reg:   2    Phys reg:  -1            Arch reg:   2    Phys reg:  -1
Arch reg:   3    Phys reg: 109            Arch reg:   3    Phys reg: 109
Arch reg:   4    Phys reg:  61            Arch reg:   4    Phys reg:  61
Arch reg:   5    Phys reg:  30            Arch reg:   5    Phys reg:  30
Arch reg:   6    Phys reg:  36            Arch reg:   6    Phys reg:  36
Arch reg:   7    Phys reg:  37            Arch reg:   7    Phys reg:  37
Arch reg:   8    Phys reg:  32            Arch reg:   8    Phys reg:  32
Arch reg:   9    Phys reg: 101            Arch reg:   9    Phys reg: 101
Arch reg:  10    Phys reg:   5            Arch reg:  10    Phys reg:   5
Arch reg:  11    Phys reg:  95            Arch reg:  11    Phys reg:  95
Arch reg:  12    Phys reg:  82            Arch reg:  12    Phys reg:  82
Arch reg:  13    Phys reg:  85            Arch reg:  13    Phys reg:  85
Arch reg:  14    Phys reg:  39            Arch reg:  14    Phys reg:  39
Arch reg:  15    Phys reg: 104            Arch reg:  15    Phys reg: 104
Arch reg:  16    Phys reg:  74            Arch reg:  16    Phys reg:  74
Arch reg:  17    Phys reg:  45            Arch reg:  17    Phys reg:  45
Arch reg:  18    Phys reg:  94            Arch reg:  18    Phys reg:  94
Arch reg:  19    Phys reg:  -1            Arch reg:  19    Phys reg:  -1
Arch reg:  20    Phys reg:  -1            Arch reg:  20    Phys reg:  -1
Arch reg:  21    Phys reg:  -1            Arch reg:  21    Phys reg:  -1
Arch reg:  22    Phys reg:  -1            Arch reg:  22    Phys reg:  -1
Arch reg:  23    Phys reg:  51            Arch reg:  23    Phys reg:  -1
Arch reg:  24    Phys reg:  -1            Arch reg:  24    Phys reg:  -1
Arch reg:  25    Phys reg:  27            Arch reg:  25    Phys reg:  27
Arch reg:  26    Phys reg:  25            Arch reg:  26    Phys reg:  25
Arch reg:  27    Phys reg: 118            Arch reg:  27    Phys reg: 118
Arch reg:  28    Phys reg:  -1            Arch reg:  28    Phys reg:  -1
Arch reg:  29    Phys reg:  76            Arch reg:  29    Phys reg:  76
Arch reg:  30    Phys reg: 122            Arch reg:  30    Phys reg: 122
Arch reg:  31    Phys reg:  -1            Arch reg:  31    Phys reg:  -1
Arch reg:  32    Phys reg:  88            Arch reg:  32    Phys reg:  88
Arch reg:  33    Phys reg: 107            Arch reg:  33    Phys reg: 107
Arch reg:  34    Phys reg:  -1            Arch reg:  34    Phys reg:  -1
Arch reg:  35    Phys reg:  97            Arch reg:  35    Phys reg:  97
Arch reg:  36    Phys reg:  43            Arch reg:  36    Phys reg:  43
Arch reg:  37    Phys reg:  -1            Arch reg:  37    Phys reg:  -1
Arch reg:  38    Phys reg:  99            Arch reg:  38    Phys reg:  -1
Arch reg:  39    Phys reg:  60            Arch reg:  39    Phys reg:  60
Arch reg:  40    Phys reg:  49            Arch reg:  40    Phys reg:  49
Arch reg:  41    Phys reg:  19            Arch reg:  41    Phys reg:  19
Arch reg:  42    Phys reg:  -1            Arch reg:  42    Phys reg:  -1
Arch reg:  43    Phys reg:  -1            Arch reg:  43    Phys reg:  -1
Arch reg:  44    Phys reg:  -1            Arch reg:  44    Phys reg:  -1
Arch reg:  45    Phys reg:  17            Arch reg:  45    Phys reg:  17
Arch reg:  46    Phys reg: 103            Arch reg:  46    Phys reg: 103
Arch reg:  47    Phys reg:  14            Arch reg:  47    Phys reg:  14
Arch reg:  48    Phys reg:  35            Arch reg:  48    Phys reg:  35
Arch reg:  49    Phys reg: 125            Arch reg:  49    Phys reg: 125
Arch reg:  50    Phys reg:  92            Arch reg:  50    Phys reg:  92
Arch reg:  51    Phys reg:  -1            Arch reg:  51    Phys reg:  -1
Arch reg:  52    Phys reg:  -1            Arch reg:  52    Phys reg:  -1
Arch reg:  53    Phys reg:  69            Arch reg:  53    Phys reg:  69
Arch reg:  54    Phys reg:  41            Arch reg:  54    Phys reg:  41
[... 315 lines cut ...]
Phys reg: 112	 Arch reg: 102	 Producer: 117296661	 Valid: true	 Busy: true
Phys reg: 113	 Arch reg: 115	 Producer: 0	 Valid: true	 Busy: true
Phys reg: 114	 Arch reg: 106	 Producer: 116960950	 Valid: true	 Busy: true
Phys reg: 115	 Arch reg:  76	 Producer: 0	 Valid: true	 Busy: true
Phys reg: 116	 Arch reg:  40	 Producer: 0	 Valid: true	 Busy: true
Phys reg: 117	 Arch reg:  12	 Producer: 0	 Valid: true	 Busy: true
Phys reg: 118	 Arch reg:  27	 Producer: 117298695	 Valid: true	 Busy: true
Phys reg: 119	 Arch reg:  68	 Producer: 114020045	 Valid: true	 Busy: true
Phys reg: 120	 Arch reg: 114	 Producer: 116961019	 Valid: true	 Busy: true
Phys reg: 121	 Arch reg: 101	 Producer: 117298717	 Valid: true	 Busy: true
Phys reg: 122	 Arch reg:  30	 Producer: 117298240	 Valid: true	 Busy: true
Phys reg: 123	 Arch reg:  29	 Producer: 0	 Valid: true	 Busy: true
Phys reg: 124	 Arch reg: 113	 Producer: 114737688	 Valid: true	 Busy: true
Phys reg: 125	 Arch reg:  49	 Producer: 117296652	 Valid: true	 Busy: true
Phys reg: 126	 Arch reg: 112	 Producer: 116961018	 Valid: true	 Busy: true
Phys reg: 127	 Arch reg:  76	 Producer: 117298702	 Valid: true	 Busy: true
```

**Desktop (please complete the following information):**
- OS: Ubuntu 20.04
- Compiler GCC
- Version 11]

**Additional context**
In `rename_src_register()`, there is a section of code that allocates source registers that haven't been allocated already. However, these registers aren't ever freed when another physical address is assigned to that architectural register.

ChristopherThomasHill, 2025-01-07:
This simple addition to the backend_RAT resolves the deadlock.

```
if (phys < 0) {
    // allocate the register if it hasn't yet been mapped
    // (common due to the traces being slices in the middle of a program)
    phys = free_registers.front();
    free_registers.pop();
    frontend_RAT[reg] = phys;
    backend_RAT[reg] = phys; // allows freeing of this register
    physical_register_file.at(phys) = {(uint16_t)reg, 0, true, true}; // arch_reg_index, producing_inst_id, valid, busy

    printf("Unknown Allocation %u %u\n", phys, reg);
  }
```

joshualmashburn, 2025-01-07:
Here is how the register rename mechanism works:
Registers are renamed at schedule. When a register-writing instruction completes execution, the destination physical register is marked Valid, signalling to waiting instructions that that operand is ready.
However, we don't recycle the previous renamed register (that corresponds to the architectural register) until this write instruction retires (*in order*), to keep a consistent architectural state and to leave open the possibility of adding the ability to roll back speculative state. Your suggestion would free the current renamed register as soon as the writing instruction retires.

As for the "-1", it's simply a marker to show that according to the frontend (speculative) or backend states, there is no physical register alias for that architectural register yet. As seen in retire_dest_register(), the "-1" is simply replaced with no side effects, because there never was a register to free. (While the accuracy may be subject to debate, ChampSim starts with an empty physical register file, and allocates registers on either first reference (read/write) or upon write)

I believe this issue may be unrelated to the register file. A full PRF could also accompany a stuck head of ROB or something stuck in the LSQs. Could you please share your command line arguments and the complete deadlock message?

joshualmashburn, 2025-01-07:
Also, can you share any other configuration parameters you have changed? Is the ROB size the default?

ChristopherThomasHill, 2025-01-07:
Sorry if I'm confused. I don't quite understand why my fix isn't a bug. The fix I propose is a single line within `rename_src_register()` for the case where you have to allocate a source register that doesn't currently exist. This allocate physical register is never freed up from what it appears. 

The only place I see a physical register being freed is in 

```
PHYSICAL_REGISTER_ID old_phys_reg = backend_RAT[arch_reg];

  // update the backend RAT with the new phys reg
  backend_RAT[arch_reg] = physreg;

  // free the old phys reg
  if (old_phys_reg != -1) {
    free_register(old_phys_reg);
  }
```

Prior to my code, the backend_RAT mapped to "-1" which means that no physical register would be freed (even though it has one mapped to it from the source register).

ChristopherThomasHill, 2025-01-07:
Here is the full json and command line

/champsim --warmup-instructions 100000000 --simulation-instructions 100000000 x264_s-39B.champsimtrace.xz

{
    "executable_name": "champsim",
    "block_size": 64,
    "page_size": 2097152,
    "heartbeat_frequency": 10000000,
    "num_cores": 1,

    "ooo_cpu": [
        {
            "frequency": 4000,
            "ifetch_buffer_size":64,
            "decode_buffer_size":32,
            "dispatch_buffer_size":32,
            "register_file_size":128,
            "rob_size": 352,
            "lq_size": 128,
            "sq_size": 72,
            "fetch_width": 6,
            "decode_width": 6,
            "dispatch_width": 6,
            "execute_width": 4,
            "lq_width": 2,
            "sq_width": 2,
            "retire_width": 4,
            "mispredict_penalty": 20,
            "scheduler_size": 128,
            "decode_latency": 1,
            "dispatch_latency": 1,
            "schedule_latency": 0,
            "execute_latency": 0,
            "branch_predictor": "perceptron",
            "btb": "basic_btb"
        }
    ],

    "DIB": {
        "window_size": 16,
        "sets": 32,
        "ways": 8
    },

    "L1I": {
        "sets": 64,
        "ways": 8,
        "rq_size": 64,
        "wq_size": 64,
        "pq_size": 32,
        "mshr_size": 8,
        "latency": 4,
        "max_tag_check": 2,
        "max_fill": 2,
        "prefetch_as_load": false,
        "virtual_prefetch": true,
        "prefetch_activate": "LOAD,PREFETCH",
        "prefetcher": "no"
    },

    "L1D": {
        "sets": 64,
        "ways": 8,
        "rq_size": 64,
        "wq_size": 64,
        "pq_size": 8,
        "mshr_size": 16,
        "latency": 4,
        "max_tag_check": 2,
        "max_fill": 2,
        "prefetch_as_load": false,
        "virtual_prefetch": false,
        "prefetch_activate": "LOAD,PREFETCH",
        "prefetcher": "no"
    },

    "L2C": {
        "sets": 512,
        "ways": 8,
        "rq_size": 32,
        "wq_size": 32,
        "pq_size": 16,
        "mshr_size": 32,
        "latency": 8,
        "max_tag_check": 1,
        "max_fill": 1,
        "prefetch_as_load": false,
        "virtual_prefetch": false,
        "prefetch_activate": "LOAD,PREFETCH",
        "prefetcher": "no"
    },

    "ITLB": {
        "sets": 16,
        "ways": 4,
        "rq_size": 16,
        "wq_size": 16,
        "pq_size": 0,
        "mshr_size": 8,
        "latency": 1,
        "max_tag_check": 2,
        "max_fill": 2,
        "prefetch_as_load": false
    },

    "DTLB": {
        "sets": 16,
        "ways": 4,
        "rq_size": 16,
        "wq_size": 16,
        "pq_size": 0,
        "mshr_size": 8,
        "latency": 1,
        "max_tag_check": 2,
        "max_fill": 2,
        "prefetch_as_load": false
    },

    "STLB": {
        "sets": 128,
        "ways": 12,
        "rq_size": 32,
        "wq_size": 32,
        "pq_size": 0,
        "mshr_size": 16,
        "latency": 8,
        "max_tag_check": 1,
        "max_fill": 1,
        "prefetch_as_load": false
    },

	"PTW": {
		"pscl4_set": 1,
		"pscl4_way": 4,
		"pscl3_set": 2,
		"pscl3_way": 4,
		"pscl2_set": 4,
		"pscl2_way": 8,
		"rq_size": 16,
		"mshr_size": 5,
		"max_read": 2,
		"max_write": 2
	},

    "LLC": {
        "frequency": 4000,
        "sets": 2048,
        "ways": 16,
        "rq_size": 32,
        "wq_size": 32,
        "pq_size": 32,
        "mshr_size": 64,
        "latency": 20,
        "max_tag_check": 1,
        "max_fill": 1,
        "prefetch_as_load": false,
        "virtual_prefetch": false,
        "prefetch_activate": "LOAD,PREFETCH",
        "prefetcher": "no",
        "replacement": "lru"
    },

    "physical_memory": {
        "data_rate": 2400,
        "channels": 1,
        "ranks": 1,
        "bankgroups": 8,
        "banks": 4,
        "bank_rows": 65536,
        "bank_columns": 1024,
        "channel_width": 8,
        "wq_size": 64,
        "rq_size": 64,
        "tCAS":  16,
        "tRCD": 16,
        "tRP": 16,
        "tRAS": 39,
        "refresh_period": 32,
        "refreshes_per_period": 8192
    },

    "virtual_memory": {
        "pte_page_size": 4096,
        "num_levels": 4,
        "minor_fault_penalty": 200,
        "randomization": 1
    }
}

ChristopherThomasHill, 2025-01-07:
I added the full deadlock error.

The important thing to note is that the instruction in entry 0 of the ROB hasn't been scheduled yet. From my understanding, this can be attributed to these checks. I observed the first if condition is the one failing since the deadlock file shows that the other conditions are met. From the deadlock file we can also observe that there is currently only 1 free register, likely attributing the cause to that. We can also see that multiple physical registers are valid and mapped to the same architectural register even though nothing in the ROB is between scheduled and retired.

```
if (reg_allocator.count_free_registers() < (sources_to_allocate + rob_it->destination_registers.size())) {
  break;
}
if (!rob_it->scheduled && rob_it->ready_time <= current_time) {
  do_scheduling(*rob_it);
  ++progress;
}
```

Full Deadlock Error
```
DEADLOCK! CPU 0 cycle 57650048
cpu0_IFETCH empty

cpu0_DECODE empty

cpu0_DISPATCH empty

[cpu0_ROB] entry:   0 instr_id: 117298798 fetch_issued: true fetch_completed: true scheduled: false executed: false completed: false num_reg_dependent: 0 num_mem_ops: 0 event: 57649082
[cpu0_ROB] entry:   1 instr_id: 117298799 fetch_issued: true fetch_completed: true scheduled: false executed: false completed: false num_reg_dependent: 0 num_mem_ops: 0 event: 57649093
[cpu0_ROB] entry:   2 instr_id: 117298800 fetch_issued: true fetch_completed: true scheduled: false executed: false completed: false num_reg_dependent: 0 num_mem_ops: 0 event: 57649093
[cpu0_ROB] entry:   3 instr_id: 117298801 fetch_issued: true fetch_completed: true scheduled: false executed: false completed: false num_reg_dependent: 0 num_mem_ops: 0 event: 57649093
[cpu0_ROB] entry:   4 instr_id: 117298802 fetch_issued: true fetch_completed: true scheduled: false executed: false completed: false num_reg_dependent: 0 num_mem_ops: 0 event: 57649093
[cpu0_ROB] entry:   5 instr_id: 117298803 fetch_issued: true fetch_completed: true scheduled: false executed: false completed: false num_reg_dependent: 0 num_mem_ops: 1 event: 57649093
[cpu0_ROB] entry:   6 instr_id: 117298804 fetch_issued: true fetch_completed: true scheduled: false executed: false completed: false num_reg_dependent: 0 num_mem_ops: 1 event: 57649094
[cpu0_ROB] entry:   7 instr_id: 117298805 fetch_issued: true fetch_completed: true scheduled: false executed: false completed: false num_reg_dependent: 0 num_mem_ops: 0 event: 57649094
[cpu0_ROB] entry:   8 instr_id: 117298806 fetch_issued: true fetch_completed: true scheduled: false executed: false completed: false num_reg_dependent: 0 num_mem_ops: 0 event: 57649094
[cpu0_ROB] entry:   9 instr_id: 117298807 fetch_issued: true fetch_completed: true scheduled: false executed: false completed: false num_reg_dependent: 0 num_mem_ops: 0 event: 57649094
[cpu0_ROB] entry:  10 instr_id: 117298808 fetch_issued: true fetch_completed: true scheduled: false executed: false completed: false num_reg_dependent: 0 num_mem_ops: 1 event: 57649094
[cpu0_ROB] entry:  11 instr_id: 117298809 fetch_issued: true fetch_completed: true scheduled: false executed: false completed: false num_reg_dependent: 0 num_mem_ops: 0 event: 57649095

Frontend Register Allocation Table        Backend Register Allocation Table
Arch reg:   0    Phys reg:  -1            Arch reg:   0    Phys reg:  -1
Arch reg:   1    Phys reg:  -1            Arch reg:   1    Phys reg:  -1
Arch reg:   2    Phys reg:  -1            Arch reg:   2    Phys reg:  -1
Arch reg:   3    Phys reg: 109            Arch reg:   3    Phys reg: 109
Arch reg:   4    Phys reg:  61            Arch reg:   4    Phys reg:  61
Arch reg:   5    Phys reg:  30            Arch reg:   5    Phys reg:  30
Arch reg:   6    Phys reg:  36            Arch reg:   6    Phys reg:  36
Arch reg:   7    Phys reg:  37            Arch reg:   7    Phys reg:  37
Arch reg:   8    Phys reg:  32            Arch reg:   8    Phys reg:  32
Arch reg:   9    Phys reg: 101            Arch reg:   9    Phys reg: 101
Arch reg:  10    Phys reg:   5            Arch reg:  10    Phys reg:   5
Arch reg:  11    Phys reg:  95            Arch reg:  11    Phys reg:  95
Arch reg:  12    Phys reg:  82            Arch reg:  12    Phys reg:  82
Arch reg:  13    Phys reg:  85            Arch reg:  13    Phys reg:  85
Arch reg:  14    Phys reg:  39            Arch reg:  14    Phys reg:  39
Arch reg:  15    Phys reg: 104            Arch reg:  15    Phys reg: 104
Arch reg:  16    Phys reg:  74            Arch reg:  16    Phys reg:  74
Arch reg:  17    Phys reg:  45            Arch reg:  17    Phys reg:  45
Arch reg:  18    Phys reg:  94            Arch reg:  18    Phys reg:  94
Arch reg:  19    Phys reg:  -1            Arch reg:  19    Phys reg:  -1
Arch reg:  20    Phys reg:  -1            Arch reg:  20    Phys reg:  -1
Arch reg:  21    Phys reg:  -1            Arch reg:  21    Phys reg:  -1
Arch reg:  22    Phys reg:  -1            Arch reg:  22    Phys reg:  -1
Arch reg:  23    Phys reg:  51            Arch reg:  23    Phys reg:  -1
Arch reg:  24    Phys reg:  -1            Arch reg:  24    Phys reg:  -1
Arch reg:  25    Phys reg:  27            Arch reg:  25    Phys reg:  27
Arch reg:  26    Phys reg:  25            Arch reg:  26    Phys reg:  25
Arch reg:  27    Phys reg: 118            Arch reg:  27    Phys reg: 118
Arch reg:  28    Phys reg:  -1            Arch reg:  28    Phys reg:  -1
Arch reg:  29    Phys reg:  76            Arch reg:  29    Phys reg:  76
[... cut ...]
[WQ] entry:  41 empty
[WQ] entry:  42 empty
[WQ] entry:  43 empty
[WQ] entry:  44 empty
[WQ] entry:  45 empty
[WQ] entry:  46 empty
[WQ] entry:  47 empty
[WQ] entry:  48 empty
[WQ] entry:  49 empty
[WQ] entry:  50 empty
[WQ] entry:  51 empty
[WQ] entry:  52 empty
[WQ] entry:  53 empty
[WQ] entry:  54 empty
[WQ] entry:  55 empty
[WQ] entry:  56 empty
[WQ] entry:  57 empty
[WQ] entry:  58 empty
[WQ] entry:  59 empty
[WQ] entry:  60 empty
[WQ] entry:  61 empty
[WQ] entry:  62 empty
[WQ] entry:  63 empty
```

joshualmashburn, 2025-01-07:
Ah yep I see it now! I thought I covered both reads and writes to previously unallocated registers in testing, but I'll have to go back and add some. Good catch.

joshualmashburn, 2025-01-07:
Still getting the same deadlock with the fix. While this does correct the bug @ChristopherThomasHill identified, I'll have to do some deeper debug to diagnose this one.

joshualmashburn, 2025-01-07:
Should be fixed now. In the period of interest (1.1M-1.2M), IPC plummets for a while:
[image]

I'd have to step through the simulator to see exactly what the trace is doing here, but it's possible that this particular trace is just that unfriendly to instruction-level-parallelism in certain blocks.

ngober, 2025-01-09:
Resolved with #574.

closed 2025-01-09 by ngober