ChampSim/ChampSim #579: I have configured the champsim, through with I am able to generate number of writes and reads per core, I want number of write and read per core per epoch (as per 1 million instructino), how can I get to know this?

Abdul8081, 2025-01-10:
### Discussed in https://github.com/ChampSim/ChampSim/discussions/578

<div type='discussions-op-text'>

<sup>Originally posted by **Abdul8081** January 10, 2025</sup>
Qn 1 : How will I know as how many instruction is coming to each core, If I am simulating champsim for 10 million simulation instruciton for the 4 core?

Qn 2 : How can I seperate out instruction comming at one core and divide it 4 equal parts? and then, how can I generate the result for like number of writes and read per core per divided part?

Any help will be very usefull for me.

My generated result
[image]

I wanted to modify in this, in which I want to divide numer of instrction coming out in a particular core and then I want to divide that number of intruction into some part for example 4 parts, then I wanted number of writes as per parts for that core. and I want this for all 4 core.

Please Help me to find out this.
</div>

maccoymerrell, 2025-01-10:
This is something you will need to program yourself. There are a couple of ways you could do this.
**The easiest would be to print out each core's total number of reads and writes at each epoch, and then calculate the reads and writes within each epoch afterwards.**

Another way would be to introduce additional phases to ChampSim's simulation, but I am not super familiar with how that would work.

Abdul8081, 2025-01-10:
> This is something you will need to program yourself. There are a couple of ways you could do this. **The easiest would be to print out each core's total number of reads and writes at each epoch, and [...]

How I will print out each cores's total number of reads and writes at each epoch? how would I know how much instruction is coming a particular core? and how to divide that in the epoch, like where to do changes, in the cache.h file, cache.cc file or any other file of the champsim?

ngober, 2025-01-13:
There is currently no provision to do this. The only strategy that I can think of is to run ChampSim many times with increasing lengths and noting the difference between the statistics.

We have in the past discussed implementing "phases" into the execution of ChampSim, but there has never been a champion for the feature. If you are interested in implementing it into the simulator, I would be happy to point you in the right direction.

Abdul8081, 2025-01-13:
I want to implement it in the champsim, please tell me the approach

Abdul8081, 2025-01-15:
> The main loop is driven by a vector of phase information defined here
> Currently, there are only two phases: warmup and simulation. In theory, it
> could be modified to perform more phases than that. The challenge, as I
> have seen it, is to design a useful interface to such functionality.

Thankyou, I will check now, if I can do somethings

ngober, 2026-02-14:
I am closing this since no progress has been made in the past year.

closed 2026-02-14 by ngober