ChampSim/ChampSim #580: Store buffer simulation of Champsim

gh1403, 2025-01-10:
**Describe the bug**

**_This might not be a bug but rather a feature request._**

Hi, I am studying the core-cache interaction part of ChampSim. Currently, the `inflight_writes` in the `CACHE` structure seems to simulate the functionality of a store buffer. However, it appears to function quite independently. For example, In the `try_hit` function, it should first check whether the target cache line exists in the store buffer but this step seems to be skipped. Additionally, the store buffer doesn't seem to have a size limit (typically 56 entries in some Skylake architecture products). Are these intended behaviors of `inflight_writes`? Any insights would be appreciated. Thank you!

**To Reproduce**

**Expected behavior**

**Output**

**Desktop (please complete the following information):**
 - OS: 
 - Compiler 
 - Version [both dev and master branch]

**Additional context**
Add any other context about the problem here.

maccoymerrell, 2025-01-10:
Forgive me if I am misunderstanding you, but the functionality you seem to be describing is the load-store queue (LSQ) that is inside the core model rather than the cache.

gh1403, 2025-01-10:
Hi,
I’m confused about which component in processors the `CACHE::inflight_writes` structure is trying to simulate. Currently, it appears to handle pending store requests independently rather than using `MSHR`. Do real-world processors typically work this way?
https://github.com/ChampSim/ChampSim/blob/f7f919c41721e0adb3bbcfece55b5715461b7cea/src/cache.cc#L384
Thanks!

maccoymerrell, 2025-01-10:
I believe this is a caveat of how the cache implementation works.

Please others correct me if I am wrong about this:
The inflight writes queue is reserved specifically for writebacks of cache lines from upper-level caches, rather than demand writes (stores) from the core.
Since writebacks of cache lines are much larger operations than standard stores, there is a latency associated with it (in this case the same latency as a fill) before that data can become available after arriving at the cache, hence the need for a tracking data structure.
The MSHR handles traffic coming from below the cache, while the inflight-writes handles traffic coming from above the cache.

Regular stores are indeed handled by the MSHR.

Hope this helps.

maccoymerrell, 2025-01-10:
As to whether real processor caches work this way, I would think of this queue as less of a real hardware structure and more of a time-keeping device so ChampSim knows when that particular data has arrived and finished writing back. A more complex cache model would probably have these writes queue up at particular cache bank and take FILL_LATENCY time before the bank is free again. Since our current cache model doesn't model banks, we are really only making sure that reads can't be returned from that block until the writeback has been filled.

gh1403, 2025-01-10:
Thanks for your clarification. I'll further check this. One more question, does champsim implement any kind of support for store buffer simulation? As I understand it, the store queue in the LSQ and the store buffer are distinct components. I am not very familiar with the cpu-side implementation in champsim but it seems that there is no component in that part serving a similar function to the store buffer.

maccoymerrell, 2025-01-10:
I am not sure what a store buffer would be doing differently than the LSQ in the CPU.

The LSQ does store-forwarding, where if we issue a load to a pending store in the LSQ, we forward that store data to the load instead of going to the caches for the load data.
That seems to be doing what you describe the store buffer doing.

maccoymerrell, 2025-01-10:
I think the confusion here is that ChampSim doesn't have any coherence/shared memory system in place, meaning that things in the LSQ are ready to release to memory directly. There is no intermediate buffering between the cache and the LSQ.

maccoymerrell, 2025-01-10:
From looking at our code in the core more, we are probably modeling an infinite store buffer,  where no LSQ entries are prevented from being issued to the L1D due to a pending store. We don't wait for a response from the cache that a store is complete, and we don't cap the number of store misses that we have oustanding, so that seems like it would mimic a store buffer of infinite size.

maccoymerrell, 2025-01-13:
@gh1403 I would encourage you that if you are interested in modeling this, and you want to use ChampSim, authoring a pull request and adding it to ChampSim yourself. We are always looking for contributions. I don't think this is too difficult to add using ChampSim's existing infrastructure. Since the performance overheads may be relevant for some workloads (and apparently the workloads you are interested in), I think this would be wise if you want to use ChampSim for your work.

Without support for things like shared memory, I am not sure if the performance differences would be significant or not, and whether you could pursue your own research using ChampSim.

gh1403, 2025-01-13:
Thanks for your suggestion! I'll look into it and keep you updated if I make any progress :)

closed 2026-02-14 by ngober