ChampSim/ChampSim #596: Replacement Policy Not Changing: Only LRU Being Applied

ksharat45, 2025-02-18:
### **_Description:_**

I changed the replacement policy in ChampSim from LRU to DRRIP, but the simulator still applies LRU to the LLC. No matter which replacement policy I set in config.json, it defaults to LRU.

### **_Reproduce:_**
Steps to reproduce the behavior:

1. Changed  `config.json` to set the replacement policy to DRRIP (or any other policy).
2. Ran make clean followed by make to ensure a fresh build.
3. Executed ChampSim with a trace.
4. Observed that the results remain the same as LRU, regardless of the selected replacement policy.

maccoymerrell, 2025-02-19:
Just to be clear, you are doing the following steps?
1. Modifying config.json
2. ./config.sh config.json
3. Make

ksharat45, 2025-02-19:
I followed the mentioned steps everytime.

While I am debugging, I just changed the default policy to drrip in the file parse.py. Then I just figured out that the drrip is being called but still the stats are same even when I changed the LLC policy to lru. I added stats in paste bin, I provided the link below
```

 ({ 'name': k,
                # Mark queues that need to match full addresses on collision
               '_queue_check_full_addr': cache.get('_first_level', False) or cache.get('wq_check_full_addr', False),

                # Get module path names and unique module names
               _replacement_data': list(map(replacement_parse, util.wrap_list(cache.get('replacement', 'drrip')))),
               '_prefetcher_data': [*map(functools.partial(prefetcher_parse, cache=cache), util.wrap_list(cache.get('prefetcher', 'no')))]
            } for k,cache in caches.items())
        )

```
Paste bin: https://pastebin.com/GAUxBkAx

closed 2025-02-26 by ksharat45