ChampSim/ChampSim #597: Snapshot/Checkpoint Functionality After Warmup

Vvamos12, 2025-02-20:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

I often find it time-consuming to re-run the warmup phase every time I need to experiment with different simulation parameters. The current workflow forces me to repeat the warmup cycle, which delays testing and debugging efforts.
**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

maccoymerrell, 2025-02-20:
ChampSim has a lot of state-containing components, including modules like replacement policies, prefetchers, and branch predictors that would all need to be saved along with the caches, vmem, and core-state.
I have given some thought to this feature before, and we would need the following infrastructure in place:
1. A base class that all state-containing classes of ChampSim could derive from (say class checkpointable)
2. A checkpoint file format. There are likely tools/libraries that already exist that could be used to generate these.
3. ChampSim components would need to write and read their checkpoint data when prompted to (via the base class interface). This will take the most work and debugging. We would need to ensure we call in the exact-same order for reads and writes, or have a good way of determining whose data goes where when resuming, which I don't think we have a good way of doing this in ChampSim yet (for example, duplicate modules like LRU in the caches are difficult to distinguish from each other).
4. Safety checks in place to ensure that the binary is the same as the one that generated the checkpoint originally when loading from a checkpoint.

This is a lot of work, so I don't think this feature will come anytime soon.
There are some PRs in the pipeline (like the component class) which may make this easier to implement in time.

ngober, 2025-02-20:
Maccoy is absolutely right about the challenges of checkpointing. I would put solutions to this problem into two categories: lossy and lossless.

### Lossy

There are a lot of expedient ways to create a state dump. Take the example of a cache: you could create a JSON file that represents the blocks, the MSHRs, and maybe the inflight checks (including the ones waiting for translation). This could then be loaded back in and the behavior can be mostly relaunched.

However, as noted above, this is not the full representation of the cache state. Notably, the prefetch and replacement are user-submitted modules that do not have a trivial state to be dumped. Surely, there is a module interface that could be conceived, but it still relies on the user to care about this. There simply isn't a uniform way to dump and restore the state like this, and this is only one example.

Now, the error that arises from the incorrect restored state might be acceptable. Wunderlich et al. in the SMARTS paper suggest that the error is O(1). If the state dump captures a sufficient amount of state, the simulator can be warmed more quickly to recover the rest (note that the error cannot be thus eliminated, all simulation is approximation). If the error introduced is defensible, then results can still be extracted. However, I should note that the units under test are often user-supplied modules that are precisely the most difficult to save and restore.

### Lossless

There are external ways to dump the program state by querying the operating system and writing the mapped memory to a file. This method is highly dependent on the operating system to provide the relevant interfaces, and it will necessarily be different across operating systems. Furthermore, it will generally not be portable across systems that have even the most subtle differences.

The advantage of this method is that it will bit-for-bit recreate the program state, and could be relaunched from that state. However, when the program contains significant internal pointers (ChampSim has a few, as noted in #471, but the number of problematic spots are probably small), care must be taken to point them at the correct places in the restored program.

------

In summary, I agree that it's difficult. There are changes in flight that could make it easier, but there are always going to be edge cases that get in the way. Working on this is a noble goal, and the benefits are clear. But, in my opinion, the merit of the solution is based on the amount of error that the user is willing to tolerate.

Vvamos12, 2025-02-21:
Hello,

Thank you very much for your detailed explanation and valuable insights.
Your comments has greatly deepened my understanding of the challenges and approaches to implementing the checkpointing feature.
I will take your suggestions into account as I explore the best way forward.

Best regards,

closed 2025-02-21 by Vvamos12