ChampSim/ChampSim #598: update_replacement_state is not called for misses at fill time

Quangmire, 2025-02-21:
**Describe the bug**
The documentation describes update_replacement_state as:

This function has different behavior depending on whether replacement_cache_fill() is defined. If it is defined, this function is called when a tag check completes, whether the check is a hit or a miss. If it is not defined, this function is called on hits and when a miss is filled (that is, with the same timing as replacement_cache_fill()).

However, when replacement_cache_fill is absent, it doesn't look like update_replacement_state is called for misses at fill time.  This occurs for DRRIP which doesn't have replacement_cache_fill.

**To Reproduce**
Steps to reproduce the behavior:
1. Configure champsim_config.json to have DRRIP as a replacement policy.
2. Modify DRRIP to log misses.
3. Compile
4. Run

**Expected behavior**
update_replacement_state is called at cache fills

**Output**
If applicable, add the output from ChampSim to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Compiler [e.g. GCC, LLVM]
 - Version [e.g. 22]

**Additional context**
The issue appears to be at cache.h:515.

It calls impl_update_replacement_state, but since that just checks for hits and whether the replacement policy has replacement_cache_fill, nothing happens.

maccoymerrell, 2025-02-21:
I believe this is being done since update_replacement_state() would be called when the replacement state needs to update itself.
Since replacement_cache_fill is not defined (we aren't tracking fills), we must be assuming that update_replacement_state() doesn't need called.

I think this was brought up in another issue (and there may be an open PR).

maccoymerrell, 2025-02-21:
Issue #570 
PR #571

ngober, 2025-02-21:
The intent of the change in #434 was to retain the existing behavior of `update_state()`, where it was called on fills and on hits. If `cache_fill()` was defined, we would instead call it on fills and `update_state()` on hits or misses. 

It looks like #571 has gone dormant. I think it's important to see the tests for this, as mentioned in that thread. I thought we had a test for this behavior [here](https://github.com/ChampSim/ChampSim/blob/19f8c80672d697f2d766be848b68dc4a71052cf1/test/cpp/src/443-replacement-interface-selection.cc#L211). Is it not working as expected?

maccoymerrell, 2025-02-21:
@ngober 
From what I can tell, that test only shows that the interface is either the same as the previous call or not called at all. I am guessing it is actually the latter instead of the former.

Is there a place where the update_interface_discerner is reset? That might solve this problem.

ngober, 2025-02-21:
I think you're right. Resetting `update_interface_discerner` might expose a bug. I wonder if the state is being preserved between test sections.