ChampSim/ChampSim #599: DRRIP Set Dueling Broken

Quangmire, 2025-02-21:
**Describe the bug**
Two Main Issues:
- DRRIP's rand_sets is incorrectly populated with huge numbers beyond the number of sets.  See the output section below for its contents.
- DRRIP's leader set detection is likewise broken.  Even if rand_sets were correctly populated, there would only be BRRIP and SRRIP would only have one leader set each.  Assuming that the rand_sets is properly allocated, the if conditions should probably be to the effect of "if(std::distance(begin, leader) % 2 == 0)" and "if(std::distance(begin, leader) % 2 == 1)"

**Output**
Printing out the contents of rand_sets, I get:
16807
1887638
57716395
101027544
101929267
152607844
194847408
269220094
280090412
304555640
316824712
322842082
347221711
348318738
388471006
471180773
515204530
524102504
563613512
578354438
589673557
678030193
681910962
784170963
823378840
889119397
915896220
944975825
977764947
992663534
1004016855
1023129506
1044788124
1070477904
1075260298
1083454666
1102246882
1104740033
1108728549
1154112991
1190959745
1227619358
1356425228
1358580979
1400285365
1459413496
1509658266
1532062767
1534827968
1561939997
1624379149
1640687929
1732267506
1737195272
1755699915
1777724115
1784484492
1882410547
1998097157
2035308228
2044747317
2051724831
2110010672
2128236579

ngober, 2025-02-21:
I'll also add a link here to #585, which is probably more useful for the modulo operation.

Quangmire, 2025-02-21:
Yes, traditionally for a 2MB LLC, SRRIP/BRRIP was implemented to have 32 leader sets each.

A different issue arises in that number of leader sets was picked for the L2 / LLC. While I don't think it's common for either L1 cache to use DRRIP, a 64 set L1D would consequently only have leader sets and no follower sets.

ngober, 2025-02-21:
So then the number of sets needs to be set in the constructor, and not a constant. Could you submit a patch for these things?

adeelurehman, 2025-02-26:
We're having the same issue with the provided SHiP implementation. The RNG is not % with the number of sampled sets.