ChampSim/ChampSim #610: About the prefetch counting

deanding99, 2025-03-12:
I am using ChampSim to try to verify that the total prefetched  = useful prefetch + useless prefetch, but I found that printing the three variables pf_fill, pf_useful, and pf_useless does not satisfy the equation. I want to ask what I need to modify or I misunderstood this？

[image]

[image]

maccoymerrell, 2025-03-12:
This is due to the uncertainty of whether a prefetch is useful or not.

We know for certain if a prefetched line is evicted without being used it was useless, and we also know that a prefetched line that results in a demand hit is useful.

There are prefetched lines that exist in the cache that have not been evicted and have not been used by demand. This is why that equation is not satisfied.

ngober, 2025-03-12:
See also: #81, #105, #485.

deanding99, 2025-03-13:
> See also: [#81](https://github.com/ChampSim/ChampSim/issues/81), [#105](https://github.com/ChampSim/ChampSim/issues/105), [#485](https://github.com/ChampSim/ChampSim/discussions/485).

Thank you for ur reply. But my understanding is that useful prefetch, useless prefetch and fill all have duplicate counts, and there are prefetched lines that exist in the cache that have not been evicted and have not been used by demand, which causes the equation to be unsatisfied. If I change the judgment logic, can I achieve total prefetched actual data = useful + useless? It seems that the logic of the three variables needs to be modified?

maccoymerrell, 2025-03-13:
To do so, you would need a way of evaluating every prefetch that has been filled into the cache as useful or useless instantly. There is no way to do so without either speculating on the quality of the prefetch or having foreknowledge about whether the line will be used in the future before eviction.

What you could do is count these unused and unevicted prefetches as useless, though this won't be entirely accurate. It should result in filled_prefetches = useful + useless.

deanding99, 2025-03-15:
So,you mean only useless counting caused this? Because in my understanding, number of total prefetches is the count of every cacheline that was brought into L1D by the prefetcher. It is possible that a cacheline is evicted and then brought again. This should count as 2.

A prefetch is useful if it was accessed at least once after it was brought into the cache and before it was evicted from the cache. If it never got evicted, then check if this was accessed at least once.

A prefetch is useful if it was never accessed after it was brought into the cache and before it was evicted from the cache. If it never got evicted, then check if this was accessed at least once. If not, then this is a useless prefetch.
And then total = useful+useless, but it seems like the logic of pf_fill and pf_ useful meets the definition.

maccoymerrell, 2025-03-15:
The useful counter is updated immediately when a prefetch is used. Then, its internal prefetch flag is set to false, preventing double counting.
The useless counter is updated upon eviction from the cache, if the prefetch flag was never set to false.

The problem has to do with the prefetch fills that don't meet either of these criteria. It would be incorrect to label them one way or another, and its also unlikely that all prefetched lines in the cache fall into the useless/useful bins at any given point in time.

If for whatever reason you must classify all fills into the cache, then you should do a search through the cache and add any cache lines with the prefetch bit still high to the total of useless prefetches.
This will result in higher uselessness counts, significant amounts if you are only running short simulations, but should result in the numbers adding up.

I would caution against this if you are looking for meaningful numbers. A prefetch may sit in the cache for many cycles without being used. To count these as useless would be incorrect, but may happen if you are taking the above approach.

deanding99, 2025-03-16:
Yeah，I am using ChampSim to test the IPC of different benchmarks around 100M instruction. This is part of my Master's project. My supervisor asked me to modify the program so that the total number of prefetches is equal to the sum of useful prefetches and useless prefetches. At the same time, he also mentioned that the prefetches may stay in the cache for many cycles before being used, so I think this is contradictory. I am new in the cache and prefetch. I found that the logic of the source code pf_useless and useful is completely consistent with the definition, but I can't find out that their sum should be equal to a certain variable. I am trying to understand the replacement and eviction strategy of code to see if it is possible to achieve. By the way, if I wana count the prefetch  sit in the cache for many cycles without being used as useless what function should I modify?Emmm... But if I create a new function or modify a new function to count them, i agree with you that this is definitely not right because I cannot be sure whether the cache in the next cycle will hit the block that is considered useless prefetched in this cycle.

maccoymerrell, 2025-03-16:
To answer your first question, I would modify the end_phase() function in the cache.cc file. In this function, the roi (region of interest) stats are set to the current simulation phase stats (sort of like reporting the data we collected during the phase, the stats transferred to the roi stats are what will be reported in the json and command line printouts).

Inside the end_phase() function, search through the entire cache and tally up the prefetches that are in the cache (valid lines with the prefetch flag set true). Set roi_stats.pf_useless = sim_stats.pf_useless + tallied_prefetches.

If you want a better solution to this problem, then a third category besides useless and useful would be best. Something like "undetermined" or "active" prefetches, which would just be the tallied prefetches as described above.
Then, filled = useless + useful + undetermined. This way you don't compromise what the useless value is telling you.

deanding99, 2025-03-16:
[image]

[image]

As you can see, I follow your suggestion to add the counting in the end_phase function, but it seems useless+ useful +undertermined not equal to fill. Is it casued by my coding logic?

maccoymerrell, 2025-03-16:
Two things,

1. MSHR merges. I am not familiar with whether ChampSim counts merges of prefetches into demands as useful. If it does, the fill into the cache may not be marked as a prefetch type, which would mess with your numbers. 
2. Prefetches may be merged with other prefetches in MSHRs, leading to only 1 fill when multiple prefetches were issued.

deanding99, 2025-03-16:
I just use the counting of pf_fill in handle_fill function as Filled prefetch

maccoymerrell, 2025-03-16:
There might be an issue because you are using multi-level prefetching. Pf_fill will count all prefetches filled into the cache, while pf_useful and pf_useless don't tally prefetches that originate from other levels (so a L1D prefetch doesn't get tallied towards L2C useful or useless counts).

I don't think we have a way of differentiating whether a prefetch getting filled is from one level or another, so you would need to add something to do that (unless the mshr_type already has state for this, which it might).

This differentiation would need applied to unevicted prefetched sitting in the cache as well, since some might have been filled into the L2C by the L1D prefetcher.

Once you are only counting prefetches that originate on the given cache level, the numbers should add up.

deanding99, 2025-03-16:
I  get what you mean, I use this logic to do the  pf_fill counting,  It seems much more closer but still not equal like useful+useless+undertermined＞filled. Do you know if some counting still has problem? The pf_filled become 8000 from 11000 after modify.

[image]

[image]

maccoymerrell, 2025-03-16:
Could you print out the useful, useless, and unevicted counts per level? That might help to see if this is another multi-level prefetch issue or something else.

deanding99, 2025-03-16:
you mean this？

[image]

maccoymerrell, 2025-03-16:
Out of curiosity, are you running with any warmup instructions?

deanding99, 2025-03-16:
Of course, I just set the target instruction, and the warm-up instruction is automatically set to 1/5 of the target instruction. I use command like  bin/champsim --simulation-instructions 1000000 --json 437_134B.json ~/Hermes/traces/437.leslie3d-134B.champsimtrace.xz

maccoymerrell, 2025-03-16:
That may be another issue, since during warmup, prefetches will be dispatched. Once the waemup ends, there will be lines within the cache that were brought in by the prefetcher that don't have a fill associated with them, since the stats were reset.

I would try modfying begin_phase() to clear the prefetch bits on every block in the cache, which will prevent the untracked warmup fills from impacting the useless/useful statistics.

deanding99, 2025-03-16:
Yeah，that is point, I clear the bits and it works, thanks a lot

deanding99, 2025-03-28:
Hi, sorry to bother you again. After I found the equation, I did some benchmark simulation tests with 100M instructions, and I found that some results did not match the percentage of useful prefetches. For example, in the figure mcf_217, sphinx3_234B, sphinx3_417B, you can clearly see that their useful prefetches account for a high percentage (blue), but the IPC is reduced. I don't know what went wrong. In Figure 1, I used the default configuration json file and the prefetcher configuration with Ip_stride enabled in L1D to compare the IPC.

[image]

maccoymerrell, 2025-03-28:
There are several metrics to concern yourself with when prefetching. Besides usefulness, you should also consider demand latency as well as overall prefetch coverage (how many demands did I not successfully predict).

deanding99, 2025-03-28:
u mean miss latency and some data in output json file？I have check it. AVG DBUS CONGESTED CYCLE is reduced L1 and L2 hit rate increased The only problem is  that it seems like LLC miss rate increase a lot when using prefetch. So it means excessive prefetching cause the  original useful data evicted？ Is this u called timeliness? I really do not know whch data in the output file can explain this. I even doubt something wrong in the code logic or configuration file 

[429.mcf-217B100M.json](https://github.com/user-attachments/files/19510666/429.mcf-217B100M.json)
[429.mcf-217B100Mip_stride.json](https://github.com/user-attachments/files/19510667/429.mcf-217B100Mip_stride.json)

maccoymerrell, 2025-03-28:
LLC miss rate increased drastically, which can cause performance to go down, even if the ratio of useful/useless is high.

Timeliness would be that even if the data is useful, if it causes evictions of other useful data then it may be bad for performance.

There are few good ways of quantifying "good" prefetching outside of usefulness, coverage, and timeliness. Even these metrics are not foolproof. The exact performance impact of cache hits and misses in an out of order cpu is difficult to quantify.

ngober, 2025-03-28:
Maccoy is right, a useful prefetch is not desirable by the fact alone. There are many factors that go into whether and what to prefetch. The details about this are an active area of research.

I'll add to this discussion that the way that ChampSim presents its statistics may be insufficient to answer the questions we're asking here. (Again, active area of research). All of the statistics regarding cache accesses are presented in aggregate: either a count or an average. In order to assess whether a prefetch is timely, we would need to correlate the prefetch request and the corresponding demand request. This would require more robust analysis tools than ChampSim currently provides. I would love to have these tools included but, well, I suppose that takes work.