ChampSim/ChampSim #616: Performance Difference in Latest ChampSim Version

Gracre, 2025-04-21:
I'm currently using the latest version (2024-12) of ChampSim for simulation, but I've noticed it runs significantly slower than the previous version (2023-08). The latest version takes approximately 3-10 times longer to complete the same simulations, which has forced me to revert to the older version to meet my course assignment deadline.
I would appreciate any insights into what might be causing this performance regression in the newer version. Is this a known issue, or could there be specific configuration settings I should adjust?
Thank you for your assistance.

maccoymerrell, 2025-04-21:
Have you updated since pr #584 ?

Gracre, 2025-04-21:
Thank you very much for your assistance. Following the reference about the register allocator mentioned in issue #584, I was able to achieve a 5× speedup in simulation time.

closed 2025-05-05 by ngober