ChampSim/ChampSim #617: About the calculation of  accuracy

deanding99, 2025-04-29:
I am confused about how the accuracy define when I use a prefetcher. In some issues before. I saw the formula looks like useful prefetches/ prefetch miss( or prefetch filled). But when I see the paper of IPCP which is based on ChampSim.  It use useful prefetches/  PREFETCH ISSUED TO LOWER LEVEL as the accuracy. So I do not know which one is correct. 

[image]

maccoymerrell, 2025-04-29:
Prefetch usefulness as a metric doesn't have an official definition. A paper may define it one way, or another.

Generally, we could say usefulness should be the number of useful prefetches over the total prefetches filled into the cache, but this may not be what others use.

ChampSim by default doesn't report "usefulness" as a statistic, just absolute numbers of useful and useless prefetches, so it is feasible to calculate either way within ChampSim.

Just generally, you shouldn't be too concerned with the difference between calculating those different ways, assuming their difference is within a reasonable bound.
Usefulness is just a single criteria, and by itself does not indicate the quality of a prefetcher.

deanding99, 2025-04-30:
Ok, thank you for explaining it.By the way if I want to caculate coverage and timeliness to judge the prefetcher performance，what formula I could use, because It is hard for me to find the defination of these two. In my opinion based on the figure, timeliness = timely prefetches/timelt +late prefetches, coverage =perfetches access/ total miss or useful prefetches /total miss?

maccoymerrell, 2025-04-30:
Timeliness would probably be best to characterize as the difference between prefetch fill time and demand hit time, with ideal being 0. It may be best to bound it one way (so late prefetches are discarded, and end up in the useless/useful calculation instead.

Coverage would be useful prefetches over total targeted misses I think (so if you are only prefetching loads, don't include writebacks, etc...). There can be some ambiguity here, since the prefetcher could induce thrashing which can artificially inflate these numbers.
It may best best to characterize it as cache misses reduced, where 0 cache misses would indicate perfect coverage.