ChampSim/ChampSim #626: Clarification on cache latency

harry-132, 2025-05-07:
I'm trying to clearly understand how cache latencies are handled in ChampSim. Given the default latencies set as follows:

L1 latency: 5 cycles

L2 latency: 10 cycles

LLC latency: 20 cycles

If a request misses in L1 and L2, but hits in LLC, does the total latency become:

Option 1: L1 miss latency (5) + L2 miss latency (10) + LLC hit latency (20) = 35 cycles, or

Option 2: Only the LLC hit latency = 20 cycles

Could someone confirm how ChampSim calculates cache access latencies in this scenario?

Thank you!

maccoymerrell, 2025-05-07:
Option 1.

ngober, 2025-05-07:
It is actually neither of the options. The configuration file takes any of the following options: `latency`, `hit_latency`, and `fill_latency`. The `latency` option is the sum of the other two. If it is the only one specified, we divide the latency evenly between the hit and fill latencies. The hit latency is incurred on any access, the fill latency is incurred when a miss is fulfilled.

So, in your example, the total latency would be:

L1 hit latency (3) + L2 hit latency (5) + LLC hit latency (10) + L2 fill latency (5) + L1 fill latency (2) = 25 cycles

harry-132, 2025-05-07:
Thanks @maccoymerrell and @ngober — I had been confused about this for a while because of some weird results, but your explanation finally made things click. Really appreciate the clarification!

closed 2025-05-07 by harry-132