ChampSim/ChampSim #629: Cannot compile master on Ubuntu 22.04

matthewpipie, 2025-05-30:
**Describe the bug**
On a clean clone of the latest repo, I run into compile issues when running `make` for the first time.

**To Reproduce**
1. Clone repo
2. Run the vcpkg setup:
```
git submodule update --init
vcpkg/bootstrap-vcpkg.sh
vcpkg/vcpkg install
```
3. `./config.sh champsim_config.json`
4. make

**Expected behavior**
A successful build.

**Output**
```
g++ @global.options @absolute.options -MM -MT .csconfig/test/instr.d -MT .csconfig/test/instr.o -I.csconfig -MF .csconfig/test/instr.d test/cpp/src/instr.cc
cc1plus: error: to generate dependencies you must specify either ‘-M’ or ‘-MM’
g++ @global.options @absolute.options -MM -MT .csconfig/test/900-btb-bounds-check.d -MT .csconfig/test/900-btb-bounds-check.o -I.csconfig -MF .csconfig/test/900-btb-bounds-check.d test/cpp/src/900-btb-bounds-check.cc
cc1plus: error: to generate dependencies you must specify either ‘-M’ or ‘-MM’
g++ @global.options @absolute.options -MM -MT .csconfig/test/803-vmem-offset.d -MT .csconfig/test/803-vmem-offset.o -I.csconfig -MF .csconfig/test/803-vmem-offset.d test/cpp/src/803-vmem-offset.cc
cc1plus: error: to generate dependencies you must specify either ‘-M’ or ‘-MM’
g++ @global.options @absolute.options -MM -MT .csconfig/test/802-pte-page.d -MT .csconfig/test/802-pte-page.o -I.csconfig -MF .csconfig/test/802-pte-page.d test/cpp/src/802-pte-page.cc
cc1plus: error: to generate dependencies you must specify either ‘-M’ or ‘-MM’
g++ @global.options @absolute.options -MM -MT .csconfig/test/801-vmem-duplicated.d -MT .csconfig/test/801-vmem-duplicated.o -I.csconfig -MF .csconfig/test/801-vmem-duplicated.d test/cpp/src/801-vmem-duplicated.cc
cc1plus: error: to generate dependencies you must specify either ‘-M’ or ‘-MM’

...
```

Full output here: https://termbin.com/e3eu

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04.5 LTS
 - Compiler: latest version of g++ in apt (looks like g++-13)
 - Version [e.g. 22]

**Additional context**

Strangely, the compiler is complaining about -MM not being present when it is. When I remove the strange-looking `@global.options @absolute.options` from the g++ commands and rerun them manually, they seem to have different errors. For instance,
```bash
$ g++ -MM -MT .csconfig/test/instr.d -MT .csconfig/test/instr.o -I.csconfig -MF .csconfig/test/instr.d test/cpp/src/instr.cc
In file included from test/cpp/src/instr.cc:1:
test/cpp/src/instr.h:4:10: fatal error: instruction.h: No such file or directory
    4 | #include "instruction.h"
      |          ^~~~~~~~~~~~~~~
compilation terminated.
```
Seems like something is quite broken here---any ideas if this is on my end (too new or old of a compiler) or on yours? I don't think upgrading to Ubuntu 24 is an option, sadly, since this is a shared machine.

maccoymerrell, 2025-05-30:
Those errors are a result of a previously failed build (for example, previously trying to build before installing vcpkg or running the config script).

Delete .csconfig and absolute.make and repeat those steps.

closed 2025-05-30 by matthewpipie

maccoymerrell, 2025-05-30:
Make clean should remove .csconfig, though I believe it fails to remove absolute.options.

There are some changes in the pipe to the build mechanism in ChampSim, so this may be fixed when those changes come to fruition.

ngober, 2025-05-30:
I think `make configclean` might have gotten most of the way there. I don't think this would have removed absoute.make, though. Maybe we need a `make much_more_cleaner`.